RaspbianFrance/raspisms #113: Impossible d'envoyer ou de recevoir

LaurentThiol, 2018-07-02:
Bonjour
Je ne peux ni recevoir ni envoyer !

L'interface web fonctionne à 100%
Juste un message comme quoi mon numéro de phone est incorrect soit en +336XXXXXXXX ou avec le 06XXXXXXXX
Curl fonctionne avec {"error":0} en réponse

Rien vu du coté de la base de données
Gammu fonctionne bien en mode manuel direct en envoie et en réception - clé huawei E177

Cron vérifié et ok

J'ai réinstallé à blanc - 3 fois la version 2.01 et toujours la même chose...

Comment faire ?
Merci par avance pour votre aide

Mikegyver42, 2018-07-13:
Bonjour,
J'ai le même problème.
Avez-vous trouvé la solution ?
Merci

LaurentThiol, 2018-07-13:
controler si gammu-smsd est installé en faisant :
`sudo gammu-smsd`
Si pas installé : 
`sudo apt-get update && sudo apt-get upgrade && sudo apt-get install gammu-smsd`
Par sécurité :
`sudo reboot`

Dans la mesure où gammu est bien configuré, cela devrait fonctionner.

Dans mon cas de figure, j'ai toujours un soucis de connection avec la clé huawei E177.
Je continue de chercher.

closed 2018-07-16 by LaurentThiol