RaspbianFrance/raspisms #114: Suivi conso

tornlouses, 2018-07-13:
Bonjour,

Super interface et pratique.
J'arrive bien a envoyer des SMS.

En revanche j'aimerai pouvoir envoyer un SMS à mon suivi conso et pouvoir le recevoir en discussion, mais on ne peut pas ajouter de numéro court du type 555.
Est-ce possible?

Merci d'avance.
Cordialement.

LaurentThiol, 2018-07-13:
Oui c'est possible en modifiant le contrôle des numéros dans controllers/internalTools.php

tornlouses, 2018-07-14:
Merci pour le retour!
Pas bête d'aller voir dans le code! Je n'y avais pas pensé.
Par contre j'ai bien pu créé un utilisateur 555, mais je ne reçois pas le retour au sms.
Pourtant dans les discussions je vois bien le sms parti vers 555 (j'ai aussi testé +33555).
Une idée?
Merci d'avance.
Cordialement.

Steve543, 2018-12-04:
Bonjour,

Avez-vous résolu le problème ? Je dois également recevoir des SMS d'un numéro court comme 555 mais également d'autres numéros. Comment avez-vous résolu votre problème ?

Merci

closed 2021-02-14 by OsaAjani