RaspbianFrance/raspisms #116: Pas de dossier scripts après l'installation

hack701, 2018-09-08:
Bonjour,

J'ai voulu installé RaspiSMS.

Tout s'est bien passé la première fois que je l'ai installé j'ai pu testé mon script en mettant le fichier dans le dossier "scripts".

J'ai depuis reformaté mon Raspberry et lors de l'installation de RaspiSMS en 2.0.1 je ne vois plus le dossier "scripts".

J'ai essayé de créer le dossier mais mes scripts ne se lancent pas.

Merci d'avance.

LaurentThiol, 2018-09-18:
Bonjour, désolé, pas vu ton message avant,
As tu suivi les points [https://github.com/RaspbianFrance/RaspiSMS/issues/115](url) ?
Quelle version de Debian ?
As tu ré-essayé de zéro depuis ?

hack701, 2018-09-18:
Bonsoir,

Je débute sous Raspberry.

J’ai installé la dernière version de NOOBS sur le raspberry. Ensuite j’ai installé Raspbian.

Dans un premier temps, j’ai installé directement RaspiSMS mais je ne recevais pas les sms et je ne pouvais pas en envoyer.

J’ai donc installé Gammu et ensuite Gammu-smsd en suivant ces deux tutoriaux :

https://tutoandco.colas-delmas.fr/software/envoyer-sms-gammu-deamon/

https://www.journaldulapin.com/2017/03/17/sms-serveur-raspberry-pi/

Tout était fonctionnel la première fois sauf que entre temps j’ai voulu formater ma carte SD car un autre logiciel ne voulait s’installer....

Depuis aucun moyen d’avoir le dossier « scripts » dans le dossier d’installation et je ne reçois aucun sms sur raspisms.

Le reste comme l’envoi est fonctionnel.

J’ai essayé de copier le dossier « scripts » du GibHub dans le dossier RaspiSMS avec la commande mkdir et de créer mon script .sh qui marchait initialement dans le dossier «scripts» mais quand je lance le sms de commande rien ne fonctionne.

Merci d’avance.

LaurentThiol, 2018-09-19:
Ah...
Pourquoi ne pas installer Raspbian directement au formatage de la carte sans passer par NOOBS ?
La clé 3G - quel est le modèle ?
Tu installes bien la version 2.0.1 ? et pas la V 2.0 ?

hack701, 2018-09-19:
Je vais tenter de réinstaller Raspbian sans passer par NOOBS. 

La clé 3g c’est une Huawei E169.

Oui j’installe la dernière version la 2.0.1

hack701, 2018-10-04:
Il y a du nouveau. Tout est fonctionnel. Je suis reparti de zéro en installant Raspbian directement après formatage et en réapplicant le topic "synthèse des bugs"

Mon problème maintenant c'est que la clé 3G ou le raspberry se met en veille tout seul au bout d'un certain temps.

Je peux donc lancer des commandes par SMS juste après un reboot et pendant un certain temps (une heure environ) c'est encore fonctionnel mais après 12 heures ou 24 heures le SMS n'exécute pas la commande demandée.

J'ai effectué des recherches de mon problème il est connu j'ai essayé de changer différents paramètres comme préconisé mais cela ne fonctionne toujours pas.
Ci-joint les liens pour résoudre mon problème que j'ai regardé.

https://tutoandco.colas-delmas.fr/software/fixer-ports-huawei-gammu/

http://blogmotion.fr/diy/tutoriel-gammu-cle-3g-dongle-16409

https://forum.raspbian-france.fr/t/fixe-le-port-ttyusb0-sur-la-cle-3g/265/3

Merci d'avance.

LaurentThiol, 2018-10-04:
Oui, il faut empêcher la mise en veille de la clé USB dans Debian
J'ai pas le temps de tester aujourd'hui mais, je pencherai pour une action de type :  

`$ sudo nano /etc/modprobe.d/XXXX.conf`

Repérer la ligne contenant : 

`options XXXX rtw_power_mgnt=0 rtw_enusbss=0`

Remplacer le "0" par un "1" // XXXX correspond à la carte USB

hack701, 2018-10-06:
Je trouve pas le fichier XXXX.conf.

J'ai essayé la commande gammu --identify et j'ai le message "Aucune réponse dans le temps d'attente spécifié: Le téléphone n'est peut-être pas connecté."

Je ne sais plus quoi faire. Si je redémarre le Raspberry tout fonctionne correctement (envoi et réception) mais après quelques heures plus rien n'est fonctionnel et je suis obligé de redemarrer le Raspberry.

La commande gammu --identify donne toujours le même message même après redémarrage.

J'ai trouvé un topic ou certaines personnes rencontrent mon problème.

https://github.com/RaspbianFrance/RaspiSMS/issues/93

closed 2018-10-20 by hack701