RaspbianFrance/raspisms #118: Pas de copie entre inbox Gammu et receiveds RaspiSMS

Kiurg, 2018-10-26:
Bonjour à tous,

Je mets en place un serveur de SMS sur un Raspberry Pi 3 B+ dans le cadre du boulot. J'ai voulu tester RaspiSMS, l'interface étant simple claire et en français (critères obligés pour nos utilisateurs).

Après avoir parcouru un peu toutes les problématiques relatives à la config de gammu et de RaspiSMS, je bloque sur ce problème : les messages reçus dans la inbox de Gammu ne sont pas copiés dans le répertoire receiveds de RaspiSMS. Je ne sais pas s'il manque une étape évidente qui ne serait pas dans les différents tutos que j'ai pu lire, si il y a un problème de script, de droits...

- j'utilise RaspiSMS 2.0.1 installé avec le package .deb, l'interface web et la base de données sont fonctionnels
- le module GSM (carte NadHat) fonctionne très bien après config de gammu, envoi réception de SMS
- j'ai changé les chemins pour RaspiSMS (en incluant le /html manquant) partout où cela était nécessaire
- j'ai corrigé le fichier internalConsole.php (pour le fichier caché .to-keep)
- même si RaspiSMS en installation auto donne les droits nécessaires pour lecture des fichiers, je l'ai refait manuellement, en utilisateur Pi et en utilisateur root

En vérité je ne comprends pas vraiment à quel moment se fait le lien entre gammu et RaspiSMS (c'est uniquement dans les scripts de RaspiSMS ?).
Quand je lance manuellement le script parseSMS.sh, j'ai l'impression que le script tourne dans le vide et du coup arrête après un certain moment.
J'ai lu d'un utilisateur anglais qu'il avait dû lier gammu à une nouvelle table dans la base de données du serveur web, disant que cela n'était expliqué nulle part. Je pourrais faire ça, mais du coup RaspiSMS et ses scripts ne serviraient à rien non ? Et si je lie gammu à la table raspisms et non une nouvelle, je dois indiquer à gammu les différentes colonnes où écrire j'imagine.

Je sèche complètement, si quelqu'un peut m'aider c'est avec plaisir !

RaspbianFrance, 2018-10-27:
Bonjour,
Normalement il n'y a pas de liaison particulière a faire avec une base de
données.
Le c'est le script parsesms qui devrait venir récupérer les messages dans
le dossier.

Tu devrais peut être essayer de mettre des points de débug dans le script
afin de voir où il passe, s'il voit bien les messages, etc.

RaspbianFrance, 2018-10-29:
> Merci pour la réponse !
>
> Alors mise à jour, en fait le script tournait apparemment dans le vide
> quand j'appelais la commande parseReceivedSMS de la internalConsole.
> J'ai dû oublié où j'ai laissé passer ça, mais quand je lance le script
> parseSMS.sh il me sort une erreur de droits pour créer le fichier .txt dans
> receiveds ; ce serait donc là mon problème.
>
> Voici l'output quand je veux visualiser les droits attribués sur le
> dossier de RaspiSMS...
>
> drwxr-xr-x 13 www-data www-data 4096 oct. 29 09:56 RaspiSMS
>
> ...et son contenu
>
> -rwxr-xr-x 1 www-data www-data 1644 oct. 24 15:46 console.php drwxr-xr-x 2
> www-data www-data 4096 oct. 24 16:34 controllers -rwxr-xr-x 1 www-data
> www-data 3868 déc. 4 2017 createDatabase.sql drwxr-xr-x 3 www-data www-data
> 4096 oct. 23 15:32 css drwxr-xr-x 6 www-data www-data 4096 oct. 23 15:32
> font-awesome drwxr-xr-x 2 www-data www-data 4096 oct. 23 15:32 fonts
> drwxr-xr-x 2 www-data www-data 4096 oct. 23 15:32 img -rwxr-xr-x 1 www-data
> www-data 939 déc. 4 2017 index.php drwxr-xr-x 5 www-data www-data 4096 oct.
> 23 15:32 js drwxr-xr-x 2 www-data www-data 4096 oct. 23 15:32 model
> drwxr-xr-x 2 www-data www-data 4096 oct. 24 15:13 mvc -rwxr-xr-x 1 www-data
> www-data 326 oct. 25 10:54 parseSMS.sh -rwxr-xr-x 1 www-data www-data 365
> déc. 4 2017 README.md drwxr-xr-x 2 gammu gammu 4096 oct. 29 11:09 receiveds
> drwxr-xr-x 2 www-data www-data 4096 oct. 23 15:32 sounds drwxr-xr-x 17
> www-data www-data 4096 oct. 23 15:32 templates
>
> Le script est bien exécutable (1) et Gammu a l'autorisation d'écrire (2)
> dans le dossier receiveds apparemment. Après je ne suis pas super calé en
> Linux, y aurait-il une erreur que je ne vois pas ?

Salut, tu devrais essayer de faire un chmod -R 777 ./receiveds

Tu regardes si ça fonctionne, et petit à petit tu diminues les droits
jusqu'à ce que ça ne fonctionne plus et a ce moment tu reviens a l'étape
précédente.

Ou alors tu laisses en 777 mais c'est pas top comme pratique.

Kiurg, 2018-10-31:
Mise à jour : 

J'ai réussi à faire tout fonctionner ! Je mets ici mes observations et manipulations pour info.

- j'ai commencé par tout réinstaller proprement sur une distrib stretch-lite vierge, en automatique avec le paquet Debian raspisms-2.0.1 (donc qui installe les paquets nécessaires, crée la base de données, crée le compte admin RaspiSMS)
- comme j'utilise la carte modem GSM NadHat, récupération du dépôt Git pour le script permettant
d'allumer / éteindre la carte
- configuration des fichiers Gammu (gammurc et gammu-smsdrc) avec changements nécessaire niveau des chemins d'appel du script de RaspiSMS (notamment le /html manquant) et test d'envoi de SMS : jusque là OK et les SMS sont bien répliqués dans le dossier receiveds de RaspiSMS
- ensuite configuration des différents paramètres nécessaire pour que RaspiSMS se connecte à sa base de données
- à partir de là, test de lancement du script parseSMS.sh et de la commande parseReceivedSMS :
--- les constantes de chemin dans console.php ne fonctionnaient pas chez moi, j'ai dû mettre en dur /var/www/html/RaspiSMS/mvc... au tout début dans la partie INCLUSIONS.
--- les messages de la inbox de gammu ne sont PLUS visibles dans le dossier receiveds > cela veut dire que le script les traite mais bloque quelque part (problème de droits)
- j'ai noté que l'installation automatique de RaspiSMS n'installait PAS le démon gammu-smsd, que j'ai choppé manuellement avec un apt-get install gammu-smsd (ne pas remplacer la config précédemment faite)
- je n'ai PAS donné les droits à l'utilisateur gammu:gammu pour le dossier .../RaspiSMS/receiveds
- j'ai remarqué que l'accès à la inbox de gammu dans /var/spool/gammu était bloquée. J'ai fait un chmod -R 755 sur le dossier gammu entier
- et là tous les messages essayés depuis hier se sont copiés dans la base de données de RaspiSMS.

J'espère que cela en aidera certains s'ils se lancent dans le paramétrage de cette solution :)

closed 2019-03-08 by Kiurg

waazdakka, 2019-03-21:
Hello @Kiurg , après avoir pas mal cherché, j'ai tenté l'installation manuelle de gammu-smsd, qui m'a permis de faire fonctionner les sms à l'envoi... Sauf quand je passe par l'API en curl.
Et toujours rien niveau réception (Aucune réception depuis de départ en fait). J'ai tenté un chmod sur /var/spool/gammu, toujours rien :'(

Kiurg, 2019-03-21:
Bonjour ! Je n'ai pas vraiment remis les mains dessus depuis 6 mois car le projet a pris du retard, mais en tout cas ce que j'ai posté juste au-dessus fonctionne toujours (je viens de tester).

Il me semble avoir noté à un moment dans mes recherches que désactiver le démon gammu-smsd permettait à certaines fonctionnalités de RaspiSMS de fonctionner ou mieux fonctionenr. Par contre on perd l'intérêt du service qui tourne en permanence pour déclencher l'enregistrement d'un message.
Essaye déjà en le désactivant puis regarde si envoyer un sms via Gammu fonctionne, puis via curl, puis via l'interface Web en mettant un SMS programmé.

A quoi ressemble ta requête curl ? Tu as bien fait attention à l'adresse du serveur (moi j'ai dû rajouter le dossier /RaspiSMS dans le chemin URL après l'IP ce qui n'apparaît pas sur la documentation) Je viens d'essayer avec un POST et aucun soucis. Il te retourne quelque chose quand tu exécutes ? Normalement RaspiSMS devrait te retourner un message error(x), où error(0) signifie que ton message s'est bien envoyé.

waazdakka, 2019-03-21:
Merci de ta réponse !
Je viens de jeter un coup d'oeil aux logs de gammu, et je suis un peu étonné : j'ai droit à une superbe erreur 
`Thu 2019/03/21 17:01:15 gammu-smsd[624]: Starting phone communication...
Thu 2019/03/21 17:01:15 gammu-smsd[624]: Error at init connection: Error opening device, it doesn't exist. (DEVICENOTEXIST[4])
Thu 2019/03/21 17:01:15 gammu-smsd[624]: Going to 30 seconds sleep because of too many connection errors`

... Alors qu'en passant directement par gammu, je n'ai aucun problème pour effectuer l'envoi ! 
(cf #123 )

J'ai un peu de mal à comprendre ^^

Kiurg, 2019-03-22:
Argh ça m'est arrivé plein de fois aussi cette erreur. C'est dans la conf gammurc / gammu-smsdrc, selon comment tu appelle ton chemin de device il peut ne plus le reconnaître. Moi j'ai mis "/dev/serial0" et connexion "at". Ne pas oublier le code PIN aussi (mais il sort une erreur particulière quand c'est le cas).
Après sur certains Pi ou certaine distrib Linux le chemin peut s'appeler "/dev/ttyACM0". Ça dépend du bus sur lequel ton tel / ta carte GSM est branché. Là tu vas devoir faire un peu de recherches pour trouver la correspondance selon ton device utilisé.

waazdakka, 2019-03-28:
Bon en fait, j'ai suivi l'ensemble des recommandations sur les bugs relatifs à la v2.0.1 ici #115 , et tout fonctionne correctement !
Il ne me reste plus qu'à trouver pourquoi le dongle 3G passe hors ligne au bout d'un certain temps !
Merci !!!