RaspbianFrance/raspisms #119: 403 Forbidden

StrangeBlackHole, 2018-11-23:
Bonjour,

je viens d'installer raspisms via les sources sur un Raspberry PI déjà en service.
Le dongle est un Huawei E3531 et le serveur web est Nginx avec la configuration suivante:
```
server { 
	listen 80; 
	server_name raspisms.sbh; 

	error_log /var/log/nginx/raspisms.error_log info; 

	root /var/www/RaspiSMS/;
	index index.php; 

	location / {
        	try_files $uri $uri/ /index.php;
    	}	

	location ~ \.php$ { 
		try_files $uri =404; 
		include /etc/nginx/fastcgi.conf;
		fastcgi_pass unix:/var/run/php5-fpm.sock; 
	} 
}
```

Tous fonctionne très bien en émission comme en réception par contre impossible via l'interface d'aller sur  Logs->SMS reçus (http://MON_DOMAIN/receiveds/).

403 Forbidden

je pensais initialement que cela venait du fait que le dossier est configuré avec les droits  gammu:gammu mais en passant les droits à www-data:www-data c'est exactement la même chose.

Une idée ??

merci

romainguerrero, 2018-11-27:
Bonjour,

Question bête mais est-ce que les autres pages fonctionnent bien ou est-ce qu’elles renvoient aussi des 403 ?

StrangeBlackHole, 2018-11-28:
Salut,

Toutes les autres pages fonctionnent très bien. La 403 est uniquement sur receiveds.

closed 2019-01-19 by StrangeBlackHole

StrangeBlackHole, 2019-01-19:
Bon finalement avec cette config tout fonctionne bien ;)

```
server {
   listen 80;
   server_name sms.sbh.sx;

   if ($http_x_forwarded_proto != "https") {
     set $ssl 1;
   }
   if ($ssl) {
     rewrite ^(.*)$ https://$server_name$1 permanent;
   }

  error_log /var/log/nginx/raspisms.error_log info;

   root /var/www/RaspiSMS/;
   index index.php;

   location / {
     rewrite !\.(js|ico|ICO|gif|GIF|jpg|JPG|jpeg|JPEG|png|PNG|css|woff|ttf|wav|ogg|mp3)$ /index.php;
     try_files $uri $uri/ /index.php;
   }

   location /receiveds {
     rewrite ^/(.*)$ /index.php?/$1 last;
     break;
   }
   location ~ \.php$ {
     try_files $uri =404;
     include /etc/nginx/fastcgi.conf;
     fastcgi_pass unix:/var/run/php5-fpm.sock;
   }
}
```