RaspbianFrance/raspisms #127: Réponse après une commande

titithen00b, 2019-08-17:
Bonjour,

Est-il possible d'envoyer un sms au numéro qui à intenté la commande avec le résultat de celle-ci ?

leknoppix, 2019-08-23:
Bonjour

Je souhaite t'aider mais là, j'ai du mal à comprendre ce que tu veux. Tu veux en gros, quand un sms par vers un utilisateur A recevoir un sms de confirmation sur la ligne qui envoie l'SMS. Est ce cela? Si oui, je pense que tu devrais regarder du coté de fichier receivesSMS.sh (si mes souvenirs sont bon). Ce script permet de récupérer le statut des sms traité. Ainsi, si tu arrives à comprendre le script, lorsqu'il y a l'accusé de reception, tu peux émettre un sms vers une autre ligne. attention, cela risque de tourner en boucle si tu fais mal ton script, je m'explique.

Tu envoies un sms a toto.
Le téléphone de toto le reçoit, donc envoie de l'accusé de réception.
Analyse de l'accusé et envoi d'un SMS vers l'admin.
A sa réception, nouveau accusé de réception, envoi d'un SMS vers admin et ainsi de suite.

L'idée, mais là, jamais tester serait de bannir le numéro de l'admin pour éviter cette boucle indefini.

Raspisms tel que ne permet pas ce que tu veux et bien sur, si c'est bien ce que j'ai compris.

Si je suis parti en live et que cela ne correspond pas à ta question, reformule là. Si tu as d'autres questions n'hésite pas à poser.

Leknoppix

titithen00b, 2019-08-24:
Hello,

En gros, prenant une commande toute bête, un apt upgrade --yes (ou un install xxx) je voudrais recevoir le résultat de ma commande. J'avais pensé à utiliser par exemple (dans mon script)  `sudo apt upgrade | gammu  --sendsms TEXT 06xxxxxxxx`

closed 2021-02-14 by OsaAjani