RaspbianFrance/raspisms #129: Now we need MariaDB instead of MySql

sbpbontant, 2019-10-30:
Hi,
I wanted to install the last version on my Pi 4  with the latest version of the OS
And it refuses to install Mysql, forcing to use MariaDB instead

I installed all, forced the install so the DB is ready

But when I want to send an sms, there's an Error .. But don't know what is the error..

Could you check it ? Tell me how to get more informations about the error ?

I tried addind the message to send throw the database, adding the 2 rows on the tables worked and the sms was sent as expected

Thanks to fix the use of MariaDB and help me solve my problem.
Bye and thanks for all. hope to use if fast !

sbpbontant, 2019-10-30:
J'suis un peu idiot... Ou formaté.. J'écris en anglais. Désolé :p

abdelhakel, 2019-11-11:
> Hi,
> I wanted to install the last version on my Pi 4 with the latest version of the OS
> And it refuses to install Mysql, forcing to use MariaDB instead
>
> I installed all, forced the install so the DB [...]

J'ai aussi essayé l'installation sur Rasbian Buster (v10) et ça m'affiche beaucoup d'erreurs. Systématiquement, ce sont les actions qui font appel à des fichiers dans  `/var/www/html/RaspiSMS` qui posent problème. Il s'avère que ce dossier est .... vide. Si quelqu'un a un tuto à jour pour Raspbian Buster je suis preneur

theos10, 2019-11-25:
Bonjour, J'ai installé RaspiSMS sur Rasbian  Buster, avec Nginx php-fpm et Mariadb.
Tous fonctionne excepté l'envoi de sms lorsque l'on répond a un sms, ou l'ajout d'un sms programmé.
Le message 'Improssoble de cree le sms' s'affiche.
Je ne trouve rien dans les logs.
Merci de votre aide.

abdelhakel, 2019-11-25:
T'es donc plus avancé que moi. je suis preneur des consignes pour arriver à arriver au bout de l'installation, et voir ensuite si j'ai le même bug que toi...

sbpbontant, 2019-11-25:
J'ai corrigé de mon côté phpmyadmin qui mettait systématiquement une erreur
(via google, j'ai lu qu'il fallait rajouter une série de  parenthèses sur
un des fichiers)
et j'ai fait une modif sur une insertion en base de mémoire
j'ai changé (comme un cochon) le système d'envoi des sms en automatique (ça
fonctionne via le site et via l'api)
seuls soucis pour le moment, je n'arrive pas à recevoir les sms et je ne
sais pas où chercher
Dès que ça marche, je mets une image de mon système pour une installation
100% terminée

Ghislain, essaie de créer les 2 enregistrements dans la base de donnée
directement pour voir si ça part bien.
Je suis preneur d'infos sur la réception des messages qui reste pour moi un
gros soucis (je ne sais pas si c'est gammu ou le site ou autre chose qui
déconne)

Merci

theos10, 2019-11-26:
> T'es donc plus avancé que moi. je suis preneur des consignes pour arriver à arriver au bout de l'installation, et voir ensuite si j'ai le même bug que toi...
On va donc s'entraider, mais je vais avoir besoin d'information :
serveur apache ou nginx, version de php, etc.
tu a fait un git clone dans /var/www/html/ ?
A tu ajouté les droits qui vont bien ?

theos10, 2019-11-26:
> J'ai corrigé de mon côté phpmyadmin qui mettait systématiquement une erreur (via google, j'ai lu qu'il fallait rajouter une série de parenthèses sur un des fichiers) et j'ai fait une modif sur une [...]

Pour la réception, la premier chose a vérifier, c'est si le demons gammu-smsd reçois bien les sms.
Pour ça, regarde dans /var/spool/gammu/inbox/ si ton sms arrive correctement.
Si c'est le cas, c'est un pb de raspisms, en l'occurrence, le parseur qui ne fonctionne pas.

sbpbontant, 2019-11-26:
J'ai envoyé un sms tout à l'heure et il n'apparait pas.
J'ai des fichiers dans le dossier mais ils datent. (pourtant la date est récente mais ils sont plus anciens il me semble)
Et le dernier envoyé n'apparait pas.
Une idée de ce que je dois corriger ?

[image]

theos10, 2019-11-26:
tu a reçu des sms le 26/11/2019 a 12h11 donc les sms arrive bien.
peux tu envoyer le fichier /etc/gammu-smsdrc ?

sbpbontant, 2019-11-26:
Voici des captures du fichier /etc/gammu-smsdrc et du shell

Je ne sais plus si j'ai précisé, mais j'utilise apache

abdelhakel, 2019-11-26:
> T'es donc plus avancé que moi. je suis preneur des consignes pour arriver à arriver au bout de l'installation, et voir ensuite si j'ai le même bug que toi...
> On va donc s'entraider, mais je vais [...]

J'ai suivi les instructions suivantes : 
```
wget http://raspisms.raspberry-pi.fr/download/raspisms-2.0.1.deb
sudo apt-get update
sudo dpkg -i ./raspisms-2.0.1.deb
sudo apt-get -f install
```
J'ai eu quelques messages d'erreur avec mysql (j'ai rentré à la main le mot de passe dans le fichier `/var/www/html/RaspiSMS/mvc/constants.php.` Mon interface fonctionne. Mais impossible d'envoyer un sms. l'url de l'API ne fonctionne pas.
Et bizarrement, je n'ai pas de dossier 'gammu' dans `/var/spool/`

[image]
[image]

sbpbontant, 2019-11-27:
> J'ai suivi les instructions suivantes :
>
> ```
> wget http://raspisms.raspberry-pi.fr/download/raspisms-2.0.1.deb
> sudo apt-get update
> sudo dpkg -i ./raspisms-2.0.1.deb
> sudo apt-get -f install
> ```
>
> J'ai [...]

Deux soucis (de mémoire)
Dans l'insertion dans une des deux tables, il faut ajouter un champ qui n'a pas de valeur par défaut et donc la forcer

Et l'autre soucis, c'est qu'il te manque le package du démo gammu-smsd (de mémoire aussi)

abdelhakel, 2019-11-27:
> Deux soucis (de mémoire)
> Dans l'insertion dans une des deux tables, il faut ajouter un champ qui n'a pas de valeur par défaut et donc la forcer
> 
> Et l'autre soucis, c'est qu'il te manque le package du démo gammu-smsd (de mémoire aussi)

Justement, c'est tout le mystère. Gammu est bien installé et fonctionnel. Cf l'image ci-dessous. Mais il n'existe pas de "service' Gammu pour faire du `status, start, stop`.
quand tu dis 

> Dans l'insertion dans une des deux tables, il faut ajouter un champ qui n'a pas de valeur par défaut et donc la forcer

tu aurais un moyen de retrouver la manip exacte et le fichier concerné ?
[image]

abdelhakel, 2019-11-27:
> Deux soucis (de mémoire)
> Dans l'insertion dans une des deux tables, il faut ajouter un champ qui n'a pas de valeur par défaut et donc la forcer
> 
> Et l'autre soucis, c'est qu'il te manque le package du démo gammu-smsd (de mémoire aussi)

> Deux soucis (de mémoire)
> Dans l'insertion dans une des deux tables, il faut ajouter un champ qui n'a pas de valeur par défaut et donc la forcer
> 
> Et l'autre soucis, c'est qu'il te manque le package du démo gammu-smsd (de mémoire aussi)

Justement, c'est tout le mystère. Gammu est bien installé et fonctionnel. Cf ci-dessous. Mais il n'existe pas de "service' Gammu pour faire du `status, start, stop`.

```
pi@raspberrypi:~ $ gammu sendsms TEXT 06XXXXXXXX -text "Test"
Si vous désirez interrompre, appuyez sur Ctrl + C...
Envois du SMS 1 1 / 1 2.... Attente d'une réponse réseau..OK, message de référence=69 1
pi@raspberrypi:~ $ systemctl status gammu-smsd.service
Unit gammu-smsd.service could not be found.
pi@raspberrypi:~ $ ls /var/spool/
cron/        exim4/       libreoffice/ mail/        rsyslog/     samba/ 
```
Et quand tu dis 

> Dans l'insertion dans une des deux tables, il faut ajouter un champ qui n'a pas de valeur par défaut et donc la forcer

tu aurais un moyen de retrouver la manip exacte et le fichier concerné ?
[image]

abdelhakel, 2019-11-27:
En fait, je n'avais pas installé gammu-smsd. C'est fait. Mais reste ce problème de sms impossible à créer. @sbpbontant  je crois que mon dernier problème réside dans ces soucis de tables et de champs à rajouter. Ton aide me serait précieuse !

theos10, 2019-11-27:
> Message du 27/11/19 22:31> De : "Abdelhak" > A : "RaspbianFrance/RaspiSMS" > Copie à : "FOURNIER Ghislain" , "Comment" > Objet : Re: [RaspbianFrance/RaspiSMS] Now we need MariaDB instead of MySql (#129)> >

J'ai fait des tests avec mysql, ce n'est pas un probleme de base de donné.
d'autant que ca fonctionne sur debian jessie.

theos10, 2019-11-28:
Perso je n'ai pas fait l'installation via le .deb, mais par git clone.
Dans une premier temps, j'ai utiliser une debian jessie, et ça fonctionnait parfaitement.
J'ai donc effectuer la même install sur une debian 10 Buster. Et c'est la que le problème est apparut.
En Jessie, php7.0 était installé, sous Buster, c'est 7.1, 7.2, 7.3. j'ai essayer les 3 et même résultat. Pour moi c'est un fonction PHP qui est déprécié.
J'ai créé un Fork, et j'ai recherche dans chaque autres Fork, les modifs, pour avoir tous les correctifs de chacun, J'ai même ajouter un fonction pour afficher la puissance du signal sur la dashboard.
Nous sommes donc plusieurs a avoir le même problème. Je ne suis pas tres calé en PHP POO,donc ca va etre compliqué.
Mise a part, il ne faut pas confondre gammu et gammu-smsd, ce dernier est le daemon qui tourne en tache de fond pour envoyer et recevoir les sms. D’ailleurs quand gammu-smsd tourne, il est impossible d'envoyer des sms avec gammu, car le périphérique est utilisé. il faut utiliser gammu-smsd-inject .
Autre chose, il faut ajouter deux lignes dans le fichier sudoers:
Ajouter gammu sans mot de passe dans le fichier sudoers
sudo visudo
ajouter :
gammu ALL=(ALL) NOPASSWD: ALL
www-data ALL=(ALL) NOPASSWD: /usr/bin/gammu-smsd-inject
démarrer gammu avec l’utilisateur root
dans /etc/init.d/gammu-smsd
changer USER=gammu en USER=root
et aussi : 
sudo usermod -a -G dialout www-data
Puis ajouter les droits sur le répertoire receiveds
chown -Rf gammu:gammu /var/www/html/receiveds

Je pense avoir fait le tour.

theos10, 2019-12-01:
J'ai isolé le bout de code qui pose probleme :
[image]
C'est un souci de champ Location dans le header http : 
[image]
Reste a trouver comment resoudre cette erreur.

vincent1890, 2020-02-23:
Pour revenir a la question de départ 👍 
Maintenant, nous avons besoin de MariaDB au lieu de MySql
Comment faire SVP ?
----------------------------------------
Back to the original question 
Now we need Mariadb instead of Mysql
How can I please ?

OsaAjani, 2020-02-24:
Salut, la dernière version du .deb ne demande plus MySQL mais bien mariadb comme il le devrait.

vincent1890, 2020-02-24:
> Salut, la dernière version du .deb ne demande plus MySQL mais bien mariadb comme il le devrait.

Salut @OsaAjani 
Merci pour l'info je ne savais pas que la 2.0.2 était sortie j'étais toujours sur la raspisms-2.0.1.deb
Merci super colle vais pouvoir tester sur le RasPi 4
Merci encore pour tous ce boulot ce pack deb qui est super génial (je l'utilise depuis deux ans)

closed 2021-02-14 by OsaAjani