RaspbianFrance/raspisms #132: Message d'erreur alors que ça fonctionne.

Pozinux, 2020-01-17:
Bonjour,

Lorsque j'envoi un sms programmé à un contact enregistré dans le répertoire, j'ai un message d'erreur : 

"Le SMS a bien été créé, mais certains numéros ne sont pas valides."

Pourtant le message arrive bien chez le destinataire. 

Une idée ?

Cordialement,

Timothé

Pitoo67, 2020-01-17:
Bonsoir,

Effectivement, j’ai aussi été confronté à ce bug. 
Le script affiche ce message dès lors que le champs de n° de destinataire est vide, sans se soucier si un autre destinataire est présent dans la liste des contacts existants, mais cela n’empêche pas le SMS de partir.

Il manque un test pour éviter ce message, malheureusement, ayant corrigé ceci sur mon système il y a plusieurs années, je suis bien incapable de vous préciser comment faire.

Cordialement

_____________________
Pierre BLANCHE

pierre@pitoo.com

Pozinux, 2020-01-17:
D'accord !! Je n'avais pas pensé que le problème venait de là. Ok merci beaucoup je devrais pouvoir le résoudre assez simplement dans ce cas.

Merci encore !

Pozinux, 2020-01-17:
Code corrigé avec ce test if : 

```
			# s'il y a des numéros dans "Numéros cibles"
			if ($numbers[0])
			{
				foreach ($numbers as $number)
				{				
					if (!$number = internalTools::parsePhone($number))
					{
						$errors = true;
						continue;
					}

					if (!$db->insertIntoTable('scheduleds_numbers', ['id_scheduled' => $id_scheduled, 'number' => $number]))
					{
						$errors = true;
					}
				}
			}
```

Merci Pittoo67 !

closed 2020-01-17 by Pozinux