RaspbianFrance/raspisms #135: Contrôler soft à distance ? (dyndns)

FredericFsa, 2020-02-03:
Bonjour,

Je viens de tomber sur votre projet.. waw ! quel boulot !
J'aurais aimé utiliser le soft a distance .

Mais je n'arrive pas a la faire ! J'aimerais par exemple l'utiliser sur le port 9999 et faire un transfert vers mon routeur.
Au niveau du routeur, pas de soucis.. mais pour la config du raspberry.. je n'arrive pas !

J'ai beau lire les commentaires en dessous.. rien n 'y fait , je dois m'y prendre comme un manche :-)
Pouvez-vous m’éclairer ?

Merci

OsaAjani, 2020-02-03:
Salut, si tu veux utiliser un port autre que 80 il faut déjà que tu t'assures d'avoir configuré Apache pour écouter sur ce port !

FredericFsa, 2020-02-04:
Hello,

Après moult recherches, j'ai réussi a forwarder le port !
J'ai choisi le port 86.

J'explique un peu pour les novices comme moi au cas où :
dans /etc/apache2/ports.conf
Ajouter ligne -> Listen 86

dans /etc/apache2/sites-enabled/000-default.conf
changer <VirtualHost *:80> par <VirtualHost *:86>

dans /var/www/html/RaspiSMS/mvc/constants.php
changer  define('HTTP_PORT', '80'); //On définit le port sur lequel est raspisms
par  define('HTTP_PORT', '86'); //On définit le port sur lequel est raspisms

Ne pas oublier de forwarder le porte 86 vers l'ip de la pi

Et le tour est joué.

Pozinux, 2020-02-11:
Parfait merci pour le partage. 

Grâce à toi j'ai également réussi à forwarder sur le port de mon choix.
 
J'ai également créer un nom dns chez OVH que j'ai rattaché à la freebox dans "DNS dynamique" et ça fonctionne nickel. J'ai dû mettre un .htaccess également dans /var/www/html/ pour pouvoir réécrire l'url en "sms.mon_domain.fr/RaspiSMS" pour que la page d'accueil arrive directement sur le login RaspiSMS lorsque j'entre sms.mon_domain.fr.

Par contre suite au changement de port comme tu l'as indiqué, je ne peux accéder au site qu'en précisant le port de la manière suivante : 

sms.mon_domain.fr:mon_port

alors que j'aimerais pouvoir faire tout simplement :

sms.mon_domain.fr

Mais je ne sais pas pourquoi je bloque quelque part... Je ne sais pas si c'est ce que j'entre dans la freebox pour forwarder vers mon_port ou s'il faut modifier le .htaccess.

Merci en tous cas.

Tim

OsaAjani, 2020-02-27:
Salut, c'est parce que par défaut ton navigateur cherche sur le port 80 pour du HTTP. Comme tu es réglé en 86 bah forcément il trouve pas. 
Si tu veux écouter sur le port 86 pour le Pi, mais accéder via le port 80 coté navigateur tu dois pouvoir configurer le port forwarding de la box pour forwarder son port 80 à elle vers le port 86 du pi.

futuriste, 2020-09-22:
Merci @FredericFsa  grâce à toi j'ai pu configurer mon service.  Je trouve l'outil très intéressant ce serait encore mieux si durant le processus de configuration automatique il proposait de modifier le port du serveur.

closed 2021-01-10 by OsaAjani