RaspbianFrance/raspisms #137: Problème d'installation de RaspiSMS avec une erreur chown: utilisateur incorrect: « gammu:gammu »

StrangeW, 2020-02-28:
Bonjour, 
Je suis en train de faire une intallasion de raspisms sur mon Raspberry Pi 3 en suivant ce tuto :
https://raspberry-pi.fr/raspisms-raspberry-pi-sms-internet/
J'ai pris le choix de la méthode 2 soit : Installer RaspiSMS sur un système de type Debian avec le paquet .deb

Après avoir lancer "sudo dpkg -i ./raspisms-2.0.2.deb" J'ai une erreur mais c'est normal :
pi@raspberrypi:~/RaspisSMS $ sudo dpkg -i ./raspisms-2.0.2.deb
Sélection du paquet raspisms précédemment désélectionné.
(Lecture de la base de données... 93529 fichiers et répertoires déjà installés.)
Préparation du dépaquetage de ./raspisms-2.0.2.deb ...
Dépaquetage de raspisms (2.0) ...
dpkg: des problèmes de dépendances empêchent la configuration de raspisms : raspisms dépend de apache2 ; cependant :  Le paquet apache2 n'est pas installé.
 raspisms dépend de php ; cependant :  Le paquet php n'est pas installé.
 raspisms dépend de php-mysql ; cependant :  Le paquet php-mysql n'est pas installé.
 raspisms dépend de php-mbstring ; cependant :  Le paquet php-mbstring n'est pas installé.
 raspisms dépend de gammu ; cependant :  Le paquet gammu n'est pas installé.
 raspisms dépend de mariadb-server ; cependant :  Le paquet mariadb-server n'est pas installé.

dpkg: erreur de traitement du paquet raspisms (--install) : problèmes de dépendances - laissé non configuré
Des erreurs ont été rencontrées pendant l'exécution : raspisms

Je lance donc "sudo apt-get -f install" et là j'obtiens, après avoir dit oui à la configuration automatique de gammu  : 
Le dossier a bien été donné à www-data, avec les droits 755
Attribution du dossier /var/www/html/RaspiSMS/receiveds à l'utilisateur gammu
chown: utilisateur incorrect: « gammu:gammu »
Le dossier a bien été donné à gammu, avec les droits 755

Dans les infos, le fichier /etc/gammurc est vide et le fichier /etc/gammu-smsdrc lui, pointe vers des dossiers qui n'existe pas dans /var/spool 

Avez vous des idées pour moi?? 
Merci d'avance.

StrangeW, 2020-03-02:
Donc je vais étoffer ma demande. 
La carte lecteur SIm est : https://www.itead.cc/wiki/RPI_SIM800_GSM/GPRS_ADD-ON_V2.0#Download
Le Raspberry pi 3 Model B VI.2 
La carte reçoit bien les SMS (Diode clignote et AR), mais impossible de lire avec GAMMU le port /dev/ttyAMA0 avec la connection at115200 [ce tuto](https://blog.garatronic.fr/index.php/fr/tutoriel-fr/27-envoyez-des-sms-avec-nadhat-et-gammu-fr)
Si quelqu'un a une idée sur comment interfacer les 2, je suis preneur.

StrangeW, 2020-03-03:
Donc problème matériel pour le fonctionnement avec Gammu. 
Réussi à faire tournée la carte avec les commande AT après l'activation du port en mode Sérial si j'ai bien compris.
Il reste un problème avec RaspiSMS qui ne veux créer que des User Admin et aucun SMS, mais je vais regarder de mon coté et voir si je ré ouvre une nouvelle issue pour ça. 
Merci pour ce bel outils

closed 2020-03-03 by StrangeW

MattBHub, 2020-03-22:
@StrangeW j'ai eu la même erreur que toi concernant l'utilisateur incorrect à l'installation.
L'interface raspiSMS fonctionne visiblement bien (j'ai modifié ce matin le fichier model.php qui empêchait de créer un sms dans l'interface en suivant le fix #138) mais les sms ne sont pas envoyés, ni reçus. 
Une idée de là où je peux regarder stp ? (J'ai vérifié le code pin de la sim évidement)

RaspbianFrance, 2020-03-22:
Salut, si tu as fait l'installation il y a quelques jours il y avait un bug
gammu-smsd était pas installé.

StrangeW, 2020-03-22:
@MattBHub Je ne sais pas où tu en est, mais sur ma première demande, le problème venait pas de RaspiSMS mais de moi. 
Après recherche, j'ai d'abord validé que mon extension ne marchait pas grâce au commande AT. Une fois que tes commandes AT peuvent envoyé un SMS, il faut le faire avec les commandes Gammu. Si tu as ces 2 pré requis, il ne devrait pas y avoir de problème.

MattBHub, 2020-03-22:
j'ai passé un bon moment à chercher. Visiblement gammu-detect ne renvoi rien, du coup le fichier de conf gammurc est vide lui aussi. Donc gammu-smsd ne se lance pas...
Merci pour votre aide. J'essaierai à nouveau plus tard (le dongle est neuf et je suis sûr qu'il a été détecté à un moment car je me souviens d'un gammurc paramétré)

StrangeW, 2020-03-22:
@MattBHub Ok, je pars donc du principe que les commandes AT sont passé. 

Pour ma part, ce n'était pas un dongle USB mais une carte sur les pins. J'ai du configurer gammu à la main. Faudrait donc chercher l’interfaçage entre ton dongle et gammu. 

Bonne chance à toi.