RaspbianFrance/raspisms #143: Erreur inconnue. (UNKNOWN[27]) avec dongle Huwaï E3531

FredMa01, 2020-10-10:
Bonjour,

J'ai installé avec succès RaspiSMS sur mon pi4, avec une clé Huwaï E3531.
La clé était reconnue comme un stockage de masse en ''sdb''
`Bus 001 Device 006: ID 12d1:14dc Huawei Technologies Co., Ltd. E33372 LTE/UMTS/GSM HiLink Modem/Networkcard`
Pour la mettre en mode modem, j'ai crée le fichier /etc/usb_modeswitch.d/12d1:1f01 en le remplissant comme indiqué :
```
Huawei E353 (3.se)
TargetVendor= 0x12d1
TargetProduct= 0x14dc
MessageContent="55534243123456780000000000000011063000000100010000000000000000"
NoDriverLoading=1
```
Le lsusb | grep -i huawei a changé : `Bus 001 Device 006: ID 12d1:155e Huawei Technologies Co., Ltd.`

Pour fixer le chemin afin qu'il ne change pas si d'autres périphériques sont connectés. J'ai choisi : ttyUSB-3G.
 nano /etc/udev/rules.d/98-usb-serial.rules
`SUBSYSTEM=="tty", ATTRS{idVendor}=="12d1", ATTRS{idProduct}=="155e", SYMLINK+="ttyUSB-3G"`

J'ai modifié la config de Gammu :
nano /etc/gammu-smsdrc
```; Configuration file generated by gammu-detect.
; Please check The Gammu Manual for more information.

[gammu]
GammuCoding = utf8
GammuLoc = fr_FR.UTF8
device = /dev/ttyUSB-3G
name = Phone on USB serial port HUAWEI HUAWEI_Mobile
connection = at
atgen_setCNMI=1,2,0,0,0

# SMSD configuration, see gammu-smsdrc(5)
[smsd]
service = files
logfile = syslog
# Increase for debugging information
debuglevel = 0
RunOnReceive = /var/www/html/RaspiSMS/parseSMS.sh
# Paths where messages are stored
inboxpath = /var/spool/gammu/inbox/
outboxpath = /var/spool/gammu/outbox/
sentsmspath = /var/spool/gammu/sent/
errorsmspath = /var/spool/gammu/error/
```
J'ai redémarré le daemon :
```
systemctl daemon-reload
systemctl start gammu-smsd
```

J'ai mis la bonne variable dans le fichier de service de Gammu. 
nano /lib/systemd/system/gammu-smsd.service
en remplaçant: EnvironmentFile=-/etc/sysconfig/gammu-smsd par EnvironmentFile=-/etc/gammu-smsdrc 

Tout à alors très bien fonctionné toute la soirée. 
Malheureusement, le lendemain cela ne fonctionnait plus : ni envoi, ni réception de SMS

```
Oct 10 13:14:21 pi4 gammu-smsd[20074]: Error getting send status of message: Erreur inconnue. (UNKNOWN[27])
Oct 10 13:14:22 pi4 gammu-smsd[20074]: Message without SMSC, assuming you want to use the one from phone
Oct 10 13:14:48 pi4 gammu-smsd[20074]: Error getting send status of message: Erreur inconnue. (UNKNOWN[27])
Oct 10 13:14:49 pi4 gammu-smsd[20074]: Message without SMSC, assuming you want to use the one from phone
Oct 10 13:15:01 pi4 CRON[22184]: (root) CMD (cd /var/www/html/RaspiSMS && /var/www/html/RaspiSMS/console.php -c sendScheduleds > /dev/null 2>&1)
Oct 10 13:15:01 pi4 CRON[22185]: (root) CMD (cd /var/www/html/RaspiSMS && /var/www/html/RaspiSMS/console.php -c parseReceivedSMS > /dev/null 2>&1)
Oct 10 13:15:01 pi4 CRON[22188]: (root) CMD (cd /var/www/html/RaspiSMS && /var/www/html/RaspiSMS/console.php -c sendTransfers > /dev/null 2>&1)

```
Une idée ?
Un problème de mise en veille du dongle ?

Merci

OsaAjani, 2020-10-10:
Salut, c'est un problème de gammu à priori donc il faudrait voir direct sur le github du projet, mais j'ai trouvé ça qui pourrait aider https://github.com/gammu/gammu/issues/462 (https://github.com/gammu/gammu/issues/462)

FredMa01, 2020-10-10:
Super, cela fonctionne. Merci !

closed 2020-10-10 by FredMa01

FredMa01, 2020-10-11:
Enfin, cela fonctionne mais ce n'est pas encore tip top.
En effet, le script purge les SMS en erreur :
`RunOnFailure = /var/spool/gammu/on-error.sh`

Cela dépanne bien jusqu'à un moment où il y a trop souvent de SMS en erreur, sans que l'on comprenne vraiment pourquoi, avec des destinataires qui ne posaient pas pb.... 
```
 Starting run on failure: /var/spool/gammu/on-error.sh OUTC20201011_144830_00_+336xxxxxxxx_sms0.smsbackup
gammu-smsd[32457]: Process finished successfully
gammu-smsd[32457]: Message without SMSC, assuming you want to use the one from phone
gammu-smsd[32457]: Error getting send status of message: Erreur inconnue. (UNKNOWN[27])
gammu-smsd[25306]: Starting run on failure: /var/spool/gammu/on-error.sh OUTC20201011_145033_00_+336xxxxxxxx_sms0.smsbackup
gammu-smsd[32457]: Process finished successfully
gammu-smsd[32457]: Message without SMSC, assuming you want to use the one from phone
gammu-smsd[32457]: Error getting send status of message: Erreur inconnue. (UNKNOWN[27])
gammu-smsd[25512]: Starting run on failure: /var/spool/gammu/on-error.sh OUTC20201011_145709_00_+336xxxxxxxx_sms0.smsbackup
```
Exemple de rapport 
```
; This file format was designed for Gammu and is compatible with Gammu+
; See <http://www.gammu.org> for more info
; Saved 20201011T150639 (dim. 11 oct. 2020 15:06:39 )

[SMSBackup000]
; Le SMS a bien été créé, mais certains numéro ne sont pas valides.
PDU = Status_Report
DateTime = 20201011T150639
State = Sent
Number = "+33667xxxxxx"
NumberUnicode = 002B00330033003600360037003000390038003300390031
Name = ""
NameUnicode =
Text00 = 004C006500200053004D0053002000610020006200690065006E002000E9007400E900200063007200E900E9002C0020006D0061006900730020006300650072007400610069006E00730020006E0075006D00E90072006F002$
Text01 = 006E00740020007000610073002000760061006C0069006400650073002E
Coding = Default_No_Compression
Folder = 1
Length = 65
Class = -1
ReplySMSC = False
RejectDuplicates = False
ReplaceMessage = 0
MessageReference = 0

```
Une piste ? 
Merci
Fred