RaspbianFrance/raspisms #146: Envoi SMS via l'API dans le crontab

FredMa01, 2020-12-20:
Une idée de la raison pour laquelle cette requête GET traditionnelle ne pose aucun problème (espace et accents) :
`curl http://URLCIBLE/smsAPI/?email=admin@email.tld\&password=p455w0rd\&numbers=0612345678\&text=Texte%20du%20SMS%20à%20envoyer`
et s'arrête au premier % quand on la colle dans crontab ?
'30 7 * * *  /usr/bin/curl http://URLCIBLE/smsAPI/?email=admin@email.tld\&password=p455w0rd\&numbers=0612345678\&text=Texte%20du%20SMS%20à%20envoyer'

J'ai essayé de mettre des guillemets à différents endroits mais sans succès....

OsaAjani, 2021-01-10:
Salut, si la requête fonctionne dans curl et et pas dans la crontab c'est le que le problème est à chercher du coté de la crontab.
D'après la doc de crontab, `%` est [un caractère spécial pour crontab ](https://stackoverflow.com/questions/27123367/percent-sign-not-working-in-crontab).

Il faudrait donc remplacer les `%` par `\%`.

closed 2021-01-10 by OsaAjani

FredMa01, 2021-01-10:
Merci pour cette réponse !

Fred