RaspbianFrance/raspisms #147: Erreur d'envoie / reception de SMS

Oliviercreativ, 2021-01-03:
Bonjour !
J'ai réussi à l'installer RASPISMS, après avoir mis à jour en php 7.3.
Je n'arrive pas à envoyé ou recevoir un SMS ... En regardant dans les log j'ai trouvé cette ligne : 

[2021-01-03T17:58:53.741368+01:00] RaspiSMS Daemon Phone 7.INFO: Error reading received smss : Cannot unlock sim. [] []

Et pourtant quand j'ai ajouté le téléphone, j'ai bien mis mon code PIN ...
Avez-vous une idée svp ?

Merci

OsaAjani, 2021-01-03:
Bonjour, il faudrait vérifier si gammu arrive bien à unlock la carte sim directement en ligne de commande.

Oliviercreativ, 2021-01-03:
Quand je fais " gammu-detect ", il me dit que la carte est busy ou sans les droits ... Je vais refaire une install correctement.
Merci

RaspbianFrance, 2021-01-03:
Premier truc, j'essaierai de vérifier si la carte sim est dans le bon sens.
Ça m'est arriver TELLEMENT souvent.

Oliviercreativ, 2021-01-03:
Elle est dans le bon sens ! 
Avec gammu-smsd, j'arrive à envoyer un sms et les recevoir.
Du coup j'ai remis la dernière version de l'OS et je vais tenter l'installer comme sur le site.
J'ai pris l'installation sur raspisms.fr.

Oliviercreativ, 2021-01-03:
J'utilise un carte waveshare SIM800C GSM/GPRS, je l'ai trouvé sur amazon : https://amzn.to/2X8FNMl

closed 2021-01-06 by Oliviercreativ

OsaAjani, 2021-01-06:
Tu peux indiquer comment tu as résolu ton problème ?