RaspbianFrance/raspisms #148: Envoi SMS Groupé

shaquillou, 2021-01-04:
Bonjour,
Je découvre RaspiSMS et je souhaite gérer les SMS entrants.
Le but est d’envoyer un SMS au Pi pour qu’il le transfère aux membres d’un groupe de RaspiSMS.
J’imagine qu’il me faut un script qui fasse le lien entre gammu, RaspiSMS et la base mysql ?
Le script doit-il être inclus sous forme de commande ? (ex: X + « texte » enverra un sms avec le contenu « texte » aux membre du groupe x)
Par avance merci pour vos réponses.

OsaAjani, 2021-01-05:
Salut, pour l'instant aucun système de redirection d'un SMS n'est prévu.
La solution serait donc, à priori, de créer une commande RaspiSMS qui appellera un script lors de la réception du SMS. À partir de là il suffit que le script fasse une requête sur l'API de RaspiSMS pour envoyer un message au groupe/contact de ton choix.

shaquillou, 2021-01-07:
Bonjour,
Je vais chercher une bonne structure de script pour l'adapter à mon besoin.
Merci pour ta réponse.

closed 2021-01-10 by OsaAjani