RaspbianFrance/raspisms #150: Réception SMS : Ignoring incoming SMS info as not a Status Report in SR memory. (parse.sh)

StephanSales, 2021-02-08:
Bonjour,
Merci dans un premier temps pour ce projet, c'est vraiment top !

Cependant je rencontre un problème pour la réception des SMS. je les reçois que de façon occasionnelle quand le message n'est pas reçu, je constate dans les log le message : 

Ignoring incoming SMS info as not a Status Report in SR memory.

En modifiant le fichier parse.sh par un script proposé dans la doc de gammu cela fonctionne, je reçois le message dans /var/spool/gammu/inbox mais sinon j'ai se problème de réception. Pensez-vous que cela peu venir de parse.sh ou c'est un problème dans ma configuration ?

StephanSales, 2021-02-08:
Une autre chose quand j'exécute `bash -x /var/www/html/RaspiSMS/parseSMS.sh` il me retourne un problème de droit que j'ai corrigé, mais aussi:

seq: opérande manquant

Voilà si cela peut vous aider à m'aider 

Bonne journée à tous

OsaAjani, 2021-02-08:
Salut, l'erreur semble être du coté de Gammu, https://github.com/gammu/gammu/issues/490

StephanSales, 2021-02-08:
Merci pour votre retour, 
 je ne comprends pas dans ce cas, pourquoi le message ignoring .... disparaît et le sms arrive bien dans le fichier /var/spool/gammu/inbox quand je modifie le fichier parseSMS.sh par l'exemple dans la doc de gammu ?

RaspbianFrance, 2021-02-08:
Pourrais-tu mettre le contenu du fichier fonctionnel ? Sans ça difficile de
dire ^^

StephanSales, 2021-02-08:
Alors je ne l'ai plus sous les yeux (je tente de réinstaller), mais c'est un fichier txt nommé INdate-00-numeroEmetteur.txt et qui contient le message envoyé. Il se trouve dans le dossier cité dans mon message précédent.
Merci encore pour vos réponses.

StephanSales, 2021-02-08:
Oui pardon, l'exemple fonctionnel est celui de la doc gammu : 
```
#!/bin/sh
INBOX=/path/to/smsd/inbox
PROGRAM=/bin/cat
for ID in "$@" ; do
    $PROGRAM < $INBOX/$ID
done
```

StephanSales, 2021-02-08:
Après avoir refait quelques tests, j'obtiens :
-script initial :
2/10 reçus (le reste en Ignoring ...)
-script ex gammu :
10/10 reçus

Cependant il faut modifier les droits d'accès au fichier parse.sh et le résultat varie en fonction des réglages de gammu mais il est clair que l'on reçoit tous les messages avec le script gammu.

StephanSales, 2021-02-08:
Un dernier retour ^^
J'ai modifié le fichier parse.sh :

bin/sh date=$(date +%Y%m%d%H%M%S%N) first_time=1
PROGRAM=/bin/echo for i in `seq $SMS_MESSAGES` ; do         eval "sms_number=\"\${SMS_${i}_NUMBER}\""         eval "sms_text=\"\${SMS_${i}_TEXT}\""         if [ $first_time -eq 1 ]         then                 sms="$sms_number:"                 first_time=0         fi         sms="$sms$sms_text" done echo "$sms" >> /var/www/html/RaspiSMS/receiveds/"$date".txt

`

En rajoutant cette ligne 'PROGRAM=/bin/echo' je constate de meilleurs résultats, par contre je ne suis pas certain que cela est dû à cette modification.

OsaAjani, 2021-02-14:
Suite à la sortie de la version 3.0 de RaspiSMS le fonctionnement interne de RaspiSMS a été largement modifié, bon nombre de fix pour des bugs mineurs ont été intégrés, et plusieures fonctionnalitées ajoutées.

Ce changement de version a donc résolu, directement ou indirectement, la grande majorité des "issues" actuellement ouvertes. Il a donc été décidé de fermer l'ensemble des tickets.

Si toutefois vous pensez que votre "issue" est toujours pertinente pour la version 3.0, vous pouvez ouvrir une nouvelle "issue", en y incluant un lien vers l'ancienne.

-----------------

Forwarding the release of RaspiSMS v3.0, internal functioning of RaspiSMS have been greatly modified, lot of fix have been implemented, and many features added.

This version change thefore resolved, directly or inderectly, the vast majority of currently opened "issues". Consequently, it have been decided to close all of the tickets.

However, if you think your "issue" is still relevant for the v3.0, you may open a new "issue", including a link to the previous one.

closed 2021-02-14 by OsaAjani