RaspbianFrance/raspisms #151: De raspisms 2 à raspisms 3 avec gammu

FredMa01, 2021-02-19:
Bravo pour cette nouvelle version de raspisms.
Par contre, depuis le passage de la version 2 à 3, l'envoi et réception via Gammu ne fonctionne plus.
J'ai bien créé le téléphone Gammu comme indiqué dans la doc : https://documentation.raspisms.fr/users/adapters/gammu.html
Le fichier de configuration Gammu est celui-ci /etc/gammu-smsdrc et je n'utilise pas de code PIN.
A la création du téléphone, j'ai cette info sur l'interface :
```
Réception d'un SMS :
Non disponible.

Changement de statut d'un SMS :
Non disponible.
```
Ce qui est bien une réalité ;)
Voici le contenu de mon /etc/gammu-smsdrc (qui fonctionnait bien en version 2) 
```
; Configuration file generated by gammu-detect.
; Please check The Gammu Manual for more information.

[gammu]
GammuCoding = utf8
GammuLoc = fr_FR.UTF8
device = /dev/ttyUSB-3G
name = Phone on USB serial port HUAWEI HUAWEI_Mobile
connection = at
atgen_setCNMI=1,2,0,0,0
logfile = /var/log/gammu.log

#Ajout Fred Raspisms 3.0
log_format = textalldate

# SMSD configuration, see gammu-smsdrc(5)
[smsd]
service = files
logfile = /var/log/gammu-smsd.log

# Increase for debugging information
debuglevel = 2
#Desactivation Fred RapiSMS3.0 RunOnReceive = /var/www/html/RaspiSMS/parseSMS.sh

# Ajout Fred
RunOnFailure = /var/spool/gammu/on-error.sh
# Paths where messages are stored
inboxpath = /var/spool/gammu/inbox/
outboxpath = /var/spool/gammu/outbox/
sentsmspath = /var/spool/gammu/sent/
errorsmspath = /var/spool/gammu/error/
```
On dirait que raspisms ne va plus chercher les SMS dans les répertoires /var/spool/gammu/...

Sinon j'ai vu qu'il est nécessaire de lancer le service raspisms :
`service raspisms start`
Par contre pas possible de lancer au démarrage via : ' update-rc.d raspisms defaults'.

Il n'y a plus de parseSMS.sh et plus d'appel de console.php en cron :
`
* * * * * cd /usr/share/raspisms && /usr/share/raspisms/console.php -c parseReceivedSMS  >/dev/null 2>&1
* * * * * cd /usr/share/raspisms && /usr/share/raspisms/console.php -c sendScheduleds >/dev/null 2>&1
* * * * * cd /usr/share/raspisms && /usr/share/raspisms/console.php -c sendTransfers >/dev/null 2>&1
`

Merci de votre aide
Fred

OsaAjani, 2021-02-19:
Avec le passage à la version 3.0 RaspiSMS n'utilise plus le démon gammu, mais plutôt gammu-python, il faut donc désinstaller celui-ci, sinon il bloque gammu et il n'est plus possible pour RaspiSMS de recevoir les messages. De la même façon il n'utilise donc plus non plus parseSMS.sh

Je te conseils donc de désinstaller gammu-smsd, et idéalement d'installer RaspiSMS sur une distribution neuve.

Pour le service RaspiSMS, il s'agit d'un service systemd. Pour l'activer au démarrage il te suffit donc de faire `systemctl enable raspisms`.

closed 2021-02-19 by OsaAjani

FredMa01, 2021-02-19:
Merci OsaAjani.
Je confirme désinstallation de gammu-smsd et installation de python-gammu solutionnent le passage de raspisms 2 à 3 !

Le transfert des SMS par mail ne semble plus se faire avec la fonction mail() mais avec PHPMailer du coup ça bloque chez moi.

Il doit y avoir un paramétrage à faire. Rien ne semble accessible sur l'interface pour renseigner le SMTP.
J'ai loupé un truc ?
Merci

Ulrick17, 2021-02-22:
Bonjour 

Vous avez installé quelle version de RASPIAN?

stretch-buster-jessie

FredMa01, 2021-02-22:
Buster. Tout est ok, c'est moi qui avait fait une fausse manip lors de l'installation. En allant corriger dans env.prod.php mes paramètres SMTP, c'est corrigé.
Peut-être à préciser dans la doc l'emplacement du fichier de paramétrage.
Fred

Ulrick17, 2021-02-22:
tu arrives à envoyer des SMS depuis l'interface graphique?

Si oui le statut des SMS reste-t'il en inconnu?

Ulrick17, 2021-02-22:
j'ai installé le raspian buster et rien que la 1er commande de la doc ne passe pas

root@raspberrypi:/home/pi# apt-get update -y
Get:1 http://archive.raspberrypi.org/debian buster InRelease [32.8 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Ign:3 https://apt.raspisms.fr buster InRelease
Err:4 https://apt.raspisms.fr buster Release
  Certificate verification failed: The certificate is NOT trusted. The certificate chain uses not yet valid certificate.  Could not handshake: Error in the certificate verification. [IP: 51.77.156.39 443]
Reading package lists... Done
E: Release file for http://archive.raspberrypi.org/debian/dists/buster/InRelease is not valid yet (invalid for another 39d 7h 56min 48s). Updates for this repository will not be applied.
E: Release file for http://raspbian.raspberrypi.org/raspbian/dists/buster/InRelease is not valid yet (invalid for another 41d 21h 16min 47s). Updates for this repository will not be applied.
E: The repository 'https://apt.raspisms.fr buster Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

Pouvez-vous m'aider

Ulrick17, 2021-02-22:
Moi j'utilise un dongle GSM 

Si on ne doit plus utiliser GAMMU quel type de téléphone faut-il créer?

Ulrick17, 2021-02-22:
le sms reste dans l'etat inconnu dans l'interface.

Pourtant celui-ci est bien recu 

Merci de votre aide ...§?

OsaAjani, 2021-02-22:
Si l'envoi est fait avec gammu c'est normal, les accusés de réception ne sont pas disponibles sur gammu car trop peu fiables.

OsaAjani, 2021-02-22:
Je ferme la conversation sur cette issue pour éviter qu'elle ne dérive vers un autre sujet que celui de base étant donné que celui-ci a été résolu.