RaspbianFrance/raspisms #152: Erreur PhpMailer

FredMa01, 2021-02-19:
Les transferts de SMS par mail ne partent pas.
Avec raspisms3, ce n'est plus la fonction mail() qui semble utilisée mais PHPMailer.
A voir s'il y a un paramétrage à prévoir (SMTP, port à ouvrir...)

`Mailer.ERROR: Error sending mail : PHPMailer\PHPMailer\Exception: SMTP Error: Could not connect to SMTP host. in /usr/share/raspisms/vendor/phpmailer/phpmailer/src/PHPMailer.php:2063 Stack trace: #0 /usr/share/raspisms/vendor/phpmailer/phpmailer/src/PHPMailer.php(1887): PHPMailer\PHPMailer\PHPMailer->smtpConnect(Array) #1 /usr/share/raspisms/vendor/phpmailer/phpmailer/src/PHPMailer.php(1609): PHPMailer\PHPMailer\PHPMailer->smtpSend('Date: Fri, 19 F...', 'Vous avez re\xC3\xA7u...') #2 /usr/share/raspisms/vendor/phpmailer/phpmailer/src/PHPMailer.php(1442): PHPMailer\PHPMailer\PHPMailer->postSend() #3 /usr/share/raspisms/controllers/internals/Mailer.php(86): PHPMailer\PHPMailer\PHPMailer->send() #4 /usr/share/raspisms/daemons/Mailer.php(77): controllers\internals\Mailer->send(Array, 'Vous avez re\xC3\xA7u...', 'Vous avez re\xC3\xA7u...', NULL) #5 /usr/share/raspisms/daemons/AbstractDaemon.php(161): daemons\Mailer->run() #6 /usr/share/raspisms/daemons/Mailer.php(44): daemons\AbstractDaemon->start() #7 /usr/share/raspisms/controllers/internals/Console.php(48): daemons\Mailer->__construct() #8 [internal function]: controllers\internals\Console->mailer() #9 /usr/share/raspisms/descartes/Console.php(240): call_user_func_array(Array, Array) #10 /usr/share/raspisms/console.php(12): descartes\Console::execute_command(Array) #11 {main} [] []
`

OsaAjani, 2021-02-19:
Effectivement avec le passage à la 3.0 l'envoi des mail se fait via PHPMailer avec une connection SMTP. Ce fonctionnement a été mis en place car la fonction mail est trop limitée et notoirement complexe à configurer correctement.

La configuration SMTP se fait normalement via l'interface de dialogue lors de l'installation de RaspiSMS par les dépôts. Si la configuration SMTP n'a pas été faite à ce moment, il est possible de le faire manuellement après coup.

Pour cela, il faut modifier le fichier `env.prod.php` dans le dossier de RaspiSMS (par défaut dossier `/usr/share/raspisms`), et modifier le tableau `MAIL` pour renseigner l'utilisateur SMTP, son mot de passe, l'adresse du serveur SMTP, si la connection doit être en TLS ou pas (`true` ou `false`), le port SMTP et l'adresse mail qui sera affichée au destinataire (pour répondre, etc.).

FredMa01, 2021-02-19:
J'avais effectivement oublié cette étape lors de l'installation. 
J'ai fait une erreur que j'ai pu corriger en modifiant le `env.prod.php`.
Les mails arrivent maintenant mais pas très bavards :
```
Vous avez reçu un nouveau SMS : 
Date : Origine : Destination : Message 

```
Les logs affichent :
RaspiSMS Daemon Mailer.INFO: Try sending email : {"destinations":["frederic@xxxxxxxx"],"subject":"Vous avez re\u00e7u un SMS","body":"Vous avez re\u00e7u un nouveau SMS : \nDate : Origine : Destination : Message : ","alt_body":null} [] []

OsaAjani, 2021-02-20:
Ah, ça doit être un problème lors de la transmission des données au template, je regarde ça aujourd'hui.

OsaAjani, 2021-02-20:
C'était bien un problème côté template, j'ai fix et release une version 3.0.3 avec le correctif.

closed 2021-02-20 by OsaAjani

FredMa01, 2021-02-20:
C'est mieux mais pas encore parfait car il n'y a pas de saut de ligne :)

`Date : 2021-02-20 16:46:03Origine : +336xxxxxxxxxxDestination : VictorMessage : Bis`