RaspbianFrance/raspisms #153: Import liste de contacts : type de fichier n'est pas valide.

FredMa01, 2021-02-20:
L'import d'une liste de contacts ne passe pas. J'ai l'erreur suivante : "Le type de fichier n'est pas valide. "
Mon fichier est le suivant :
```
0,1,Nom,Prenom
Fred,+33667676767,TEST,Frederic
```
D'ailleurs lors que je crée les contacts puis exporte la liste de contacts, cela ne marche pas mieux au moment de l'import. 
Le fichier n'est pas valide alors que c'est raspisms qui l'a créé !
Fred

OsaAjani, 2021-02-20:
Merci du retour, après test il semblerait que l'export au format CSV provoque des erreures si on mélange des contacts avec des données enrichies et des contacts sans données enrichies. Pareil lors de l'import. J'ai corrigé ça et publié une mise à jour. 

Le bug pouvait provoquer le fait que des utilisateurs ne soient pas insérés, ou bien mal exportés, mais pas l'erreur "Le type de fichier n'est pas valide.".
Cette erreure n'apparait que si le fichier soumis n'est pas un csv ni un json. Tu devrais regarder les logs PHP, et vérifier le mimetype du fichier que tu essaies d'importer. Pour ça tu peux utiliser la commande `mimemtype /path/to/your/file` sous Linux.

FredMa01, 2021-02-20:
Je suis sous windows :(
J'ai testé mon fichier sous Linux : 
```
mimetype carnet.csv
carnet.csv: text/csv
```
Ce que je ne comprends pas c'est que l'export de la liste de contacts via raspisms n'est pas importable et génère l'erreur "type de fichier n'est pas valide".

OsaAjani, 2021-02-20:
Effectivement le mimetype semble ok, donc la seul chose que je vois c'est un problème sur les droits d'upload ou un truc du genre.
Il faut aller regarder du côté des logs PHP si une erreure apparait.

De mon coté impossible de reproduire le bug et après la mise à jour 3.0.4 aucun bug d'export ni d'import.

FredMa01, 2021-02-21:
Je ne vois pas d'erreur de log non plus. Je vais chercher...
Par contre, j'ai mis à jour raspisms 3.0.5 via ''apt-get upgrade'' et je n'ai pas la modif de transfert SMS dans le template.
Celui-ci est resté en l'état : 
```
Vous avez reçu un nouveau SMS :
Date : <?= $sms['at'] ?>
Origine : <?= $sms['origin'] ?>
Destination : <?= $sms['destination'] ?>
Message : <?= $sms['text'] ?>
```

OsaAjani, 2021-02-21:
Tu devrais peut-être essayer un petit chmod -R 777 /usr/share/raspisms histoire de voir si ça résoud le problème, ce qui indiquerait un problème de droits.

Pour la mise à jour ça devrait être reglé avec la 3.0.7, le pipeline de déploiement tapait sur une ancienne branche ^^.

FredMa01, 2021-02-21:
J'ai tout mis en 777 et raspisms:www-data mais toujours la même erreur :
Voir à gauche répertoire modifié et à droite répertoire initial :
[image]

Ok avec la 3.0.7 qui prend en compte les mises à jour !

OsaAjani, 2021-02-21:
Comme je n'arrive pas à reproduire je ne vois que deux possibilités. Soit le fichier est mal upload, soit le mimetype n'est plus bon après l'upload.
Il faudrait modifier le fichier `controllers/internals/Tool.php`, et dans la fonction `read_uploaded_file` ajouter une ligne `var_dump($result);var_dump(mime_content_type($tmp_filename));die();` après la ligne 268, pour voir le mimetype du fichier coté PHP. Si tu peux faire ça et me poster le résultat je pourrais peut-être trouver ce qui ne fonctionne pas.

FredMa01, 2021-02-21:
J'ai tenté mais je ne sais pas où voir le résultat...
Y a rien dans  /var/log/apache2/error.log 
Je peux t'envoyer un accès admin à raspisms en MP si tu veux tester sinon attendons de voir si je suis le seul dans ce cas ;)

Fred

FredMa01, 2021-02-21:
En fait l'import/export fonctionne très bien sur un PC Linux. C'est sous Windows que se pose ce pb.
J'aurais dû commencer par ce test...

RaspbianFrance, 2021-02-21:
Si tu fais un var_dump comme décrit tu devrais voir le résultat dans le
navigateur directement. Supprime les fichiers dans le dossier data de
RaspiSMS avant de tester.

Sous windows tu utilises quel navigateur ?

OsaAjani, 2021-02-23:
À priori windows envoie des mimetypes abberants pour les fichiers CSV, et pour PHP c'est toujours du text/plain et pas du text/csv. La version 3.0.10 utilise désormais l'extension du fichier au lieu du mimetype déclaré par le navigateur s'il n'arrive pas à reconnaitre un fichier CSV ou JSON côté serveur.

closed 2021-02-23 by OsaAjani

FredMa01, 2021-02-23:
Problème résolu nickel !
Si on voulait chipotter, on pourrait fermer automatiquement la fenêtre suivante une fois l'exportation terminée :
[image]

OsaAjani, 2021-02-23:
Effectivement, mais je ne pense pas qu'il soit possible de savoir facilement si l'exportation a été terminée. On pourrait fermer directement au clic sur le bouton, mais bon pour l'instant ça va rester tel quel, ce n'est pas une priorité.