RaspbianFrance/raspisms #154: Activation des commandes par SMS

FredMa01, 2021-02-21:
Par défaut les commandes par SMS ne sont disponibles.
Il faut à priori mettre à true dans env.ph :
`        'ENABLE_COMMAND' => true,`
Peut-être à préciser dans la doc.

Fred

OsaAjani, 2021-02-21:
Effectivement, il pourrait être intéressant d'ajouter une entrée dans la doc sur le sujet, à priori dans la page https://documentation.raspisms.fr/users/commands/overview.html.

Je regarde pour ajouter cette info à la doc.

OsaAjani, 2021-02-21:
C'est ajouté. https://documentation.raspisms.fr/users/commands/overview.html#activer-desactiver-le-support-des-commandes-par-sms

closed 2021-02-21 by OsaAjani

FredMa01, 2021-02-22:
Je me demande si cette activation des commandes SMS via env.php est suffisante.
Je tente un l'envoi d'une commande :
`{"login":"documentation@raspisms.fr","password":"p4ssw0rd","command":"lampe"}`
en mettant bien mes identifiants .
Mais, rien ne se passe.
D'ailleurs le mot de passe n'est pas caché dans l'interface raspisms, preuve à priori que la commande n'est pas interprêtée...
Rien non plus dans les logs n'évoque la réception d'une commande....
Fred

OsaAjani, 2021-02-22:
Modifier la constante dans env.php active bien la réception des SMS, mais en vérifiant je me suis rendu compte que l'analyse des commandes ne marchait plus (d'autant plus étrange que je l'avais justement testée juste avant de release la 3.0).

Après pas mal de recherche il se trouve qu'à un endroit dans le code qui analyse les commandes, un `!=` a été remplacé par un `!==` par l'outil utilisé pour harmoniser le style du code de RaspiSMS (PHPCsFixer), lequel été mal réglé et pouvais tenter des modifications dangereuses. J'ai donc corriger les réglages de l'outil et le bug, et j'ai publié le tout dans une version 3.0.8 qui devrait résoudre le problème.

FredMa01, 2021-02-23:
L'envoi de commande ne fonctionne pas mieux, désolé ;(
```
[2021-02-23T07:50:46.205295+01:00] RaspiSMS Daemon Phone 8.INFO: Receive message : {"at":"2021-02-23 07:50:41","text":"{\"login\":\"fred@domaine.fr\",\"password\":\" motdepasse\",\"command\":\"sms\"} \u0000\"","origin":"+33667xxxxxx"} [] []
[2021-02-23T07:50:46.616730+01:00] RaspiSMS Daemon Phone 8.INFO: Message received successfully. [] []
[2021-02-23T07:50:47.061931+01:00] RaspiSMS Daemon Mailer.INFO: Try sending email : {"destinations":["fred@domaine.fr"],"subject":"Vous avez re\u00e7u un SMS","body":"Vous avez re\u00e7u un nouveau SMS : \nDate : 2021-02-23 07:50:46\nOrigine : +33667xxxxxx\nDestination : Raspi\nMessage : {\"login\":\"fred@domaine.fr\",\"password\":\" motdepasse\",\"command\":\"sms\"} \u0000\"","alt_body":null} [] []
```
Sinon oui, l'issue #143 peut être fermée, le pb était résolue.

OsaAjani, 2021-02-23:
Alors pour le coup ça marche chez moi, le paramètre des commandes est-il toujours sur true, ta commande s'appelle-t-elle "sms" ? Ton fichier /usr/share/raspisms/VERSION contient bien v3.0.8 ?

FredMa01, 2021-02-23:
Oui oui je suis bien en v3.0.8.
Ma commande s'appelle sms. Elle renvoie vers sms.sh
[image]

Bizarre...

OsaAjani, 2021-02-23:
le compte que tu utilises est bien admin ? Les scripts sont bien executables par root ?

FredMa01, 2021-02-23:
Oui oui je suis bien sur un compte admin.
Les scripts sont en` raspisms:www-data` et `755`.

OsaAjani, 2021-02-24:
Ah, ça pourrait-être le `\u0000` à la fin du message qui serait responsable de la mauvaise analyse de la commande. Normalement le message est passé dans la fonction trim de php pour supprimer les caractères vides de fin de ligne, mais il est possible que différents passages entre différents encodages cassent ce retour.). Essaie de voir si tu reçois/envoi bien les accents par SMS.

Peux-tu mettre un copier/coller de ton fichier de config pour gammu ici ?

FredMa01, 2021-02-24:
Voici mon fichier de config Gammu :
```
[gammu]
#GammuLoc = fr_FR.UTF8
device = /dev/ttyUSB-3G
name = Phone on USB serial port HUAWEI HUAWEI_Mobile
connection = at
atgen_setCNMI=1,2,0,0,0
log_file = /var/log/gammu.log
log_format = textalldate
gammu_coding = utf8
```
Pour l'instant je ne peux pas tester, cf #155

FredMa01, 2021-02-25:
> Tu peux aussi essayer d'ajouter la ligne `file_put_contents(PWD_DATA . '/tst.txt', print_r($decode_message, true));` après la ligne 88 du fichier `controllers\internals\Command.php`, redémarrer les services RaspiSMS, envoyer une commande par SMS et aller voir le contenu du fichier `/usr/share/raspisms/data/tst.txt`. Ça permettra de voir si le message est bien décodé et si oui son contenu.

C'est ce que j'ai fait :

```
        public function analyze_and_process(int $id_user, string $message)
        {
            if (!ENABLE_COMMAND)
            {
                return false;
            }
            $extracted_command = [];
            $decode_message = json_decode(trim($message), true);
/* Fred */
file_put_contents(PWD_DATA . '/tst.txt', print_r($decode_message, true));
```
mais aucun fichier tst.txt créé lors de l'envoi d'une commande.
Les commandes sont considérées comme un sms quelconque.

FredMa01, 2021-03-01:
Problème résolu sans que je n'en comprenne la raison.  Le fichier tst.txt est bien créé et les mots de passe bien cachés. Désolé du bruit...

OsaAjani, 2021-03-01:
Ok, je ferme donc l'issue.