RaspbianFrance/raspisms #155: Rapisms 3.0 ne recoit pas les sms

Ulrick17, 2021-02-23:
Bonjour 

J'ai installé la version 3.0 de raspisms avec un dongle USB et Gammu.

J'ai bien compris qu'en utilisant Gammu le statut des sms restent en inconnu car trop peu fiables.

Par contre quand j'envoie un message depuis un téléphone vers le raspi j'ai bien l'accusé de réception sur le téléphone mais aucun message n'arrive au niveau du Raspi.

Est aussi parce que j'utilise gammu ou ai-je omis un paramètre dans la config.

Merci de votre aide et de vos réponses 

Ci joint mon fichier de config

[gammu]

port = /dev/ttyUSB1
model =
connection = at19200
synchronizetime = yes
logfile = /var/log/gammu.log
log_format = textalldate
use_locking =
gammuloc =
atgen_setCNMI=1,2,0,0,0

DeliveryReport = sms

DebugLevel = 255

OsaAjani, 2021-02-23:
Est-ce que les SMS partent bien depuis RaspiSMS vers le téléphone ?

Ulrick17, 2021-02-23:
Oui pour ca aucun soucis à part le statut qui reste inconnu 

Ulrick Biaga SAMA
Société Développement Informatique International (D.I.I) 
(Groupe UBIPHARM)
Espace des prés - Immeuble B
27950 SAINT MARCEL
Tél : 02 32 64 38 89
Port : 06 01 07 41 60

(#155)

Ulrick17, 2021-02-23:
oui ils le sont bien

OsaAjani, 2021-02-23:
Dans ce cas le problème se fait lors de l'échange entre RaspiSMS et gammu sur le lecture. Aucun autre logiciel n'utilise gammu en même temps (genre un démon, ou autre) ?

Il faudrait désactiver les démons RaspiSMS avec la commande `service raspisms stop`. Puis envoyer un SMS vers le Raspberry Pi, puis lancer la commande `/usr/share/raspisms/bin/gammu_get_unread_sms.py /le/chemin/vers/ton/fichier/de/config/gammu`.
Puis tout de suite après, `echo $?`.

Met moi le retour des deux dernières commandes ici.

Tu peux aussi regarder les logs dans `/var/log/raspisms/daemons.log` pour voir si une erreure se produit.

Ulrick17, 2021-02-23:
voici le output des 2 commandes 

root@raspberrypi:/var/log/raspisms# /usr/share/raspisms/bin/gammu_get_unread_sms.py /root/.gammurc
root@raspberrypi:/var/log/raspisms# echo $?
0

FredMa01, 2021-02-23:
Maintenant que Raspisms 3.0 n'utilise plus gammu-smsd, il n'est plus possible d'envoyer de sms en ligne de commande ?
On doit donc forcément passer par l'API de raspisms ?

Ulrick17, 2021-02-23:
Fred si tu dois pouvoir envoyer en ligne de commande.

Chez moi ca marche le CLI

FredMa01, 2021-02-23:
Merc Ulrick17. C'est quoi le CLI ? Hors Gammu ? Hors raspisms ?

Ulrick17, 2021-02-23:
CLI : Command Line 

Celle là marche chez moi : echo "Tapez ici votre SMS" | gammu --sendsms TEXT 06XXXXXX

Ulrick17, 2021-02-23:
OsaAjani

une idée pour la reception de SMS

FredMa01, 2021-02-23:
Bah oui CLI ! 
C'est ce que je faisais avec gammu-smsd :
`echo "Test Fred" | gammu-smsd-inject TEXT`
Mais vu que gammu-smsd est supprimé ça ne fonctionne plus.
Et 
`echo "Tapez ici votre SMS" | gammu --sendsms TEXT 06XXXXXX`
affiche une erreur : ''Erreur à l'ouverture du périphérique : périphérique inconnu, occupé ou problème de permissions.''

Fred

OsaAjani, 2021-02-23:
Alors à priori il n'y a pas de bugs dans le programme qui fait la lecture, puisqu'il trouve bien 0 comme retour. Mais il ne trouve aucun SMS non lu. J'imagine donc que le SMS est directement marqué comme "lu" par gammu avant que RaspiSMS ai eu le temps de demander les SMS non lus, ce qui n'est évidemment pas le comportement normal.

Tu peux peut-être commenter la ligne 40 et 41 du script `/usr/share/raspisms/bin/gammu_get_unread_sms.py` et ré-essayer les commandes que j'avais indiqué. Peut-être aussi ajouté après la ligne 43 `'state': m['State'],`, pour voir si le SMS est déjà lu ou pas.

Normalement tu devrais voir tous les messages reçus, ce qui permettra déjà de vérifier que le message parviens bien jusqu'à Gammu et est bien accessible via le script python. Après il restera à trouver pourquoi il arrive déjà lu...

OsaAjani, 2021-02-23:
> Maintenant que Raspisms 3.0 n'utilise plus gammu-smsd, il n'est plus possible d'envoyer de sms en ligne de commande ?
> On doit donc forcément passer par l'API de raspisms ?

Tu peux toujours envoyer par ligne de commande, mais en passant directement par gammu plutôt que gammu-smsd (comme indiqué par @Ulrick17). En revanche si tu essaies d'envoyer en même temps que RaspiSMS, il peut y avoir un blocage.

FredMa01, 2021-02-23:
```
echo "Tapez ici votre SMS" | gammu --sendsms TEXT 06XXXXXX
affiche une erreur : ''Erreur à l'ouverture du périphérique : périphérique inconnu, occupé ou problème de permissions.''
```
J'ai résolu le problème en créant un fichier de configuration .gammurc lié à l'utilisateur (gammu config). C'est normal que gammu ne prenne pas en compte le /etc/gammurc paramétré pour raspisms ?
Bon là je sors un peu de raspisms, désolé ;)

Ulrick17, 2021-02-23:
j'ai commenté les ligne s 40 et 41 mais quand je lance la commande j'ai des erreurs

root@raspberrypi:/dev# /usr/share/raspisms/bin/gammu_get_unread_sms.py /root/.gammurc
root@raspberrypi:/dev# echo $?
0
root@raspberrypi:/dev# vi /usr/share/raspisms/bin/gammu_get_unread_sms.py
root@raspberrypi:/dev# /usr/share/raspisms/bin/gammu_get_unread_sms.py /root/.gammurc
Traceback (most recent call last):
  File "/usr/share/raspisms/bin/gammu_get_unread_sms.py", line 57, in <module>
    main()
  File "/usr/share/raspisms/bin/gammu_get_unread_sms.py", line 20, in main
    state_machine.Init()
gammu.ERR_DEVICEOPENERROR: {'Text': 'Error opening device. Unknown, busy or no permissions.', 'Where': 'Init', 'Code': 2}
root@raspberrypi:/dev#

OsaAjani, 2021-02-23:
> ```
> echo "Tapez ici votre SMS" | gammu --sendsms TEXT 06XXXXXX
> affiche une erreur : ''Erreur à l'ouverture du périphérique : périphérique inconnu, occupé ou problème de permissions.''
> ```
>
> J'ai [...]

C'est normal, le fichier de configuration indiqué dans RaspiSMS est uniquement connu de RaspiSMS. C'est lui qui indique le chemin à gammu quand il lance sa commande d'envoi ou de lecture des SMS.

Dans ton cas, si tu veux utiliser le même fichier que RaspiSMS, il te suffit de passer à gammu le paramètre `--config /chemin/de/ton/fichier`.

OsaAjani, 2021-02-23:
> j'ai commenté les ligne s 40 et 41 mais quand je lance la commande j'ai des erreurs
> 
> root@raspberrypi:/dev# /usr/share/raspisms/bin/gammu_get_unread_sms.py /root/.gammurc
> root@raspberrypi:/dev# echo $?
> 0
> root@raspberrypi:/dev# vi /usr/share/raspisms/bin/gammu_get_unread_sms.py
> root@raspberrypi:/dev# /usr/share/raspisms/bin/gammu_get_unread_sms.py /root/.gammurc
> Traceback (most recent call last):
> File "/usr/share/raspisms/bin/gammu_get_unread_sms.py", line 57, in
> main()
> File "/usr/share/raspisms/bin/gammu_get_unread_sms.py", line 20, in main
> state_machine.Init()
> gammu.ERR_DEVICEOPENERROR: {'Text': 'Error opening device. Unknown, busy or no permissions.', 'Where': 'Init', 'Code': 2}
> root@raspberrypi:/dev#

À priori il y a quelque chose qui essai de lire/écrire en même temps, ce qui bloque le modem et donc fait planter, ça ne viens pas de la modif. As-tu bien arrêté raspisms avec `service raspisms stop` ?

FredMa01, 2021-02-23:
Depuis que j'ai joué avec le CLI et le gammu --sendsms, il n'est plus possible de recevoir des messages sur raspisms !! ;)
Les logs ne disent rien.
J'ai essayé de redémarrer raspisms mais ça succès....

FredMa01, 2021-02-23:
Dans les logs j'ai des erreurs de ce type, aussi bien pour des envois que des réceptions  :
```
RaspiSMS Daemon Sender.INFO: Transmit sms send signal to phone 9 on queue 2147483647.
 RaspiSMS Daemon Phone 9.INFO: Error reading received smss : Gammu command return failed. [] []
[2021-02-23T21:59:30.011452+01:00] RaspiSMS Daemon Phone 9.ERROR: Failed send message : {"id_user":"2","id_scheduled":"32","text":"Test 3 raspi","id_phone":"9","destination":"+336xxxxxxxxx","flash":"0"} with error : Cannot retrieve sms uid. [] []
```
Je crois qu'avec raspisms, il ne faut plus jouer avec gammu ;)
Où sont mis en liste d'attente les messages bloqués ? Peut-être qu'une purge peut aider....

OsaAjani, 2021-02-24:
Tu n'as pas simplement modifié le fichier de config gammu qui était renseigné pour RaspiSMS ? Ou bien créé deux téléphones qui utilisent le même fichier ?

FredMa01, 2021-02-24:
J'avais créé un autre fichier de config Gammu ici /root/.gammurc pour faire fonctionner le "gammu --sendsms ".
Et c'est depuis que raspisms ne fonctionne plus.
J'ai supprimé ce fichier de config /root/.gammur, relancé raspisms, remis/retiré le dongle et même redémarré le raspberry mais le pb demeure ;(

OsaAjani, 2021-02-24:
Tu peux essayer de stopper raspisms et de voir le retour de la commande `gammu --config /path/to/config/file sendsms TEXT '+33612345678' -text 'ton texte' -validity MAX - autolen 'le nombre de caractères dans ton texte'`.
C'est la commande qui est utilisé par RaspiSMS pour envoyer un message avec gammu. Le retour devrait te permettre de savoir pourquoi ça bug.

OsaAjani, 2021-02-24:
J'ai trouvé @Ulrick17 !!! En regardant à nouveau ton fichier gammu, je vois que tu utilises `atgen_setCNMI=1,2,0,0,0`. Si je me reporte à la doc sur les normes AT (utilisées pour communiquer avec le module), mettre le second paramètre CNMI à 2 indique d'afficher les messages reçus sur la connection série, et de les supprimer directement sans les stocker sur la SIM. C'est pour cette raison que les messages ne sont pas reçus par RaspiSMS.

Tu devrais donc supprimer cette ligne du fichier de conf gammu utilisé pour le téléphone RaspiSMS.

FredMa01, 2021-02-24:
@Ulrick17 est sauvé, pas moi ;)

Si Raspisms est arrêté :
gammu --config /etc/gammurc sendsms TEXT '+33612345678' -text 'texte1234' -validity MAX - autolen '9'
renvoie : `No response in specified timeout. Probably the phone is not connected.`

Si je démarre raspisms :
gammu --config /etc/gammurc sendsms TEXT '+33612345678' -text 'texte1234' -validity MAX - autolen '9'
renvoie : `Erreur à l'ouverture du périphérique : périphérique inconnu, occupé ou problème de permissions.`

OsaAjani, 2021-02-24:
Pour l'erreur quand RaspiSMS tourne c'est normal, RaspiSMS essaie de lire régulièrement donc ça bloque le téléphone.

Pour le retour quand RaspiSMS ne tourne pas, c'est que visiblement il n'arrive pas à trouver le téléphone. Donc le problème est clairement côté gammu. Peut être des lignes pourries dans ton fichier de config.

FredMa01, 2021-02-24:
Mon fichier de config n'a rien de particulier :
```
[gammu]
#GammuLoc = fr_FR.UTF8
device = /dev/ttyUSB-3G
name = Phone on USB serial port HUAWEI HUAWEI_Mobile
connection = at
atgen_setCNMI=1,2,0,0,0
log_file = /var/log/gammu.log
log_format = textalldate
gammu_coding = utf8
```

FredMa01, 2021-02-25:
**Les messages partent bien de raspisms mais ne sont plus reçus avec le téléphone Gammu.**
Voici une petite synthèse de mon problème...
J'ai supprimé puis réinstallé gammu et raspisms :
`apt-get purge gammu`

Le problème est resté le même !

J'ai réinitialisé gammu : `/usr/sbin/usb_modeswitch -W -v 12d1 -p 155e -R`
```
 ! PLEASE REPORT NEW CONFIGURATIONS !

DefaultVendor=  0x12d1
DefaultProduct= 0x155e

Look for default devices ...
  found USB ID 12d1:155e
   vendor ID matched
   product ID matched
  found USB ID 8564:4100
  found USB ID 0bc2:61b6
  found USB ID 0424:ec00
  found USB ID 0424:9514
  found USB ID 1d6b:0002
 Found devices in default mode (1)
Access device 038 on bus 001
Get the current device configuration ...
Current configuration number is 1
Use interface number 0
 with class 255

USB description data (for identification)
-------------------------
Manufacturer: HUAWEI
     Product: HUAWEI Mobile
  Serial No.: not provided
-------------------------
Warning: no switching method given. See documentation
Reset USB device .
 Device was reset
```
J'ai lancé un gammu identify

```
Périphérique       : /dev/ttyUSB-3G
Fabricant            : Huawei
Modèle              : E3531 (E3531)
Firmware             : 22.521.35.00.1217
IMEI                 : 861143035493085
SIM IMSI             : 208014706668400
```
J'ai modifié le fichier de config de gammu utilisé par le téléphone de raspisms en désactivant le name et le atgen :
```
[gammu]
device = /dev/ttyUSB-3G
#name = Phone on USB serial port HUAWEI HUAWEI_Mobile
connection = at
#atgen_setCNMI=1,2,0,0,0
Logfile = /var/log/gammu.log
Logformat = textalldate
gammu_coding = utf8
synchronizetime = yes

```Mais toujours bloqué. Je ne comprends pas ce qui a pu se passer à partir du moment où j'ai commencé à vouloir utiliser gammu sendsms et créer un fichier de configuration
/root/.gammurc que j'ai ensuite supprimé pensé que le pb venait de là...

Une fois raspisms arrêté, la commande :
`gammu --config /etc/gammurc sendsms TEXT '+33612345678' -text 'texte1234' -validity MAX -autolen '9'
`Fonctionne maintenant et renvoie :
```
9 / 9
Si vous désirez interrompre, appuyez sur Ctrl + C...
Envois du SMS 1 1 / 1 2.... Attente d'une réponse réseau..OK, message de référence=175 1
```
Le problème ne viendrait donc pas de gammu ?

OsaAjani, 2021-02-25:
Si les SMS sont bien envoyés mais pas reçus, je penche pour le même problème que Ulrick. Le réglage setCNMI désactive le stockage des SMS reçus, ce qui fait que RaspiSMS ne peut pas les lire puisqu'ils disparaissent directement.

Il est tout à fait possible que le fait d'avoir set le CNMI à `1,2,0,0,0` à un moment ai modifié la config interne du modem, et que celui-ci reste dans ce mode. Tu devrais peut-être essayer de set le CNMI à `1,1,0,0,0`.

Ulrick17, 2021-02-25:
Bonjour 

J'ai appliqué les modification au niveau du CNMI et j'ai fait un test du téléphone vers le raspi.

J'ai envoyé un message du téléphone vers le raspi  et je l'ai  reçu toutes les minutes et ca ne s'arrêtait pas : j'ai du arrêter le raspi pour cesser de recevoir les messages.

J'ai rallumé et depuis que j'ai fait ca  le raspi ne reçoit pas de messages du téléphone

Ulrick17, 2021-02-25:
quelque soit le CNMI que je met j'ai ca au redémarrage du daemon

 RaspiSMS Daemon Phone 2.INFO: Error reading received smss : Gammu command return failed. [] []

FredMa01, 2021-02-25:
> Il est tout à fait possible que le fait d'avoir set le CNMI à `1,2,0,0,0` à un moment ai modifié la config interne du modem, et que celui-ci reste dans ce mode. Tu devrais peut-être essayer de set le CNMI à `1,1,0,0,0` et de relancer les démons RaspiSMS.
ça n'a rien changé...
Dans le fichier de configuration de gammu : /etc/gammurc, j'ai  ajouté : 
`debuglevel = 255`
J'ai alors 
```
Thu 2021/02/25 15:38:27: SENDING frame type 0x00/length 0x05/5
Thu 2021/02/25 15:38:27: 41A|54T|45E|311|0D                                              ATE1.
Thu 2021/02/25 15:38:37: Phone does not support enabled echo, it can not work with Gammu!
Thu 2021/02/25 15:38:37: It might be caused by other program using the modem.
Thu 2021/02/25 15:38:37: See <https://wammu.eu/docs/manual/faq/general.html#echo> for help.
Thu 2021/02/25 15:38:37: Init:GSM_TryGetModel failed with error TIMEOUT[14]: No response in specified timeout. Probably the phone is not connected.
Thu 2021/02/25 15:38:37: Entering GSM_SetIncomingSMS
Thu 2021/02/25 15:38:37: Getting available SMS memories
Thu 2021/02/25 15:38:37: SENDING frame type 0x00/length 0x0A/10
Thu 2021/02/25 15:38:37: 41A|54T|2B+|43C|50P|4DM|53S|3D=|3F?|0D                          AT+CPMS=?.

```
Il faut que j'aille voir sur le github de Gammu, peut-être ici : https://github.com/gammu/gammu/issues/517

FredMa01, 2021-02-25:
Même les envois depuis raspisms sont en erreur alors qu'ils partent bien:
```
RaspiSMS Daemon Phone 11.INFO: Try send message : {"id_user":"2","id_scheduled":"42","text":"Test raspisms\n","id_phone":"11","destination":"+33612345678","flash":"0"} [] []
[2021-02-25T19:19:41.322492+01:00] RaspiSMS Daemon Phone 11.ERROR: Failed send message : {"id_user":"2","id_scheduled":"42","text":"Test raspisms\n","id_phone":"11","destination":"+33612345678","flash":"0"} with error : Cannot retrieve sms uid. [] []
```

FredMa01, 2021-02-25:
**J'ai solutionné mon problème mais je n'ai pas bien compris la solution ;)**
En fait, comme RaspiSMS n'avait plus besoin de gammu-smsd, je l'ai désinstallé.
Et c'est à partir de ce moment où j'ai eu mes problèmes de SMS qui n'arrivaient plus dans RaspiSMS.
En réinstallant gammu-smsd, et en l'arrêtant bien afin de pouvoir lancer raspisms, tout est rentré dans l'ordre !

Il ne reste que les envois depuis raspisms où il y a un message d'erreur alors que les SMS partent bien :
```
RaspiSMS Daemon Phone 11.INFO: Try send message : {"id_user":"2","id_scheduled":"42","text":"Test raspisms\n","id_phone":"11","destination":"+33612345678","flash":"0"} [] []
[2021-02-25T19:19:41.322492+01:00] RaspiSMS Daemon Phone 11.ERROR: Failed send message : {"id_user":"2","id_scheduled":"42","text":"Test raspisms\n","id_phone":"11","destination":"+33612345678","flash":"0"} with error : Cannot retrieve sms uid. [] []
```

OsaAjani, 2021-02-25:
@FredMa01 Il semblerait que RaspiSMS n'arrive pas à parser les retours de la commande gammu pour en extraire l'identifiant du message. Il faudrait voir ce que la commande d'envoi retourne quand jouée à la main (bien désactivé RaspiSMS avant). Peut-être une partie du message attendu est désactivé par les réglages CNMI. Ou alors l'utilisation des locales françaises modifient une partie du retour qui a été pensé pour les locales anglaises (j'y pense car je vois une ligne sur le sujet dans ton fichier de conf).

D'après tes retours de log je pense que tu devrais jetter un oeil à cette page : https://docs.gammu.org/faq/general.html#echo
Ça pourrait permettre de savoir si un programme utilise déjà le téléphone.

Globalement j'aurais tendance à vous conseiller à tous les deux de refaire une installation propre, avec un fichier de config ne contenant pas d'instructions CNMI ou de locale.

Sinon pour info il semble que certaines version de gammu ne fonctionnent pas avec 5 paramètres pour le CNMI, ces versions ne fonctionnent qu'avec 4 paramètres, ça pourrait venir de là.

@Ulrick17 le fait de recevoir toutes les minutes, c'est probablement parceque tu as laissé les modifs faites sur le fichier `/usr/share/raspisms/bin/gammu_get_unread_sms.py`, qui donc reçoit tous les messages (même ceux déjà lus) comme venant d'être reçus.

De façon générale je pense que le problème est vraiment lié aux modification CNMI, mais je ne sais pas comment rétablir la configuration "d'usine".

OsaAjani, 2021-05-02:
Je considère donc qu'on peut clore le sujet étant donné que @Ulrick17 ne répond plus et que le problème de @FredMa01 a été résolué avec l'issue #159 et #160.

closed 2021-05-02 by OsaAjani