RaspbianFrance/raspisms #157: Passage de variable dans la requête CURL de l'API

FredMa01, 2021-03-01:
Je ne sais pas comment faire pour intégrer le contenu de la variable $var dans le contenu du SMS via la requête CURL de l'API :
`curl -X POST https://app.raspisms.fr/api/scheduled/ -H 'X-Api-Key: ac122e8d46faac0615c181b93d07612d' -d 'text=Mon%20SMS%20d%27exemple%20:**$var**' -d 'numbers=%2B33612345678`

OsaAjani, 2021-03-01:
Si c'est une variable bash tu devrais essayer ` curl -X POST https://app.raspisms.fr/api/scheduled/ -H 'X-Api-Key: ac122e8d46faac0615c181b93d07612d' -d 'text=Mon%20SMS%20d%27exemple%20:**'"$var" -d 'numbers=%2B33612345678`

FredMa01, 2021-03-01:
Cool merci ! Il faut effectivement fermer l'apostrophe et encadrer la variable par des guillemets !
J'avais essayé plein de trucs !

OsaAjani, 2021-03-01:
Pour le coup ça relève vraiment de l'utilisation des commandes bash, je pense que ça n'as pas vraiment sa place dans la doc du projet.

closed 2021-03-01 by OsaAjani