RaspbianFrance/raspisms #158: Sélectionner tous les contacts - tous les groupes

FredMa01, 2021-03-01:
Pas essentiel mais pratique (surtout en phase de tests d'importation/exportation de listes de contacts), il n'est pas possible de sélectionner tous les contacts/groupes en vue de les modifier ou supprimer..

OsaAjani, 2021-03-01:
Effectivement, c'est une fonctionnalité qui serait bien utile, je note ça comme amélioration possible. En attendant, si tu n'as pas de trop de contacts, etc., sache que si tu cliques sur la première coche de la liste de contact, que tu maintient Shift enfoncé et que tu cliques sur la dernière, ça va sélectionner l'ensemble des contacts entre deux. Ça permet déjà de supprimer par groupe de 25 ^^.

FredMa01, 2021-03-01:
Ah oui c'est bien déjà le SHIFT  ;)

RaspbianFrance, 2021-03-19:
Pour pousser l'idée plus loin, il pourrait être intéressant de supprimer des contacts de façon conditionnelle, comme pour les groupes conditionnels.

OsaAjani, 2021-04-25:
Cette fonctionnalitée à été intégrée à la version 3.1.0, il désormais possible de choisir le nombre d'entrées à afficher dans tous les types de liste, de 25 à tous. En utilisant la technique du Clic+Shift il est donc possible très facilement de supprimer en masse n'importe quel type de donnée.

Par ailleurs il est désormais également possible de supprimer des contacts en utilisant des conditions, comme pour la création de groupes (http://127.0.0.1/raspisms-documentation/build/html/users/groups_and_contacts/conditional_delete.html).

closed 2021-04-25 by OsaAjani

FredMa01, 2021-04-25:
Nickel tout fonctionne. Très bien cette suppression conditionnelle !
Juste une erreur dans le chemin de la doc sur ton post : https://documentation.raspisms.fr/users/groups_and_contacts/conditional_delete.html