RaspbianFrance/raspisms #160: Cannot retrieve sms uid

dokime7, 2021-03-04:
Hello,

L'envoi de SMS fonctionne (le SMS est bien reçu par le destinataire) mais RaspiSMS indique échec.

Dans la log on trouve : `Cannot retrieve sms uid`

Après investigation le problème vient du fait que la commande gammu retourne :
`Envois du SMS 1 1 / 1 2.... Attente d'une réponse réseau..OK, message de référence=127 1`
`référence` contient des accents alors que la regex recherche le pattern sans.

Ça fait longtemps que je fais plus de PHP donc je ne propose pas de pull request mais il faudrait convertir la sortie de la commande pour retirer les accents avant de passer la regex.

https://github.com/RaspbianFrance/raspisms/blob/3f84622cbe2bd7e32d48e38df663413b3bb0a71a/adapters/GammuAdapter.php#L210

Je précise que la locale est configurée en FR : 
```
root@raspisms:~# locale
LANG=fr_FR.UTF-8
LANGUAGE=
LC_CTYPE="fr_FR.UTF-8"
LC_NUMERIC="fr_FR.UTF-8"
LC_TIME="fr_FR.UTF-8"
LC_COLLATE="fr_FR.UTF-8"
LC_MONETARY="fr_FR.UTF-8"
LC_MESSAGES="fr_FR.UTF-8"
LC_PAPER="fr_FR.UTF-8"
LC_NAME="fr_FR.UTF-8"
LC_ADDRESS="fr_FR.UTF-8"
LC_TELEPHONE="fr_FR.UTF-8"
LC_MEASUREMENT="fr_FR.UTF-8"
LC_IDENTIFICATION="fr_FR.UTF-8"
LC_ALL=
```

OsaAjani, 2021-03-05:
Effectivement, lors du développement je n'ai pas envisagé que gammu pouvait offrir des locales différentes.
Le problème c'est que pour une commande en italien, ou en espagnole, etc., je ne vais pas pouvoir retrouver le pattern de façon satisfaisante. Je pourrais peut-être modifier la regex pour chercher uniquement `=([0-9]+)`.
Comme de toutes façons le suivi des messages n'est pas fiable et donc pas activé pour les téléphones gammu, cet id n'est pas réellement utilisé. Je pense donc de plus en plus à simplement retourner un uuid aléatoire.

dokime7, 2021-03-05:
Oui je vois, peut être donc avoir juste un pattern à partir de "=" pour l'instant...

OsaAjani, 2021-03-07:
C'est corrigé avec la version 3.0.13. Maintenant on recherche uniquement "=" suivi de chiffres, ce qui devrait corriger le problème en cas de locales non anglaises.

closed 2021-03-07 by OsaAjani