RaspbianFrance/raspisms #163: API RaspiSMS

skitzoo, 2021-03-12:
Hi,

I use RaspiSMS and i need to use Api for send message but i don't understand how work this API. 

Thanks

OsaAjani, 2021-03-12:
Hello, all the documentation for using the HTTP API and sending a message with the API is here : https://documentation.raspisms.fr/developpers/api/overview.html#creer-un-sms-programme-post

skitzoo, 2021-03-12:
Hey, 

Before migrate to new version i use the api as follow :

```
$ch = curl_init();
curl_setopt($ch, CURLOPT_CONNECTTIMEOUT, 0);
curl_setopt($ch, CURLOPT_TIMEOUT, 1);
curl_setopt($ch, CURLOPT_URL, "http://urlraspisms/smsAPI/?email=email&password=password&numbers=number&text=text);
curl_setopt($ch, CURLOPT_HEADER, 0);
 curl_setopt($ch, CURLOPT_RETURNTRANSFER,1);
curl_exec($ch);
curl_close($ch);
```

But now i should use https://app.raspisms.fr ? with my ApiKey because i't doesn't work.

OsaAjani, 2021-03-12:
The new version use a different API so you need to update your code. You should use app.raspisms.fr only if you have an account on the cloud version of RaspiSMS. If you use a local hosted raspisms server then you should use the address of this server. 

So you should use something like : `http://your_raspisms_instance_url/api/scheduled/` and pass params `api_key`, `numbers[]` and `text` as POST params.
```
$post_params = [
     'api_key' => 'your key',
     'numbers' => [
         '+33612345678',
    ],
    'text' => 'your message',
];

$post_params = http_build_query($post_params);

$ch = curl_init();
curl_setopt($ch, CURLOPT_CONNECTTIMEOUT, 0);
curl_setopt($ch, CURLOPT_TIMEOUT, 1);
curl_setopt($ch, CURLOPT_POST, 1);
curl_setopt($ch, CURLOPT_URL, 'http://your_raspisms_instance_url/api/scheduled/');
curl_setopt($ch, CURLOPT_POSTFIELDS, $post_params);
curl_setopt($ch, CURLOPT_HEADER, 0);
curl_setopt($ch, CURLOPT_RETURNTRANSFER,1);
$result = curl_exec($ch);
curl_close($ch);
```

 so take a look at PHP curl options `CURLOPT_POST` and `CURLOPT_POSTFIELDS`.

skitzoo, 2021-03-12:
Thanks, it's work.

I close issue.

closed 2021-03-12 by skitzoo