RaspbianFrance/raspisms #165: Dongle USB recommandé pour la version V3 de RaspiSMS

SirMishaa, 2021-03-24:
Bonjour, 

Déjà merci pour votre travail qui est incroyable. C'est vraiment très cool et la nouvelle version est superbe. 

Néanmoins, je n'arrive pas à trouver la liste des dongles recommandés (du moins, ceux qui fonctionnent bien et qui sont fiables). C'est dans le but d'y connecter une carte SIM. 

Je possède actuellement celle-ci [Huawei E3531](https://www.amazon.fr/gp/product/B00L64LSWS/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1), je n'ai pas pu la tester pour l'instant. 

Si vous avez une liste, je pense qu'il serait intéressant de l'ajouter à titre indicatif dans la documentation.

Bien cordialement.

OsaAjani, 2021-03-24:
Bonjour,
le module Huawei E3531 devrait fonctionner. En revanche il faudra peut-être utiliser usb_modeswitch pour le dongle soit reconnu correctement.

Aujourd'hui le seul modem que je soit prêt à recommander le SIM800L, directement branché via des ports GPIO. Les autres modems ont trop souvent tendance à avoir des comportement étranges, coûter trop cher, ou tout simplement voir leur production/commercialisation arrêtée.

Pour l'envoi des messages RaspiSMS passe par le logiciel Gammu, lequel propose une liste de modems compatibles : https://fr.wammu.eu/phones/.

Par ailleurs cela est déjà indiqué dans la documentation sur l'utilisation d'un téléphone gammu (https://documentation.raspisms.fr/users/adapters/gammu.html#trouver-un-modem-gsm-compatible). Peut-être l'information n'est-elle pas assez facilement trouvable, si d'autres utilisateurs font remonter la même remarque je verrais pour créer une entrée dédiée dans la documentation.

closed 2021-03-24 by OsaAjani

FredMa01, 2021-03-24:
J'utilise le module Huawei E3531 avec raspiSMS. Pas de problème tout fonctionne bien.

Fred