RaspbianFrance/raspisms #166: Perte de la réception des SMS

FredMa01, 2021-04-12:
Les SMS partent bien depuis raspisms par contre aucune réception n'arrive... (alors que tout marchait il y a encore quelques semaines).
Est-ce un problème lié à la raspisms ? à python-gammu ? à mon dongle Huawei E3531 ?

Dans les logs j'ai l'impression qu'il y a des SMS bloqués

```
==> /var/log/gammu.log <==
[Gammu            - 1.40.0]
[Connection       - "at19200"]
[Connection index - 0]
[Model type       - ""]
[Device           - "/dev/ttyUSB-3G"]
[Running on       - Linux, kernel 5.10.17-v7+ (#1403 SMP Mon Feb 22 11:29:51 GMT 2021)]
Serial device: DTR is up, RTS is up, CAR is down, CTS is up
Setting speed to 19200
[Module           - "auto"]
Escaping SMS mode
Sending simple AT command to wake up some devices
1 "AT"
2 "OK"
Checking line: OK
AT reply state: 1
Enabling echo
1 "ATE1"
2 "OK"
Checking line: OK
AT reply state: 1
Trying Motorola mode switch
1 "AT+MODE=2"
2 "ERROR"
Checking line: ERROR
AT reply state: 3
Seems not to be supported
Enabling CME errors
1 "AT+CMEE=1"
2 "OK"
Checking line: OK
AT reply state: 1
1 "AT+CSCS?"
2 "+CSCS: "GSM""
3 "OK"
Checking line: OK
AT reply state: 1
1 "AT+CSCS=?"
2 "+CSCS: ("IRA","UCS2","GSM")"
3 "OK"
Checking line: OK
AT reply state: 1
Chosen GSM as normal charset
Chosen UCS2 as unicode charset
Getting model
1 "AT+CGMM"
2 "E3531"
3 "OK"
Checking line: OK
AT reply state: 1
[Model name: `E3531']
[Model data: `E3531']
[Model data: `E3531']
1 "AT+CGMI"
2 "huawei"
3 "OK"
Checking line: OK
AT reply state: 1
Manufacturer info received
[Manufacturer: Huawei]
Checking for OBEX support
1 "AT+CPROT=?"
2 "ERROR"
Checking line: ERROR
AT reply state: 3
Checking for SYNCML/OBEX support
1 "AT+SYNCML=?"
2 "ERROR"
Checking line: ERROR
AT reply state: 3
1 "AT$TSSPCSW=?"
2 "ERROR"
Checking line: ERROR
AT reply state: 3
[Module           - "A2D|iPAQ|at|M20|S25|MC35|TC35|C35i|S65|S300|5110|5130|5190|5210|6110|6130|6150|6190|6210|6250|6310|6310i|6510|7110|8210|8250|8290|8310|8390|8850|8855|8890|8910|9110|9210"]
Escaping SMS mode
Sending simple AT command to wake up some devices
1 "AT"
2 "OK"
Checking line: OK
AT reply state: 1
Enabling echo
1 "ATE1"
2 "OK"
Checking line: OK
AT reply state: 1
Trying Motorola mode switch
1 "AT+MODE=2"
2 "ERROR"
Checking line: ERROR
AT reply state: 3
Seems not to be supported
Enabling CME errors
1 "AT+CMEE=1"
2 "OK"
Checking line: OK
AT reply state: 1
1 "AT+CSCS?"
2 "+CSCS: "GSM""
3 "OK"
Checking line: OK
AT reply state: 1
1 "AT+CSCS=?"
2 "+CSCS: ("IRA","UCS2","GSM")"
3 "OK"
Checking line: OK
AT reply state: 1
Chosen GSM as normal charset
Chosen UCS2 as unicode charset
1 "AT+CGMI"
2 "huawei"
3 "OK"
Checking line: OK
AT reply state: 1
Manufacturer info received
[Manufacturer: Huawei]
Checking for OBEX support
1 "AT+CPROT=?"
2 "ERROR"
Checking line: ERROR
AT reply state: 3
Checking for SYNCML/OBEX support
1 "AT+SYNCML=?"
2 "ERROR"
Checking line: ERROR
AT reply state: 3
1 "AT$TSSPCSW=?"
2 "ERROR"
Checking line: ERROR
AT reply state: 3
Setting date & time
1 "AT+CCLK="2021/04/12,19:28:37+01""
2 "ERROR"
Checking line: ERROR
AT reply state: 3
Getting firmware versions
1 "AT+CGMR"
2 "22.521.35.00.1217"
3 "OK"
Checking line: OK
AT reply state: 1
Received firmware version: "22.521.35.00.1217"
Number version is "22,521350"
1 "AT^CURC=0"
2 "OK"
Checking line: OK
AT reply state: 1
1 "AT+CFUN=1"
2 "OK"
Checking line: OK
AT reply state: 1
1 "AT^PORTSEL=1"
2 "OK"
Checking line: OK
AT reply state: 1
[Connected]
Entering GSM_GetSMSStatus
Getting available SMS memories
1 "AT+CPMS=?"
2 "+CPMS: ("SM","ME"),("SM","ME"),("SM","ME")"
3 "OK"
Checking line: OK
AT reply state: 1
Available SMS memories received: read: ME : ok, SM : ok, SR : N/A save: ME : ok, SM : ok, SR : N/A, Motorola = no
Getting SIM SMS status
1 "AT+CPMS="SM","SM""
2 "+CPMS: 50,50,50,50,50,50"
3 "OK"
Checking line: OK
AT reply state: 1
SMS status received
Parsing +CPMS: 50,50,50,50,50,50 with +CPMS: @i, @i, @0
Parsed int 50
Parsed int 50
Used : 50
Size : 50
Getting phone SMS status
1 "AT+CPMS="ME","ME""
2 "+CPMS: 0,20,0,20,50,50"
3 "OK"
Checking line: OK
AT reply state: 1
SMS status received
Parsing +CPMS: 0,20,0,20,50,50 with +CPMS: @i, @i, @0
Parsed int 0
Parsed int 20
Used : 0
Size : 20
Leaving GSM_GetSMSStatus
Entering GSM_GetNextSMS
Starting reading!
Number = 0, Location = -1, Folder = 0
Trying SMS PDU mode
1 "AT+CMGF=0"
2 "OK"
Checking line: OK
AT reply state: 1
Getting SIM SMS status
1 "AT+CPMS="SM","SM""
2 "+CPMS: 50,50,50,50,50,50"
3 "OK"
Checking line: OK
AT reply state: 1
SMS status received
Parsing +CPMS: 50,50,50,50,50,50 with +CPMS: @i, @i, @0
Parsed int 50
Parsed int 50
Used : 50
Size : 50
Getting phone SMS status
1 "AT+CPMS="ME","ME""
2 "+CPMS: 0,20,0,20,50,50"
3 "OK"
Checking line: OK
AT reply state: 1
SMS status received
Parsing +CPMS: 0,20,0,20,50,50 with +CPMS: @i, @i, @0
Parsed int 0
Parsed int 20
Used : 0
Size : 20
Setting SMS memory type to SM

SMS center number : "+33xxxxxxxxxxxxxxxxxxx"
SMS type: Deliver
Number Length=11
Number type 91 (1 0 0 1|0 0 0 1)
International number
Len 6
Remote number : "+33xxxxxxxxxxxxxx"
SMS PID: 0x00
SMS DCS: 0x00
SMS class: -1
Decoding date & time: ven. 12 mars 2021 08:24:55  +0100
7 bit SMS, length 4
Test
Leaving GSM_GetNextSMS
Entering GSM_GetNextSMS
Number = 0, Location = 43, Folder = 0
Cache status: Found: 43, count: 50
Getting message from cache
07913386094000F0040B913347487701F400001230020290214012331848FDAECBE7A076180F92C1D03318
Number Length=7
Number type 91 (1 0 0 1|0 0 0 1)
International number
Len 6
SMS center number : "+33xxxxxxxxxxxxxxxxxxx"
SMS type: Deliver
Number Length=11
Number type 91 (1 0 0 1|0 0 0 1)
International number
Len 6
Remote number : "+33xxxxxxxxxxxxxxxxxxx"
SMS PID: 0x00
SMS DCS: 0x00
SMS class: -1
Decoding date & time: sam. 20 mars 2021 20:09:12  +0100
7 bit SMS, length 18
Test-test 1-test 1
Leaving GSM_GetNextSMS
Entering GSM_GetNextSMS
Number = 0, Location = 44, Folder = 0
Cache status: Found: 44, count: 50
Getting message from cache
07913386094000F0640ED049B7F93DCDB7C3000012305280930140A0050003140201A6F97618D47C8BD3EC320B647DD3E56590F92D3787D374501854A01740F2B2FB5DB797D90516C8FEA6CBCB2079995DB61740E453180E2FB3E7A079591E0691D373F8DB9D16B3CB2072D83D07D540EA775D3E0791C3EE39C8FEA6CBCBA0F21C1E1E9741F0B27CFE76BBCB6C50333FCDB7C32C50783C2ECFE76931BB0C9AD7E5A0FBFDEE9AE7DB
Number Length=7
Number type 91 (1 0 0 1|0 0 0 1)
International number
Len 6
SMS center number : "+33xxxxxxxxxxxxxxxxxxx"
SMS type: Deliver, UDH included, Delivery report requested
Number Length=14
Number type d0 (1 1 0 1|0 0 0 0)
Alphanumeric number, length 8
Len 8
Remote number : "+33xxxxxxxxxxxxxxxxxxx"
SMS PID: 0x00
SMS DCS: 0x00
SMS class: -1
Decoding date & time: jeu. 25 mars 2021 08:39:10  +0100
UDH header available (length 6)
7 bit SMS, length 4
Cool
Leaving GSM_GetNextSMS
Entering GSM_GetNextSMS
Number = 0, Location = 45, Folder = 0
Cache status: Found: 45, count: 50
Getting message from cache
07913386094000F0440ED049B7F93DCDB7C300001230528093014012050003140202C2EDB738CD2EBBC6EF36
Number Length=7
Number type 91 (1 0 0 1|0 0 0 1)
International number
Len 6
```
Fred

FredMa01, 2021-04-25:
En attente d'une éventuelle piste de réponse : https://github.com/gammu/gammu/issues/600

FredMa01, 2021-05-01:
Le problème a été résolu ainsi :
La commande `gammu getallsms` affichait tout une liste de SMS (une cinquantaine !).
J'ai listé les répertoires SMS via la commande `gammu getsmsfolders`
Ce qui m'a donné : 
```
1. "          Boîte de réception", Mémoire SIM, Boite de réception
2. "                Boîte d'envoi", Mémoire SIM, Boite d'envois
3. "          Boîte de réception", mémoire du téléphone, Boite de réception
4. "                Boîte d'envoi", mémoire du téléphone, Boite d'envois

J'ai alors vidé le répertoire 1 via la commande `gammu deleteallsms 1`

Les SMS arrivent bien maintenant à nous !

OsaAjani, 2021-05-02:
Ok, donc clairement on est sur une boite de réception qui se remplie.

Peut-être faudrait-il modifier le script python qui lis les SMS pour RaspiSMS afin qu'il supprime le SMS de la mémoire une fois celui-ci lu. Néanmoins ça pose la question de savoir si c'est le role de RaspiSMS de supprimer des messages de la mémoire de gammu, ou bien si ce ne serait pas plutôt à l'administrateur système de mettre en place un script qui supprime régulièrement tous les messages lus.

Je vais y réfléchir, en attendant je marque ça comme une évolution possible.

FredMa01, 2021-05-04:
Le problème n'est pas résolu car régulièrement j'obtiens ce type d'erreur :
`RaspiSMS Daemon Phone 15.INFO: Error reading received smss : Gammu command return failed. `
Je galère pour faire repartir.
J'ai l'impression  qu'à un moment le dongle est inactif.
Du coup des SMS sont mis en queue :
RaspiSMS Daemon Sender.INFO: Transmit sms send signal to phone 15 on queue 2147483647. 
RaspiSMS Daemon Phone 15.INFO: Error reading received smss : Gammu command return failed.
Et n'arrivent plus à partir ou arriver même une fois le dongle réactivé...

Après, je ne suis pas certain que cela vienne de raspisms...

FredMa01, 2021-05-19:
Depuis 15 jours, j'ai résolu mon pb d'instabilité et d'erreur :
`RaspiSMS Daemon Phone 15.INFO: Error reading received smss : Gammu command return failed.`
Je lance toutes les 2 heures une tâche cron avec le script suivant :
```
#modeswitch correspondant à ma clé E3531 
/usr/sbin/usb_modeswitch -W -v 12d1 -p 155e -R
sleep 1
lsusb
sleep 1
systemctl restart raspisms
sleep 3
```
C'est pas le top mais ça fait le travail en attendant de trouver mieux ;)

OsaAjani, 2021-06-07:
La grande question c'est est-ce que la partie restart de RaspiSMS est utile ou bien est-ce la partie usb_modeswitch qui fait le taff ? Tu pourrais essayer sans la partie restart de RaspiSMS et nous tenir au courant ?

FredMa01, 2021-06-07:
Oui oui le restart est utile sinon l'erreur `Error reading received smss : Gammu command return failed` demeure...

PacifickFR42, 2022-02-19:
Hello everyone !
I have the same problem, no more reception or sending since 02/15/22
The device is well recognized, no error in dmesg...