RaspbianFrance/raspisms #171: Blocage automatique des appels ou transfert message répondeur.

FredMa01, 2021-05-19:
Dans le cas d'un téléphone Gammu, il pourrait être pratique de pouvoir bloquer les appels ou transférer les messages vocaux répondeur.
Chez Syma, la désactivation de la messagerie vocale ou le transfert automatique des messages par mail ne semble pas possible via l'espace client. Je ne sais pas ce qu'il en est des autres opérateurs.

Il arrive que les personnes qui reçoivent un SMS aient envie d'y répondre par un appel et tombent sur la messagerie vocale. S'il faut retirer la carte SIM et la mettre dans un téléphone pour récupérer le message c'est un peu galère... ;)

OsaAjani, 2021-06-07:
Pour l'instant récupérer le message vocale etc. semble un peu complexe, mais il serait possible d'ajouter au téléphone Gammu un signalement lors de la réception d'un appel, puisque la fonctionnalité a récemment été ajoutée à RaspiSMS. Je garde l'idée sous le coude pour une future évolution.

FredMa01, 2021-06-07:
Le signalement d'un appel est en général prévu par les opérateurs via un SMS qui propose d'aller consulter la messagerie. On a donc déjà une notification.
Le souci c'est qu'il faut sortir la carte SIM du raspberry, la mettre dans un smartphone pour entendre le message vocal en question ;(