RaspbianFrance/raspisms #172: Aide pour installer raspisms & configurer gammu-smsd

papiche, 2021-06-05:
salut à tous,

Je viens de me lancer dans l'installation de raspisms sur raspberry pi pour mon association.
J'ai suivi ce tutoriel https://raspberry-pi.fr/sim800l-gsm-gps-raspberry/

1. Adapter électriquement et connecté le SIM800L (merci pour l'astuce!!) :+1: 

2. Configurer le code PIN et envoyer un premier SMS avec le tuto https://the-empire.systems/Raspberry-Pi-GSM-Notes/  :+1: 

3. Installer raspisms en passant par le dépôt  :+1: 
```
sudo apt update -y
sudo apt install -y apt-transport-https gnupg2 curl
echo "deb https://apt.raspisms.fr/ buster contrib" | sudo tee -a /etc/apt/sources.list.d/raspisms.list
curl https://apt.raspisms.fr/conf/pub.gpg.key | sudo apt-key add -
sudo apt update -y

sudo apt-get install raspisms
```

4. Accéder et configurer http://localhost/raspisms
-> Changer identifiant et mot de passe :+1: 
-> Ajouter Téléphone :-1: 
La colonne **Callbacks** indique **"Non disponible"** !!
> Réception d'un SMS : Non disponible.
> Changement de statut d'un SMS : Non disponible.
> Notification d'appel entrant : Non disponible.
> Notification de fin d'appel : Non disponible.

---
ANALYSE ET ACTIONS ?
---

A ce stade de l'installation, le fichier ```/etc/gammu-smsdrc``` contient
```[gammu]
port = /dev/ttyS0
model = at
connection = at19200
synchronizetime = yes
logformat = textall
logfile = /tmp/raspisms.log

[smsd]
service = files
logfile = syslog
pin = 1234

# Paths where messages are stored
inboxpath = /var/spool/gammu/inbox/
outboxpath = /var/spool/gammu/outbox/
sentsmspath = /var/spool/gammu/sent/
errorsmspath = /var/spool/gammu/error/

```

D'après ce que je lis ici https://docs.gammu.org/quick/index.html#configuring-the-storage et j'en conclue que la section **[smsd]** devrait contenir quelque chose du genre
```
service = sql
driver = native_mysql
host = localhost
user = ****** ?
password = ********* ?
database = ****** ?
```

J'aurai besoin de votre aide pour continuer:

- Doit-on utiliser ce qui se trouve dans ```/usr/share/raspisms/.credentials_db``` pour renseigner /etc/gammu-smsdrc ? Cela ne risque pas de créer des problèmes dans la BDD ??
- Je remarque qu'il manque à ma configuration le paramètre **RunOnReceive = ** ?
J'ai trouvé des exemples qui utilisent ```/var/www/html/RaspiSMS/parseSMS.sh``` qui n'est pas sur ma machine... Quel script faut-il y indiquer ?
- Je voudrai pouvoir déclencher des script bash à la réception d'un SMS (pour interpréter des commandes), selon https://documentation.raspisms.fr/developpers il semble y avoir plusieurs façons de faire: utiliser l'API HTTP ou les Webhooks, avez-vous des exemples ? Dois-je placer mon script dans "RunOnReceive" ?

Pour envoyer un SMS de réponse faut-il utiliser l'API HTTP ou directement gammu ?

Merci

papiche, 2021-06-06:
Oh! Je viens de lire que raspisms ne se sert plus de gammu-smsd
https://github.com/RaspbianFrance/raspisms/issues/151#issuecomment-782230784

papiche, 2021-06-06:
Bizarre le PIN est bien accepté... Mais gammu.log indique une boucle erronée...

```
Sun 2021/06/06 05:26:43: 41A|54T|2B+|43C|50P|52R|4FO|54T|3D=|3F?|0D                      AT+CPROT=?.     
Sun 2021/06/06 05:26:43: 1 "AT+CPROT=?"
Sun 2021/06/06 05:26:43: 2 "+CME ERROR: 100"
Sun 2021/06/06 05:26:43: Checking line: +CME ERROR: 100
Sun 2021/06/06 05:26:43: AT reply state: 6
Sun 2021/06/06 05:26:43: RECEIVED frame type 0x00/length 0x1E/30
Sun 2021/06/06 05:26:43: 41A|54T|2B+|43C|50P|52R|4FO|54T|3D=|3F?|0D |0D |0A |2B+|43C|4DM AT+CPROT=?...+CM
Sun 2021/06/06 05:26:43: 45E|20 |45E|52R|52R|4FO|52R|3A:|20 |311|300|300|0D |0A          E ERROR: 100..  
Sun 2021/06/06 05:26:43: CME Error 100: "unknown"
Sun 2021/06/06 05:26:43: Checking for SYNCML/OBEX support
Sun 2021/06/06 05:26:43: SENDING frame type 0x00/length 0x0C/12
Sun 2021/06/06 05:26:43: 41A|54T|2B+|53S|59Y|4EN|43C|4DM|4CL|3D=|3F?|0D                  AT+SYNCML=?.    
Sun 2021/06/06 05:26:43: 1 "AT+SYNCML=?"
Sun 2021/06/06 05:26:43: 2 "+CME ERROR: 100"
Sun 2021/06/06 05:26:43: Checking line: +CME ERROR: 100
Sun 2021/06/06 05:26:43: AT reply state: 6
Sun 2021/06/06 05:26:43: RECEIVED frame type 0x00/length 0x1F/31
Sun 2021/06/06 05:26:43: 41A|54T|2B+|53S|59Y|4EN|43C|4DM|4CL|3D=|3F?|0D |0D |0A |2B+|43C AT+SYNCML=?...+C
Sun 2021/06/06 05:26:43: 4DM|45E|20 |45E|52R|52R|4FO|52R|3A:|20 |311|300|300|0D |0A      ME ERROR: 100.. 
Sun 2021/06/06 05:26:43: CME Error 100: "unknown"
Sun 2021/06/06 05:26:43: SENDING frame type 0x00/length 0x0D/13
Sun 2021/06/06 05:26:43: 41A|54T|24$|54T|53S|53S|50P|43C|53S|57W|3D=|3F?|0D              AT$TSSPCSW=?.   
Sun 2021/06/06 05:26:43: 1 "AT$TSSPCSW=?"
Sun 2021/06/06 05:26:43: 2 "+CME ERROR: 100"
Sun 2021/06/06 05:26:43: Checking line: +CME ERROR: 100
Sun 2021/06/06 05:26:43: AT reply state: 6
Sun 2021/06/06 05:26:43: RECEIVED frame type 0x00/length 0x20/32
Sun 2021/06/06 05:26:43: 41A|54T|24$|54T|53S|53S|50P|43C|53S|57W|3D=|3F?|0D |0D |0A |2B+ AT$TSSPCSW=?...+
Sun 2021/06/06 05:26:43: 43C|4DM|45E|20 |45E|52R|52R|4FO|52R|3A:|20 |311|300|300|0D |0A  CME ERROR: 100..
Sun 2021/06/06 05:26:43: CME Error 100: "unknown"
Sun 2021/06/06 05:26:43: [Module           - "A2D|iPAQ|at|M20|S25|MC35|TC35|C35i|S65|S300|5110|5130|5190|5210|6110|6130|6150|6190|6210|6250|6310|6310i|6510|7110|8210|8250|8290|8310|8390|8850|8855|8890|8910|9110|9210"]
Sun 2021/06/06 05:26:43: Escaping SMS mode
Sun 2021/06/06 05:26:43: SENDING frame type 0x00/length 0x02/2
Sun 2021/06/06 05:26:43: 1B |0D                                                          ..              
Sun 2021/06/06 05:26:43: Sending simple AT command to wake up some devices
Sun 2021/06/06 05:26:43: SENDING frame type 0x00/length 0x03/3
Sun 2021/06/06 05:26:43: 41A|54T|0D                                                      AT.             
Sun 2021/06/06 05:26:43: 1 "AT"
Sun 2021/06/06 05:26:43: 2 "OK"
Sun 2021/06/06 05:26:43: Checking line: OK
Sun 2021/06/06 05:26:43: AT reply state: 1
Sun 2021/06/06 05:26:43: RECEIVED frame type 0x00/length 0x09/9
Sun 2021/06/06 05:26:43: 41A|54T|0D |0D |0A |4FO|4BK|0D |0A                              AT...OK..       
Sun 2021/06/06 05:26:43: Enabling echo
Sun 2021/06/06 05:26:43: SENDING frame type 0x00/length 0x05/5
Sun 2021/06/06 05:26:43: 41A|54T|45E|311|0D                                              ATE1.           
Sun 2021/06/06 05:26:43: 1 "ATE1"
Sun 2021/06/06 05:26:43: 2 "OK"
Sun 2021/06/06 05:26:43: Checking line: OK
Sun 2021/06/06 05:26:43: AT reply state: 1
Sun 2021/06/06 05:26:43: RECEIVED frame type 0x00/length 0x0B/11
Sun 2021/06/06 05:26:43: 41A|54T|45E|311|0D |0D |0A |4FO|4BK|0D |0A                      ATE1...OK..     
Sun 2021/06/06 05:26:43: Trying Motorola mode switch
Sun 2021/06/06 05:26:43: SENDING frame type 0x00/length 0x0A/10
Sun 2021/06/06 05:26:43: 41A|54T|2B+|4DM|4FO|44D|45E|3D=|322|0D                          AT+MODE=2.      
Sun 2021/06/06 05:26:43: 1 "AT+MODE=2"
Sun 2021/06/06 05:26:43: 2 "+CME ERROR: 100"
Sun 2021/06/06 05:26:43: Checking line: +CME ERROR: 100
Sun 2021/06/06 05:26:43: AT reply state: 6
Sun 2021/06/06 05:26:43: RECEIVED frame type 0x00/length 0x1D/29
Sun 2021/06/06 05:26:43: 41A|54T|2B+|4DM|4FO|44D|45E|3D=|322|0D |0D |0A |2B+|43C|4DM|45E AT+MODE=2...+CME
Sun 2021/06/06 05:26:43: 20 |45E|52R|52R|4FO|52R|3A:|20 |311|300|300|0D |0A               ERROR: 100..   
Sun 2021/06/06 05:26:43: CME Error 100: "unknown"
Sun 2021/06/06 05:26:43: Seems not to be supported
Sun 2021/06/06 05:26:43: Enabling CME errors
Sun 2021/06/06 05:26:43: SENDING frame type 0x00/length 0x0A/10
Sun 2021/06/06 05:26:43: 41A|54T|2B+|43C|4DM|45E|45E|3D=|311|0D                          AT+CMEE=1.      
[... 115 lines cut ...]
Sun 2021/06/06 05:26:43: Parsing +CPIN: READY with +CPIN: @r
Sun 2021/06/06 05:26:43: Grabbed string from reply: "READY" (parsed 5 bytes)
Sun 2021/06/06 05:26:43: Parsed raw string "READY"
Sun 2021/06/06 05:26:43: Security status received - nothing to enter
Sun 2021/06/06 05:26:44: Leaving GSM_GetSecurityStatus
Sun 2021/06/06 05:26:44: [Terminating]
Sun 2021/06/06 05:26:44: [Closing]
tail: /tmp/gammu.log : fichier tronqué
Sun 2021/06/06 05:26:44: [Gammu            - 1.40.0]
Sun 2021/06/06 05:26:44: [Connection       - "at"]
Sun 2021/06/06 05:26:44: [Connection index - 0]
Sun 2021/06/06 05:26:44: [Model type       - "SL800"]
Sun 2021/06/06 05:26:44: [Device           - "/dev/serial0"]
Sun 2021/06/06 05:26:44: [Running on       - Linux, kernel 5.10.17-v7l+ (#1421 SMP Thu May 27 14:00:13 BST 2021)]
[... cut ...]
Sun 2021/06/06 05:26:44: [Module           - "auto"]
Sun 2021/06/06 05:26:44: Escaping SMS mode
Sun 2021/06/06 05:26:44: SENDING frame type 0x00/length 0x02/2
Sun 2021/06/06 05:26:44: 1B |0D                                                          ..              
Sun 2021/06/06 05:26:45: Sending simple AT command to wake up some devices
Sun 2021/06/06 05:26:45: SENDING frame type 0x00/length 0x03/3
Sun 2021/06/06 05:26:45: 41A|54T|0D                                                      AT.           
```

papiche, 2021-06-06:
Quand **raspisms est arrêté**, envoi de SMS :+1: 

sudo gammu --config /etc/gammurc sendsms TEXT '+33_mon_numero' -text 'texte' -validity MAX -autolen '5'
5 / 5
Si vous désirez interrompre, appuyez sur Ctrl + C...
Envois du SMS 1 1 / 1 2.... Attente d'une réponse réseau..OK, message de référence=5 1

OsaAjani, 2021-06-07:
Bonjour, nous n'assurons pas de support sur RaspiSMS pour la version gratuite, donc pour plus d'aide tu trouveras peut-être plus facilement sur forum.raspberry-pi.fr.

Quelques pistes néanmoins : 
- Vérifies que le démon gammu-smsd est bien désactivé.
- Vérifies que rien d'autre n'utilises le modem GSM.
- Assure toi que le code PIN est le bon, et vérifies si le code PIN est bien nécessaire (peut-être est-il désactivé).
- Essaie d'envoyer un message avec la première commande mais le fichier de config `gammu_raspi.rc` jusqu'à avoir un truc qui fonctionne. Une fois que ça fonctionnera et que tu auras quelque chose de 100% ok niveau gammu, alors il sera temps de voir pour l'ajouter à RaspiSMS.

papiche, 2021-06-07:
- Vérifies que le démon gammu-smsd est bien désactivé.
C'est bien le cas

- Vérifies que rien d'autre n'utilises le modem GSM.
J'ai vérifié, c'est bon. A part raspisms (quand le démon est lancé)

Assure toi que le code PIN est le bon, et vérifies si le code PIN est bien nécessaire (peut-être est-il désactivé).
Le PIN est le bon.

Essaie d'envoyer un message avec la première commande mais le fichier de config gammu_raspi.rc jusqu'à avoir un truc qui fonctionne. 
Le truc fonctionne avec gammu.

Cela fonctionne aussi avec python-gammu...

J'aurai bien aimé utiliser raspisms. Mais le log gammu indique une boucle d'initialisation qui ne s'arrête jamais...
Pour continuer mes test, il faudrait que j'essaye de lancer à la main les commandes qui initialisent controlent le téléphone..
Ce serait aimable à vous de m'indiquer les lignes du code raspisms qui sont impliquées?

mpdf123, 2021-12-26:
Hello mapuche !

Je ne sais pas si tu as trouvé la solution à ton problème mais voilà une piste.

Si tu es en français, ce que je pense à lire le résultat de tes commandes... ça ne marche pas.
En effet dans la fonction "unlock_sim()" (cf. dans "raspisms/adapters/GammuAdapter.php") la dernière ligne de code test la présence du mot "nothing" dans le retour de la commande "gammu --config.....entersecuritycode PIN .....".

En Français, il faudrait chercher "Rien à faire entrer". En Anglais ça marche ça la commande  donne "Nothing to enter" pour indiquer que le code pin est connu.

En conclusion : passe en langue anglaise et ça devrait te débloquer.

FredThx, 2022-01-23:
Hello,
J'avais le même problème que papiche.
la correction du fichier de config gammu du modem en ajoutant : 
gammucoding = utf8
gammuloc = en_GB.utf8

à bien résolu le problème.

Pour info, ce n'est pas la commande "gammu ... entersecuritycode ..." qui pose problème, mais
gammu --config /home/pi/.config/gammu/gammu_sim800l getsecuritystatus
qui renvoie en français : "Rien à faire entrer."
alors qu'en anglais : "Nothing to enter." (et qui contient bien "nothing") 

merci pour ta solution.

PR #191 (2022-10-18): Make sure Gammu speaks english
Gammu returns localized messages whereas GammuAdapter searches for string "nothing". Force Gammu to return english messages so word searches will work properly. This fixes #172
commit f790adc6 (2022-10-18): Make sure Gammu speaks english

Gammu returns localized messages whereas GammuAdapter searches for string "nothing".
Force Gammu to return english messages so word searches will work properly.
This fixes #172
closed 2022-10-18 by OsaAjani