RaspbianFrance/raspisms #173: Une erreur inconnue est survenue...

FredMa01, 2021-06-27:
Lors de la mise à jour de la 3.1.13 à la 3.1.14, j'ai ce message :
"Une erreur inconnue est survenue, vous devriez peut-être ré-essayer dans quelques minutes."

Voici ce que me donnent les logs...
Il semble manquer une table SQL....`

[Sun Jun 27 15:33:12.137957 2021] [proxy_fcgi:error] [pid 17609] [client 192.168.2.1:49242] AH01071: Got error 'PHP message: descartes\\exceptions\\DescartesExceptionSqlError: SQL Error : \nSQLSTATE : 42S02\nDriver Error Code : 1146\nDriver Error Message : Table 'raspisms.quota' doesn't exist\nSQL QUERY DEBUG :\n-----------------\nSQL: [60] SELECT * FROM `quota` WHERE 1 AND `id_user` = :where_id_user\nSent SQL: [49] SELECT * FROM `quota` WHERE 1 AND `id_user` = '2'\nParams:  1\nKey: Name: [14] :where_id_user\nparamno=-1\nname=[14] ":where_id_user"\nis_param=1\nparam_type=2\n\n-----------------\n in /usr/share/raspisms/descartes/Model.php:313\nStack trace:\n#0 /usr/share/raspisms/descartes/Model.php(333): descartes\\Model->_select('quota', Array, NULL, false, NULL, NULL)\n#1 /usr/share/raspisms/models/Quota.php(25): descartes\\Model->_select_one('quota', Array)\n#2 /usr/share/raspisms/controllers/internals/Quota.php(285): models\\Quota->get_user_quota(2)\n#3 /usr/share/raspisms/controllers/publics/Dashboard.php(81): controllers\\internals\\Quota->get_user_quota(2)\n#4 [internal function]: controllers\\publics\\Dashboard->show()\n#5 /usr/sh...', referer: https://mondomaine.fr/`

RaspbianFrance, 2021-06-27:
Bonjour,
as-tu installé RaspiSMS manuellement ou via le paquet .deb ?

Il faut lancer la commande /usr/share/RaspiSMS/vendor/bin/phinx migrate
pour mettre à jour la bdd.

FredMa01, 2021-06-27:
J'ai installé raspisms via le paquet et la mise à jour via le 'apt-get upgrade'

Pour la mise à jour, j'ai cette erreur avec ta commande :```

Phinx by CakePHP - https://phinx.org.
In FileLocator.php line 71:
  The file "phinx.yml" does not exist (in: "/usr/share/raspisms/vendor/bin", "/usr/share/raspisms/vendor/bin/").
migrate [-c|--configuration CONFIGURATION] [-p|--parser PARSER] [-e|--environment ENVIRONMENT] [-t|--target TARGET] [-d|--date DATE] [-x|--dry-run] [--fake]
```

FredMa01, 2021-06-27:
Le fichier phinx.yml est dans le répertoire ''/usr/share/raspisms''

RaspbianFrance, 2021-06-27:
Il faut peut être lancer la commande depuis le répertoire de RaspiSMS

FredMa01, 2021-06-27:
Oui oui effectivement ça marche mieux ainsi.
Par contre j'ai une nouvelle erreur lors du lancement...
```
PDOException: SQLSTATE[HY000] [1045] Access denied for user 'raspisms'@'localhost' (using password: YES) in /usr/share/raspisms/vendor/robmorgan/phinx/src/Phinx/Db/Adapter/PdoAdapter.php:79
Stack trace:
#0 /usr/share/raspisms/vendor/robmorgan/phinx/src/Phinx/Db/Adapter/PdoAdapter.php(79): PDO->__construct('mysql:host=loca...', 'raspisms', 'NobwBWOpT2tmAjH...', Array)
#1 /usr/share/raspisms/vendor/robmorgan/phinx/src/Phinx/Db/Adapter/MysqlAdapter.php(116): Phinx\Db\Adapter\PdoAdapter->createPdoConnection('mysql:host=loca...', 'raspisms', 'NobwBWOpT2tmAjH...', Array)
#2 /usr/share/raspisms/vendor/robmorgan/phinx/src/Phinx/Db/Adapter/PdoAdapter.php(148): Phinx\Db\Adapter\MysqlAdapter->connect()
#3 /usr/share/raspisms/vendor/robmorgan/phinx/src/Phinx/Db/Adapter/PdoAdapter.php(212): Phinx\Db\Adapter\PdoAdapter->getConnection()
#4 /usr/share/raspisms/vendor/robmorgan/phinx/src/Phinx/Db/Adapter/PdoAdapter.php(231): Phinx\Db\Adapter\PdoAdapter->query('SELECT * FROM `...')
#5 /usr/share/raspisms/vendor/robmorgan/phinx/src/Phinx/Db/Adapter/PdoAdapter.php(376): Phinx\Db\Adapter\PdoAdapter->fetchAll('SELECT * FROM `...')
#6 /usr/share/raspisms/vendor/robmorgan/phinx/src/Phinx/Db/Adapter/PdoAdapter.php(348): Phinx\Db\Adapter\PdoAdapter->getVersionLog()
#7 /usr/share/raspisms/vendor/robmorgan/phinx/src/Phinx/Db/Adapter/AdapterWrapper.php(204): Phinx\Db\Adapter\PdoAdapter->getVersions()
#8 /usr/share/raspisms/vendor/robmorgan/phinx/src/Phinx/Migration/Manager/Environment.php(251): Phinx\Db\Adapter\AdapterWrapper->getVersions()
#9 /usr/share/raspisms/vendor/robmorgan/phinx/src/Phinx/Migration/Manager.php(303): Phinx\Migration\Manager\Environment->getVersions()
#10 /usr/share/raspisms/vendor/robmorgan/phinx/src/Phinx/Console/Command/Migrate.php(120): Phinx\Migration\Manager->migrate('prod', NULL, false)
#11 /usr/share/raspisms/vendor/symfony/console/Command/Command.php(299): Phinx\Console\Command\Migrate->execute(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#12 /usr/share/raspisms/vendor/symfony/console/Application.php(978): Symfony\Component\Console\Command\Command->run(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#13 /usr/share/raspisms/vendor/symfony/console/Application.php(295): Symfony\Component\Console\Application->doRunCommand(Object(Phinx\Console\Command\Migrate), Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#14 /usr/share/raspisms/vendor/robmorgan/phinx/src/Phinx/Console/PhinxApplication.php(69): Symfony\Component\Console\Application->doRun(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#15 /usr/share/raspisms/vendor/symfony/console/Application.php(167): Phinx\Console\PhinxApplication->doRun(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#16 /usr/share/raspisms/vendor/robmorgan/phinx/bin/phinx(28): Symfony\Component\Console\Application->run()
#17 {main}

Next InvalidArgumentException: There was a problem connecting to the database: SQLSTATE[HY000] [1045] Access denied for user 'raspisms'@'localhost' (using password: YES) in /usr/share/raspisms/vendor/robmorgan/phinx/src/Phinx/Db/Adapter/PdoAdapter.php:82
Stack trace:
#0 /usr/share/raspisms/vendor/robmorgan/phinx/src/Phinx/Db/Adapter/MysqlAdapter.php(116): Phinx\Db\Adapter\PdoAdapter->createPdoConnection('mysql:host=loca...', 'raspisms', 'NobwBWOpT2tmAjH...', Array)
#1 /usr/share/raspisms/vendor/robmorgan/phinx/src/Phinx/Db/Adapter/PdoAdapter.php(148): Phinx\Db\Adapter\MysqlAdapter->connect()
#2 /usr/share/raspisms/vendor/robmorgan/phinx/src/Phinx/Db/Adapter/PdoAdapter.php(212): Phinx\Db\Adapter\PdoAdapter->getConnection()
#3 /usr/share/raspisms/vendor/robmorgan/phinx/src/Phinx/Db/Adapter/PdoAdapter.php(231): Phinx\Db\Adapter\PdoAdapter->query('SELECT * FROM `...')
#4 /usr/share/raspisms/vendor/robmorgan/phinx/src/Phinx/Db/Adapter/PdoAdapter.php(376): Phinx\Db\Adapter\PdoAdapter->fetchAll('SELECT * FROM `...')
#5 /usr/share/raspisms/vendor/robmorgan/phinx/src/Phinx/Db/Adapter/PdoAdapter.php(348): Phinx\Db\Adapter\PdoAdapter->getVersionLog()
#6 /usr/share/raspisms/vendor/robmorgan/phinx/src/Phinx/Db/Adapter/AdapterWrapper.php(204): Phinx\Db\Adapter\PdoAdapter->getVersions()
#7 /usr/share/raspisms/vendor/robmorgan/phinx/src/Phinx/Migration/Manager/Environment.php(251): Phinx\Db\Adapter\AdapterWrapper->getVersions()
#8 /usr/share/raspisms/vendor/robmorgan/phinx/src/Phinx/Migration/Manager.php(303): Phinx\Migration\Manager\Environment->getVersions()
#9 /usr/share/raspisms/vendor/robmorgan/phinx/src/Phinx/Console/Command/Migrate.php(120): Phinx\Migration\Manager->migrate('prod', NULL, false)
#10 /usr/share/raspisms/vendor/symfony/console/Command/Command.php(299): Phinx\Console\Command\Migrate->execute(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#11 /usr/share/raspisms/vendor/symfony/console/Application.php(978): Symfony\Component\Console\Command\Command->run(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
[... cut ...]
#13 /usr/share/raspisms/vendor/robmorgan/phinx/src/Phinx/Console/PhinxApplication.php(69): Symfony\Component\Console\Application->doRun(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#14 /usr/share/raspisms/vendor/symfony/console/Application.php(167): Phinx\Console\PhinxApplication->doRun(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#15 /usr/share/raspisms/vendor/robmorgan/phinx/bin/phinx(28): Symfony\Component\Console\Application->run()
#16 {main}

```

RaspbianFrance, 2021-06-27:
À priori le fichier de conf phinx.xml ne contient pas le bon MDP pour la
base de données.

Si le reste de RaspiSMS fonctionne tu dois pouvoir retrouver le bon MDP
dans le fichier env.prod.php

FredMa01, 2021-06-27:
C'est parfait, j'ai fait la manip. Merci à toi.
C'est étrange ce changement de mot de passe...

closed 2021-06-27 by FredMa01

GBInformatics, 2024-03-13:
Hello, I keep getting the same error even though the two passwords are identical. Can someone please help me?