RaspbianFrance/raspisms #176: Php problem

Fabien0409, 2021-09-26:
Hello i've this error when launching Raspisms :
`Fatal error: Uncaught Error: Call to undefined function descartes\mb_strtoupper() in /usr/share/raspisms/descartes/load-environment.php:8 Stack trace: #0 /usr/share/raspisms/descartes/load-environment.php(35): descartes\define_array(Array) #1 /usr/share/raspisms/descartes/load-environment.php(62): descartes\load_env() #2 /usr/share/raspisms/descartes/load.php(5): require_once('/usr/share/rasp...') #3 /usr/share/raspisms/index.php(2): require_once('/usr/share/rasp...') #4 {main} thrown in /usr/share/raspisms/descartes/load-environment.php on line 8`
I think mbstring is well installed but i don't find the solution.
Thanks for your help

OsaAjani, 2021-09-26:
Hello, i think it is a problem with mbstring installation. This error never happened so far, and the file load-environment.php haven't change for years.

You should try to temporary rename the file index.php to index.php.sav and create a new index.php file with : 
```
<?php
    phpinfo();
    die();
```
Then take a look in the return page to check if mbstring is properly installed and enabled.

Fabien0409, 2021-09-27:
Ok i found the solution.
Thanks to much.
The version of php used was the wrong one (7.0). I switched to 7.4 and everything is ok.
Thanks again.

closed 2021-09-27 by Fabien0409