RaspbianFrance/raspisms #178: erreur de validitée dans la clée publique

clem-le-lu, 2021-12-06:
après cette commande:
:~$ curl https://apt.raspisms.fr/conf/pub.gpg.key | sudo apt-key add -

sur mon update je retrouve ça:

Certificate verification failed: 
The certificate is NOT trusted. 
The certificate chain uses expired certificate.  
Could not handshake: Error in the certificate verification

ne me rappelant pas avoir utilisé curl j'ai essayé avec wegt:

:~$ wget https://apt.raspisms.fr/conf/pub.gpg.key

et j'ai une erreur de certificat :

Résolution de apt.raspisms.fr (apt.raspisms.fr)… 51.77.156.39
Connexion à apt.raspisms.fr (apt.raspisms.fr)|51.77.156.39|:443… connecté.
Erreur : le certificat de « apt.raspisms.fr » n’est pas de confiance.
Erreur : le certificat de « apt.raspisms.fr » a expiré.

donc ...

j'ai fait quelque chose de mal ?

bonne soirée ⁽^-^⁾

OsaAjani, 2021-12-06:
Bonjour,
je viens d'essayer et je n'arrive pas à reproduir l'erreur. Chez moi tout fonctionne...

```
root@dd169a1f8304:/# echo "deb https://apt.raspisms.fr/ buster contrib" | tee -a /etc/apt/sources.list.d/raspisms.list
deb https://apt.raspisms.fr/ buster contrib
root@dd169a1f8304:/# curl https://apt.raspisms.fr/conf/pub.gpg.key | apt-key add -
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  3126  100  3126    0     0  14472      0 --:--:-- --:--:-- --:--:-- 14472
OK
root@dd169a1f8304:/# apt update -y
```

Vérifies peut-être que ca-certificates est installé et à jour ?

clem-le-lu, 2021-12-09:
merci de la réponse

:~$ sudo apt install ca-certificates

Les paquets suivants seront mis à jour :
  ca-certificates
1 mis à jour, 0 nouvellement installés, 0 à enlever et 407 non mis à jour.
Il est nécessaire de prendre 166 ko dans les archives.
Après cette opération, 18,4 ko d'espace disque supplémentaires seront utilisés.
Réception de :1 http://ftp.fr.debian.org/debian buster/main amd64 ca-certificates all 20200601~deb10u2 [166 kB]
166 ko réceptionnés en 0s (441 ko/s)      
Lecture des fichiers de modifications (« changelog »)... Terminé
Préconfiguration des paquets...
(Lecture de la base de données... 397876 fichiers et répertoires déjà installés.)
Préparation du dépaquetage de .../ca-certificates_20200601~deb10u2_all.deb ...
Dépaquetage de ca-certificates (20200601~deb10u2) sur (20190110) ...
Paramétrage de ca-certificates (20200601~deb10u2) ...
Updating certificates in /etc/ssl/certs...
13 added, 4 removed; done.
Traitement des actions différées (« triggers ») pour man-db (2.8.5-2) ...
Traitement des actions différées (« triggers ») pour ca-certificates (20200601~deb10u2) ...
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...

done.
done.
puis
:~$ wget https://apt.raspisms.fr/conf/pub.gpg.key

--2021-12-09 09:31:47--  https://apt.raspisms.fr/conf/pub.gpg.key
Résolution de apt.raspisms.fr (apt.raspisms.fr)… 51.77.156.39
Connexion à apt.raspisms.fr (apt.raspisms.fr)|51.77.156.39|:443… connecté.
Erreur : le certificat de « apt.raspisms.fr » n’est pas de confiance.
Erreur : le certificat de « apt.raspisms.fr » a expiré.

sudo apt update -y

Ign :8 https://apt.raspisms.fr buster InRelease                                
Réception de :9 http://linux.teamviewer.com/deb stable InRelease [11,9 kB]     
Atteint :10 http://deb.debian.org/debian buster InRelease                      
Réception de :11 http://deb.debian.org/debian buster-updates InRelease [51,9 kB]
Ign :12 http://ppa.launchpad.net/alex-p/aegisub/ubuntu hirsute InRelease       
Err :14 https://apt.raspisms.fr buster Release                                 
  Certificate verification failed: The certificate is NOT trusted. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP : 51.77.156.39 443]

donc sans surprise

:~$ sudo apt install raspisms

Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
E: Impossible de trouver le paquet raspisms

mais tu es sous buster ?
peut être que ça ne marche plus pour buster

clem-le-lu, 2021-12-09:
:~$ sudo echo "deb https://apt.raspisms.fr/ buster contrib" | tee -a /etc/apt/sources.list.d/raspisms.list
tee: /etc/apt/sources.list.d/raspisms.list: Permission non accordée
deb https://apt.raspisms.fr/ buster contrib

:~$ sudo curl https://apt.raspisms.fr/conf/pub.gpg.key | apt-key add -
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0E: This command can only be used by root.
100  3126  100  3126    0     0  15248      0 --:--:-- --:--:-- --:--:-- 15174
(23) Failed writing body

clem-le-lu, 2021-12-09:
ok je dois prendre l'habitude d'utiliser su car sudo ne marche pas toujours 

:/home/bomd# echo "deb https://apt.raspisms.fr/ buster contrib" | tee -a /etc/apt/sources.list.d/raspisms.list
deb https://apt.raspisms.fr/ buster contrib

:/home/bomd# curl https://apt.raspisms.fr/conf/pub.gpg.key | apt-key add -
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  3126  100  3126    0     0  20038      0 --:--:-- --:--:-- --:--:-- 20038
OK

puis

:/home/bomd# apt update -y

Ign :9 https://apt.raspisms.fr buster InRelease                                
Atteint :10 http://deb.debian.org/debian buster InRelease                      
Ign :11 http://ppa.launchpad.net/alex-p/aegisub/ubuntu hirsute InRelease       
Réception de :13 http://deb.debian.org/debian buster-updates InRelease [51,9 kB]
Err :15 https://apt.raspisms.fr buster Release                                 
  Certificate verification failed: The certificate is NOT trusted. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP : 51.77.156.39 443]

apt install raspisms
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
E: Impossible de trouver le paquet raspisms

T-T
bon je ne sais pas ....

FredMa01, 2021-12-09:
Je viens de tester sur une autre machine mais pour moi il n'y a pas de souci non plus :
```
sudo apt update -y
sudo apt install -y apt-transport-https gnupg2 curl
echo "deb https://apt.raspisms.fr/ buster contrib" | sudo tee -a /etc/apt/sources.list.d/raspisms.list
curl https://apt.raspisms.fr/conf/pub.gpg.key | sudo apt-key add -
sudo apt update -y
```

clem-le-lu, 2021-12-10:
après un apt update
j'ai ça à la fin 

W: La cible Sources (main/source/Sources) est spécifiée plusieurs fois dans /etc/apt/sources.list:10 et /etc/apt/sources.list:16
W: La cible Sources (contrib/source/Sources) est spécifiée plusieurs fois dans /etc/apt/sources.list:10 et /etc/apt/sources.list:16
W: La cible Sources (non-free/source/Sources) est spécifiée plusieurs fois dans /etc/apt/sources.list:10 et /etc/apt/sources.list:16
W: La cible Packages (main/binary-amd64/Packages) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible Packages (main/binary-i386/Packages) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible Packages (main/binary-all/Packages) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible Translations (main/i18n/Translation-fr_FR) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible Translations (main/i18n/Translation-fr) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible Translations (main/i18n/Translation-en) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible DEP-11 (main/dep11/Components-amd64.yml) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible DEP-11 (main/dep11/Components-all.yml) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible DEP-11-icons-small (main/dep11/icons-48x48.tar) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible DEP-11-icons (main/dep11/icons-64x64.tar) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible Packages (contrib/binary-amd64/Packages) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible Packages (contrib/binary-i386/Packages) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible Packages (contrib/binary-all/Packages) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible Translations (contrib/i18n/Translation-fr_FR) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible Translations (contrib/i18n/Translation-fr) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible Translations (contrib/i18n/Translation-en) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible DEP-11 (contrib/dep11/Components-amd64.yml) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible DEP-11 (contrib/dep11/Components-all.yml) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible DEP-11-icons-small (contrib/dep11/icons-48x48.tar) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible DEP-11-icons (contrib/dep11/icons-64x64.tar) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible Packages (non-free/binary-amd64/Packages) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible Packages (non-free/binary-i386/Packages) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible Packages (non-free/binary-all/Packages) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible Translations (non-free/i18n/Translation-fr_FR) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible Translations (non-free/i18n/Translation-fr) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible Translations (non-free/i18n/Translation-en) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible DEP-11 (non-free/dep11/Components-amd64.yml) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible DEP-11 (non-free/dep11/Components-all.yml) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible DEP-11-icons-small (non-free/dep11/icons-48x48.tar) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible DEP-11-icons (non-free/dep11/icons-64x64.tar) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible Packages (contrib/binary-amd64/Packages) est spécifiée plusieurs fois dans /etc/apt/sources.list.d/raspisms.list:1 et /etc/apt/sources.list.d/raspisms.list:2
W: La cible Packages (contrib/binary-i386/Packages) est spécifiée plusieurs fois dans /etc/apt/sources.list.d/raspisms.list:1 et /etc/apt/sources.list.d/raspisms.list:2
W: La cible Packages (contrib/binary-all/Packages) est spécifiée plusieurs fois dans /etc/apt/sources.list.d/raspisms.list:1 et /etc/apt/sources.list.d/raspisms.list:2
W: La cible Translations (contrib/i18n/Translation-fr_FR) est spécifiée plusieurs fois dans /etc/apt/sources.list.d/raspisms.list:1 et /etc/apt/sources.list.d/raspisms.list:2
[... cut ...]
W: La cible DEP-11 (non-free/dep11/Components-amd64.yml) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible DEP-11 (non-free/dep11/Components-all.yml) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible DEP-11-icons-small (non-free/dep11/icons-48x48.tar) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15
W: La cible DEP-11-icons (non-free/dep11/icons-64x64.tar) est spécifiée plusieurs fois dans /etc/apt/sources.list:9 et /etc/apt/sources.list:15

en tout cas merci

OsaAjani, 2021-12-29:
As-tu résolu le problème ?

clem-le-lu, 2021-12-29:
non pas du tout et je n'ai plus eu le temps 
il faudrait que j'essaye avec un ordi que j'ai qui est sous xubuntu mais voila pas le temps 

en tout cas merci pour le support c'est top ⁽^-^⁾

OsaAjani, 2022-04-09:
Toujours pas reproductible donc je ferme. Après relecture du fil je pense que le problème viens du fait d'utiliser sudo avant echo au lieu de l'utiliser avant tee, puis de ne pas faire les apt update dans l'ordre.

Je conseils de faire un copier coller des commandes d'install plutôt que de les taper à la main pour éviter les erreurs de recopiage.

closed 2022-04-09 by OsaAjani