RaspbianFrance/raspisms #181: SMS reçus découpés

FredMa01, 2022-01-23:
Cela fait plusieurs fois que je reçois des SMS découpés en morceaux. 
Une idée de l'origine du pb ?
Je me suis envoyé le SMS suivant (50 mots)
```
Début - Quidem quidquid ex an princeps solet autem a esset de qui acciperet posset proprium amicitiae cogitanti de illud amicitiae de amicitia ipse iis desiderata causa et utilitates recipiendisque sit cogitanti qui minus Nam nihil esset ex maxime simulatione posset quisque causa ex minus verum iis amicitia temporis in benevolentiam profecta. - Fin
```
Il arrive en 4 morceaux : 
```
[2022-01-23T20:17:03.452611+01:00] RaspiSMS Daemon Mailer.INFO: Success sending email [] []
[2022-01-23T20:18:13.828886+01:00] RaspiSMS Daemon Phone 15.INFO: Receive message : {"at":"2022-01-23 19:17:54","text":"D\u00e9but - Quidem quidquid ex an princeps solet autem a esset de qui acciperet posset proprium amicitiae cogitanti de illud amicitiae de amicitia ipse iis d","origin":"+3366xxxxxxx"} [] []
[2022-01-23T20:18:13.905481+01:00] RaspiSMS Daemon Phone 15.INFO: Message received successfully. [] []
[2022-01-23T20:18:13.905754+01:00] RaspiSMS Daemon Phone 15.INFO: Receive message : {"at":"2022-01-23 19:17:56","text":"esiderata causa et utilitates recipiendisque sit cogitanti qui minus Nam nihil esset ex maxime simulatione posset quisque causa ex minus verum iis amicit","origin":"+3366xxxxxxx"} [] []
[2022-01-23T20:18:14.015405+01:00] RaspiSMS Daemon Phone 15.INFO: Message received successfully. [] []
[2022-01-23T20:21:35.599168+01:00] RaspiSMS Daemon Phone 15.INFO: Receive message : {"at":"2022-01-23 19:16:15","text":"i quasdam cedentem exaedificavit honorem ad Eleutheropolim et Neapolim honorem principis per.","origin":"+3366xxxxxxx"} [] []
[2022-01-23T20:21:35.701204+01:00] RaspiSMS Daemon Phone 15.INFO: Message received successfully. [] []
[2022-01-23T20:21:36.202555+01:00] RaspiSMS Daemon Phone 15.CRITICAL: Error reading MSG SEND Queue, error code :  [] []
[2022-01-23T20:23:12.298881+01:00] RaspiSMS Daemon Phone 15.INFO: Receive message : {"at":"2022-01-23 19:17:56","text":"ia temporis in benevolentiam profecta. - Fin","origin":"+3366xxxxxxx"} [] []

```
Un autre SMS de 30 mots arrive en 2 morceaux.
Un SMS de 10 ou 15 mots arrive en 1 morceau.

Bravo pour cet outil très pratique en cette période de crise sanitaire où l'envoi de SMS est bien utile.

FredMa01, 2022-03-25:
Est-ce que d'autres utilisateurs rencontrent ce type de découpage systématique des SMS ou il faut que je regarde sur mon raspi ce qui pose souci ?

FredMa01, 2022-04-09:
Un utilisateur pourrait-il tenter d'envoyer un SMS vers raspisms d'une centaine de mots du type :`
Globis truci eos repellere inter dispositos perterrens municipia ardor perterrens latius aliquotiens serpentes confertos moliens isdem latius hic quae plurima pro ardor ardor ardor plana globis repellere montium ardor nunc et quae aliquotiens confertos et obvios quae nata castella multitudine conterminant ardor lacessens inter eminus nunc viribus moliens quisque plana et aliquotiens persultat plana superabatur quisque globis et confertos pro castella et obvios et quisque persultat mollia milites et nata et isdem per dispositos eos eos persultat sed sed aliquotiens truci aliquotiens municipia isdem confertos lacessens ingenti obvios ut truci educata dispersos dispersos ululatu plurima dispersos et quae quae serpentes.`
Chez moi ce type de SMS ne pose aucun souci à l'envoi mais se retrouve découpé en 9 SMS lors de la réception (SMS reçus).
Je suis sur un raspberry 4

OsaAjani, 2022-04-09:
Bonjour,
le problème viens de la façon dont gammu gère les messages long. Il est peut-être possible de modifier les réglages gammu pour qu'il concatène automatiquement les messages longs mais je n'ai pas trouvé en jettant un rapide coup d'oeil à la doc.

Pourtant dans mon souvenir ça ne posait pas de problèmes d'habitude, mais peut-être que le daemon SMSD gerait le soucis à notre place.

Sinon il serait probablement possible de gérer le cas en modifiant le code de `bin/gammu_get_unread_sms.py`. [Cet échange](https://gammu-users.narkive.com/Y5EvVo5Y/could-you-tell-me-how-gammu-handles-long-sms-the-function-that-prints-it) semble indiquer un fichier d'exemple qui permet de décoder des SMS pour reformer un SMS long avec la fonction `linkSMS`.

Pour l'instant ce soucis n'est pas une priorité pour moi car même si les choses peuvent êtres améliorées elles restent fonctionnelles en l'état, et d'autres tâches plus importantes pour RaspiSMS me prennent déjà beaucoup de temps.

Si quelqu'un se sent de faire une modification du fichier pour gérer correctement les SMS longs lors de la lecture, j'accueillerai volontier une pull request en ce sens.

Vikingfr, 2022-11-08:
Bonjour,
juste pour info, un SMS est limité à 160 caractères. Il est normal qu'une application découpe un SMS plus long en bloc de 153 caractères (les 7 manquants servent a indiquer le fractionnement et l'ordre)

FredMa01, 2022-11-08:
Bonjour,
Sauf que dans les cas que j'évoque, les messages sont bien en dessous des 160 caractères....

Vikingfr, 2022-11-08:
non, le message "Début - Quidem quidquid ex an princeps solet autem a esset de qui acciperet posset proprium amicitiae cogitanti de illud amicitiae de amicitia ipse iis desiderata causa et utilitates recipiendisque sit cogitanti qui minus Nam nihil esset ex maxime simulatione posset quisque causa ex minus verum iis amicitia temporis in benevolentiam profecta. - Fin" fait 350 caractères. Il a donc été envoyé en SMS de 153 caractères. Par contre, il n'aurait pas du y avoir 4 sms, mais seulement 3.

PR #195 (2023-05-30): Improve Gammu SMS Receiver
This as a rewrite of `gammu_get_unread_sms.py` script that adds:
- Support for long SMS
- Added proper CLI interface (see --help)
- Added optional --delete parameter which deletes SMS after printing them as JSON
- Added optional --show-read parameter which shows all not Unread marked SMS
- Added logging and --debug option
- Retain retrocompatibility with earlier versions of this script
- Retain retrocompatibility with Python 2.7+ (hopefully)

Fixes #181.
Btw, the interface with RaspiSMS could be improved with a temporary JSON file, using stdout seems lossy.
commit 2309a0e0 (2023-05-30): Improve Gammu SMS Receiver

This as a rewrite of `gammu_get_unread_sms.py` script that adds:
- Support for long SMS
- Added proper CLI interface (see --help)
- Added optional --delete parameter which deletes SMS after printing them as JSON
- Added optional --show-read parameter which shows all not Unread marked SMS
- Added logging and --debug option
- Retain retrocompatibility with earlier versions of this script
- Retain retrocompatibility with Python 2.7+ (hopefully)

Fixes #181.
Btw, the interface with RaspiSMS could be improved with a temporary JSON file, using stdout seems lossy.
closed 2023-05-30 by OsaAjani