RaspbianFrance/raspisms #183: API refuse parfois une requête php

conichonhaa, 2022-02-03:
Suivant la syntaxe de l'API `curl -X POST http://local_IP/raspisms/api/scheduled/ -H 'X-Api-Key: XXXXXXXXXXXXXXXXXXXX' -d 'text=My%20Message' -d 'numbers=06XXXXXXXX'` j'exécute bien mon code php de la sorte

```
$ch = curl_init();

curl_setopt($ch, CURLOPT_URL, 'http://local_IP/raspisms/api/scheduled/');
curl_setopt($ch, CURLOPT_RETURNTRANSFER, 1);
curl_setopt($ch, CURLOPT_POST, 1);
curl_setopt($ch, CURLOPT_POSTFIELDS, "text=My%Message&numbers=06XXXXXXXX");

$headers = array();
$headers[] = 'X-Api-Key: XXXXXXXXXXXXXXXX';
$headers[] = 'Content-Type: application/x-www-form-urlencoded';
curl_setopt($ch, CURLOPT_HTTPHEADER, $headers);

$result = curl_exec($ch);
if (curl_errno($ch)) {
    echo 'Error:' . curl_error($ch);
}
curl_close($ch);
```

Aucune erreur tout est bon, par contre en essayant d'y inclure 2 variables $tel et $message tel que

`$data = array("text"=>$message,
"numbers"=>$tel
);`

`curl_setopt($ch, CURLOPT_POSTFIELDS, http_build_query($data));`

Je ne retrouve aucune erreur de refus de la part de raspisms, mais mon sms ne part pas et n’apparaît pas dans l'interface web

conichonhaa, 2022-02-04:
Je me réponds à moi même c'est la différence entre le formatage local (06XXXXXXXX) et international (+33) qui met le bazar. L'api accepte %2b33 mais pas 06XXXXXXXX. La v2 ne s'en souciait pas dans mes souvenirs

OsaAjani, 2022-02-04:
Depuis la version 3 RaspiSMS n'accepte que les numéros au format international, ceci afin de garantir la cohérence interne.

closed 2022-04-09 by OsaAjani