RaspbianFrance/raspisms #184: Dépôt apt.raspisms.fr pour bullseye - absence de release

FredMa01, 2022-02-03:
Lors de la mise à jour de Debian vers bullseye, j'ai modifié le fichier **/etc/apt/sources.list.d/raspisms.list**
`deb https://apt.raspisms.fr/ buster contrib`
par
` deb https://apt.raspisms.fr/ bullseye contrib"`

ce qui génère l'erreur :
`E: Le dépôt https://apt.raspisms.fr bullseye Release n'a pas de fichier Release.`

J'ai remis `buster` en attendant ;)

conichonhaa, 2022-02-04:
Installation impossible sur une version toute fraîchement installée de debian 11. Retour à buster également et plus de problème

FredMa01, 2022-02-04:
En laissant buster dans les sources, raspisms fonctionne bien sous Bullseye.

OsaAjani, 2022-02-04:
Il va falloir que j'ajoute un dépôt pour Bullseye, j'essaie de m'occuper de ça dans la semaine.

FredMa01, 2022-10-17:
Un petit up ;)
` Le dépôt https://apt.raspisms.fr bullseye Release n'a pas de fichier Release`

GBInformatics, 2024-03-28:
Bonjour est il possible de vous poser des questions ?