RaspbianFrance/raspisms #185: Campagne de SMS, plantage pendant l'envois

Ast0e, 2022-02-17:
Bonjour,
Nous nous servons de la solution RaspiSMS afin de faire des campagnes de SMS sur un grand nombres de numéros.
Nous envoyons à tous les contacts sans distinction en utilisant un Groupe conditionnel "1".

[image]

Nous constatons que l’envoie se passe bien sur les 100 premiers contacts ( 1 SMS toutes les 5 secondes ) mais ensuite ils passent en erreurs avant que la plateforme ne cesse d'envoyés.

Je suis obliger de découper l'annuaire en plusieurs "Tranches de 100" afin d'échelonner l’envoie.

Est-ce que c'est un problème connu ? Est-ce qu'il a moyen de faire un échelonnage des SMS autrement (Car actuellement je vide l'annuaire et réimporte 100 par 100 ) ?

Nous utilisons ce petit module en USB :

[image]

Cordialement

Ast0e, 2022-02-18:
Bonjour,
Je te remercie pour le retour, je vais voir avec l'opérateur même si normalement nous avons prit un forfait adapté.
Pour la partie modem USB il va falloir creuser je suppose alors.

Dans RaspiSMS il n'y pas de moyen de limiter/temporiser  l’envoie ducoup ?

FredMa01, 2022-02-18:
Cela ressemble à ce que je suggérais également ici https://github.com/RaspbianFrance/raspisms/issues/182

Ast0e, 2022-04-25:
Finalement nous avons contourné avec la dernière mise à jours qui permet de créer des CSV d'envois.
Nous envoyons par salve de 150 toute les heures.
Je suppose toute de même que l'idée de OsaAjani était bonne avec une restriction coté opérateur.

closed 2022-04-25 by Ast0e