RaspbianFrance/raspisms #187: Erreur 500 API scheduled

emnbdx, 2022-03-30:
Depuis la mise à jour dans la dernière version il est impossible de programmer un envoie via l'API

` curl -X POST https://xxxxxxxx/raspisms/api/scheduled -H 'X-Api-Key: xxxxxxx' -d 'text=Mon%20SMS%20d%27exemple' -d 'numbers=%2B336xxxxxxxx'`

En retour j'ai une erreur 500 avec un page html contenant ce message : Une erreur inconnue est survenue, vous devriez peut-être ré-essayer dans quelques minutes.

OsaAjani, 2022-03-30:
Je ne parviens pas à reproduire l'erreur. Il s'agit d'une installation auto-hébergée de RaspiSMS ? Si oui comment l'installation et la mise à jour ont-elles été faites ?

emnbdx, 2022-03-30:
C'est effectivement auto-hébergé sur un raspberry pi.
Pour la mise à jour j'ai simplement fait un apt-get update / upgrade

emnbdx, 2022-03-30:
En fait il s'agissait d'un souci lors de la mise à jour.
J'ai l'impression qu'il n'avait pas effectué les migrations de database :/

Je peux à nouveau ajouter des SMS.

closed 2022-03-30 by emnbdx