RaspbianFrance/raspisms #190: Questions API & problème de relais (si message trop vieux ?)

flapili, 2022-09-21:
Bonjour,

J'utilise la version SaaS avec un relais (sous entendre l'application RaspiSMS Relay).

Si le relais n'est pas disponible alors le message passe en status inconnu et le relais n'essaye pas de le renvoyer le message (si le message est trop vieux ?)

questions :
 - au bout de combien de temps le message est considéré comme "à ne pas envoyer", si je coupe le relai les messages en status inconnus ne s'envoient que s'il sont recents, au bout de plusieurs jours impossible de les faire renvoyer.

 - Existe t'il une documentation au format openAPI de la doc ? il semblerais que certains endpoints ne soit pas documenté (https://app.raspisms.fr/discussion/getmessage/{numéro}/{timestamp} par exemple)

- existe t'il un moyen de monitorer les messages non envoyés ? le plus simple semble d'être avec par exemple https://app.raspisms.fr/sended/json/, cet endpoint semble avoir des filtres, par contre est ce une bonne idée dans le sens où il n'est pas prévu pour l'api mais pour l'interface, et donc peut changer à tout moment en cas de MAJ de l'interface.

Cordialement

OsaAjani, 2022-09-21:
Bonjour,
concernant les messages ils sont toujours envoyés coté RaspiSMS. En revanche je ne trouve pas d'information coté Google sur le délais après lequel les messages firebase (qui sont utilisés pour la transmission entre RaspiSMS Server -> RaspiSMS Relay) sont considérés comme non livrés et sont donc perdus.

L'appli mobile peut également ignorer un message si il a déjà essayé de l'envoyer plusieurs fois mais que cela a échoué à chaque fois. Par défaut l'application est prévue pour ne faire l'envoi de message que si elle dispose d'un réseau mobile, et réessayer plusieurs fois avec un temps d'attente d'une minute entre chaque essaie.

Par ailleurs un message en status inconnu est, par définition en status inconnu ! Cela ne signifie donc pas forcément qu'il n'a pas été envoyé. Le protocole SMS est fondamentalement asynchrone, ce qui le rend très compliqué à suivre, ainsi un message en status inconnu peut avoir été : 

- Envoyé et non reçu par le destinataire.
- Envoyé, reçu par le destinataire, mais que le destinataire bloque les avis de réception.
- Avoir été envoyé, reçu par le destinataire, généré un retour indiquant sa réception, mais que le retour soit arrivé à un moment ou l'appli ne tournait pas et n'as donc pas pu enregistrer la réception.

Globalement, si le statut est failed, le message a 100% pas été reçu. Si le statut est delivered, le message a 100% été reçu. Si le statut est unknown, on ne peut pas savoir.

Je sais que c'est une situation très agaçante, surtout pour nous qui sommes habitués au web, à TCP/IP et à toujours savoir si les choses sont arrivées ou non, mais c'est hélas la nature même du protocole SMS :/. Si ça peut aider à mieux comprendre les choses pour ceux qui sont plus habitués au réseau classique, il faut d'avantage voir le SMS comme de l'UDP que comme du TCP, les paquets peuvent êtres perdus, peuvent ne pas arriver dans l'ordre, et on n'en saura jamais rien.

Concernant l'API maintenant. Il n'y a pas de doc openAPI, il y a uniquement la documentation officielle ici https://documentation.raspisms.fr/developpers/api/overview.html. Si ce n'est pas dans la doc officielle, cela signifie que ça peut changer à tout moment.

Il peut effectivement y avoir des "API" internes non documentées, mais il ne s'agit alors pas d'API à proprement parler, uniquement de moyens commodes d'exposer des données JSON à la partie client. Typiquement, si ça n'est pas sous l'URL `/api/*`, il vaut mieux éviter d'y toucher, ça peut bouger n'importe quand.

Pour trouver une liste des messages non envoyés la seule solution actuellement qui soit fiable et qui ne risque pas de bouger c'est de passer par /api/list/sended/.

Il est possible que dans l'avenir une solution permettant d'ajouter des paramètres à la requête pour filtrer les résultats soit ajoutée, mais ce n'est pas prévu à court terme, sauf à recevoir une pull request dans ce sens ou à faire une prestation dev dev payante.

flapili, 2022-09-22:
Merci pour votre retour si rapide ! malheureusement je ne fais pas du tout de PHP je ne pourrai donc pas proposer de PR 😅 

Finalement j'ai trouvé une solution simple, c'est d'utiliser les webhooks plutôt que de faire du polling sur l'api

Cordialement

closed 2022-09-22 by flapili