RaspbianFrance/raspisms #193: Usage de gammu dans RaspiSMS lorsqu'on débranche/rebranche un modem

deajan, 2022-10-18:
Bonjour,

J'ai installé un RaspiSMS en tant que VM Qemu, et j'utilise gammu via un modem 4G sur adaptateur USB que je donne à ma VM en passthrough.
Jusqu'à la, tout va bien, excepté lorsque je débranche/rebranche l'adaptateur USB, ma VM perd l'accès au modem (logique !).
J'ai fini par écrire un petit script qui utilise udev pour vérifier si un appareil USB a été branché, et s'il s'agit du modem, je le remets en passthrough pour ma VM.

Le seul soucis, c'est que RaspiSMS n'a plus envie de discuter avec mon modem s'il a été débranché/rebranché.
Le seul moyen d'arriver à refaire marcher RaspiSMS après un débranchement / rebranchement sauvage est de relancer le service "raspisms" via un `systemctl restart raspisms`.

Jusqu'à la, j'utilise mon script également à l'intérieur de la VM, pour détecter lorsque mon modem est branché, et redémarrer le service raspisms à ce moment là.

Du coup je me demandais s'il n'y avait pas un moyen plus propre, comme redémarrer seulement le sous-système gammu de RaspiSMS, via une commande php quelconque ?

PS: Au besoin, je veux bien partager mon script, qui peut éviter pas mal de maux de têtes pour ceux qui bricolent avec les modems USB ;)

OsaAjani, 2022-10-19:
Pour l'instant je ne comprends pas pourquoi RaspiSMS ne parviens pas à faire les envois si le modem a été débranché/rebranché. Si le ttyUSB a changé là c'est normal, mais si il reste le même je ne vois vraiment pas raisons.

Serait-il possible d'avoir les logs /var/log/raspisms/daemons.log après le débranchement/rebranchement du modem lors de l'envoi d'un SMS.

deajan, 2023-02-14:
J'ai écrit le script suivant qui détecte les branchements/ rebranchements du modem: https://github.com/NetInvent/udev_monitor
Ce dernier lance un script qui réinitialise le modem au besoin : https://github.com/NetInvent/usb_resetter