RaspbianFrance/raspisms #200: SIM Pleine

Vikingfr, 2022-11-06:
Bonjour,

je rencontre un nouveau soucis avec RASPISMS. J'utilise des cartes SIM de Free à 2€. Elles peuvent stocker 50 ou 100 SMS dans leur mémoire selon les versions. Et lorsque raspiSMS reçoit un SMS, il semble qu'il ne libère pas la mémoire. En effet, après quelques jours, je ne peux plus recevoir de SMS. J'ai dans les logs

`RaspiSMS Daemon Phone 1.INFO: Error reading received smss : Gammu command return failed. [] []`

Et si j'interroge la SIM via des commandes AT, j'obtient:
```
AT+CPMS?
+CPMS: "SM",50,50,"ME",0,50,"SM_P",50,50
```
Si je vide manuellement les emplacements mémoire via "gammu deleteallsms 1" (je n'ai pas encore compris comment supprimer sélectivement un sms) , je revient à 
```
AT+CPMS?
+CPMS: "SM",0,50,"ME",0,50,"SM_P",0,50
```
et la réception refonctionne
Du coup, je me demande si après la lecture des SMS, il ne manque pas une option pour supprimer le SMS lu de la mémoire de la SIM.
Je ne vois rien dans gammu_get_unread_sms.py qui efface les SMS en état 'Read'

deajan, 2022-11-07:
Verifiez ma dernière pull request pour `gammu_get_unread_sms.py` qui comporte une option d'effacement.
Il faut cependant avoir gammu >= 1.4.2 pour que cela fonctionne.

Vikingfr, 2022-11-08:
Merci pour ce retour.
La version 1.4.2 ne semble pas encore dispo sur raspbian, je vais surveiller et utiliser votre version du script (en attendant, je fait un gros deleteall bourin via crontab) .

deajan, 2022-11-08:
Il y a le risque que le deteteall bourin est de supprimer les messages juste avant lecture (bien que la fenetre de tir soit limitée).
Il serait plus intéressant d'ajouter le deleteall directement dans GammuAdapter.php après une lecture avec succès.

Vikingfr, 2022-11-08:
pour le moment, le delete bourin ne me pose pas de soucis, je le fais vers 1h du matin et il n'y a pas de réception à cette heure dans mon usage.
Mais je suis en mode dev et je veux faire du vote par SMS à terme, il faudra que je gère réellement le delete pour une mise en prod. 
Je vais voir si je peux compiler gammu 1.42 pour RPi3 parce que je pense que l'intégration dans les dépôts ne se fera jamais (vu que la version est sortie depuis 2020 si j'ai bien suivi)

deajan, 2022-11-08:
gammu 1.42 est dispo sur Debian 11, mais il va falloir installer raspisms manuellement.
Sinon c'est assez simple d'améliorer le script python pour appeler un delete une fois que le transfert à RaspiSMS a été un succès.

Vikingfr, 2022-11-08:
Le dev raspisms est plus actif que Gammu. Je vais garder Raspisms géré via apt, et je vais compiler Gammu, ça me semble plus logique.
Merci de vos conseils et de votre aide.

FredMa01, 2022-11-08:
J'utilise Raspisms sur pi4 via  Raspberry Pi OS Bullseye :
deb http://raspbian.raspberrypi.org/raspbian/ bullseye main contrib non-free rpi
J'ai laissé buster dans les sources de raspisms : ```
 /etc/apt/sources.list.d/raspisms.list
deb https://apt.raspisms.fr/ buster contrib
```
Par contre gammu est toujours en 1.41.
@Vikingfr , je veux bien profiter ton expérience de compilation de gammu 1.42....

Vikingfr, 2022-11-09:
@FredMa01 , je vais répondre de mémoire, j'espère ne pas oublier d'étape, mais elles sont simples.
J'ai un RPi3B sous Raspberry OS (bullseye)
J'ai raspisms moi aussi installé via le dépot 'buster' et cela génère un premier soucis, c'est qu'il y a une dépendance vers gammu. Elle n'est pourtant pas obligatoire, si on n'utilise que des services type ovh. Mais tans pis, il faut faire avec. A cause de la dépendance, il n'est pas possible de retirer le paquet Gammu via APT, sinon Raspisms part avec.

J'ai fait le choix de garder raspisms géré via APT et de recompiller Gammu (car projet peu actif, je ne vais pas y revenir souvent).
J'ai donc commencé par renommer /usr/bin/gammu en /usr/bin/gammu.141 (via cmde 'mv'), à défaut de pouvoir supprimer le paquet via apt. 

Ensuite, j'ai téléchargé les sources de Gammu et Python-Gammu:
```
wget https://dl.cihar.com/gammu/releases/gammu-1.42.0.tar.gz
wget https://files.pythonhosted.org/packages/11/83/2a5ce187bbe628f60f6053556d8c0cc2905e0246df8d8ed4bc11779873ad/python-gammu-3.2.4.tar.gz
```
(reprendre les liens depuis le site officiel, c'est plus prudent: https://fr.wammu.eu/download/ ))
Pour Gammu, j'ai installé quelques dépendances, mais pas toutes, je n’utilise pas les options bluetooth, sql, etc..
`sudo apt-get install cmake python3-dev pkg-config libcurl4-gnutls-dev libusb-1.0-0-dev libgudev-1.0-dev libglib2.0-dev
`
ensuite la compilation s'est faite sans soucis:
```
tar -xzf gammu-1.42.0.tar.gz
cd gammu-1.42.0
./configure
make
sudo make install
```
A noter que gammu se retrouve dans /usr/local/bin au lieu de /usr/bin (important si vous utilisez des scripts avec chemin en dur)
J'ai refait le lien symbolique qui pointait vers l'ancienne librairie (1.41->1.42):
```
sudo rm /usr/lib/libGammu.so.8
sudo ln -s /usr/local/lib/libGammu.so.8.1.42.0  /usr/lib/libGammu.so.8
```
A partir de là, Gammu doit fonctionner et renvoyer la version 1.42 (testé via 'gammu --version')

Pour python-gammu:
```
tar xzf python-gammu-3.2.4.tar.gz
cd python-gammu-3.2.4
./setup.py build
sudo ./setup.py install
```
Pour vérifier la version (doit renvoyer 3.2.4): 
`pip3 show python-gammu`

FredMa01, 2022-11-09:
@Vikingfr , tes explications sont top pour quelqu'un qui n'est jamais sorti des paquets via apt ;)
J'ai un souci à la fin pour python-gammu. La commande
./setup.py build me renvoie l'erreur :
```
  File "./setup.py", line 4
SyntaxError: Non-ASCII character '\xc2' in file ./setup.py on line 4, but no encoding declared; see http://python.org/dev/peps/pep-0263/ for details

```
Si j'ajoute l'encodage utf8 en début de fichier setup.py : 
# -*- coding: utf-8 -*-
Cela ne se passe pas mieux : 
```
./setup.py: ligne 25: $'\npython-gammu - Phone communication libary\n' : commande introuvable
import-im6.q16: impossible de se connecter au serveur X `' @ error/import.c/ImportImageCommand/346.
import-im6.q16: impossible de se connecter au serveur X `' @ error/import.c/ImportImageCommand/346.
import-im6.q16: impossible de se connecter au serveur X `' @ error/import.c/ImportImageCommand/346.
import-im6.q16: impossible de se connecter au serveur X `' @ error/import.c/ImportImageCommand/346.
import-im6.q16: impossible de se connecter au serveur X `' @ error/import.c/ImportImageCommand/346.
import-im6.q16: impossible de se connecter au serveur X `' @ error/import.c/ImportImageCommand/346.
from: trop d'arguments
from: trop d'arguments
./setup.py: ligne 39: VERSION : commande introuvable
./setup.py: ligne 40: GAMMU_REQUIRED : commande introuvable
./setup.py: ligne 43: erreur de syntaxe près du symbole inattendu « ( »
./setup.py: ligne 43: `THIS_DIRECTORY = os.path.abspath(os.path.dirname(__file__))'

```

FredMa01, 2022-11-09:
En mettant un peu après la ligne, elle n'est pas prise en compte. Du coup j'ai modifié la ligne 4 avec ses caractères particuliers.
J'ai ensuite des erreurs 
```
 ./setup.py build
  File "./setup.py", line 93
    f"--atleast-version={GAMMU_REQUIRED}",
```Je vire le f de la ligne 93 et encore une erreur :
```
  File "./setup.py", line 35, in <module>
    from setuptools import Extension, setup

```
Bref pas si simple ;)

deajan, 2022-11-09:
@Vikingfr Merci du retour ;)
@FredMa01 Cela sent quand meme fortement le script Python 2.x qui tente de compiler du python 3
Vous avez quelle version de python pour compiler python-gammu ?

FredMa01, 2022-11-09:
Bien vu, j'ai Python 3.9.2 et Python 2.7.18 qui cohabitent ;)
```
root@pi4:~# python -V
Python 2.7.18
root@pi4:~# python3 -V
Python 3.9.2

```
Il faut que je fasse une manip particulière pour compiler avec python3...
Si je modifie le lien symbolique /usr/bin/python pour qu'il pointe vers python3, je risque de casser d'autres trucs non ?
Merci de votre patience...
Peut-être faudra-t-il déplacer ces posts sur une autre issue spécifique au passage de Gammu en 1.42...

deajan, 2022-11-09:
Le plus simple, le temps de la compilation, est de créer un alias type `alias python=/usr/bin/python3`.
sinon, de mémoire, il y a moyen de spécifier python3 par défaut avec alternatives pour un utilisateur (je ne travaille pas avec debian donc google est votre ami ;)

Vikingfr, 2022-11-09:
il y a un paquet qui se nomme python-is-python3 ui crée les liens symboliques.
`sudo apt install python-is-python3`
Sinon, lancer le script via python:
`python3 setup.py build`
puis
`sudo python3 setup.py install`

FredMa01, 2022-11-09:
`python3 setup.py build` fonctionne parfaitement.
@Vikingfr est mon ami ;)

FredMa01, 2022-11-16:
Alors moi maintenant Gammu 1.42 fonctionne bien. J'ai eu un peu de mal à voir comment récupérer le pull request pour gammu_get_unread_sms.py de @deajan  ;)
Je suis passé par un  : 
`wget https://raw.githubusercontent.com/RaspbianFrance/raspisms/185d7772f70be981203c9e9b4b06bc6b0ef45a9b/bin/gammu_get_unread_sms.py`
mais il doit y avoir mieux. 
Enfin j'ai bien la nouvelle version de ''gammu_get_unread_sms.py''. J'ai relancé raspisms mais les SMS ne sont pas supprimés... Y a un truc qui a dû m'échapper.
Je cherche ;).

deajan, 2022-11-16:
@FredMa01 Normal, mon script attend un --delete de la part de RaspiSMS.
Pour faire court, modifiez le script ligne 223 `delete = False` par `delete = True` et cela devrait fonctionner.

FredMa01, 2022-11-16:
Nickel, ça marche merci !

OsaAjani, 2023-05-30:
Ça devrait être fix avec la 3.7.0

closed 2023-05-30 by OsaAjani