RaspbianFrance/raspisms #201: Interface REST classique

deajan, 2022-11-07:
Bonjour,

Certains des logiciels que je souhaite interfacer avec RaspiSMS utilisent des URL GET simples pour l'envoi de SMS, type `http://raspisms/envoi?numero=%2B33XXXXXXX&text=Mon%20Super%20Message` avec les paramètres dans l'URL.

RaspiSMS ne propose que des URL qui requirent des champs de données ou de formulaire ainsi qu'un header.
Existe-t-il une option de rétrocompatibilité pour une bonne vieille requête GET ?

Merci.

Vikingfr, 2022-11-16:
Bonjour,

j'ai créé un bout de code php pour faire ça, car je suis confronté au même soucis. Il faudra peut-être l'adapter à votre besoin et votre configuration. J'ai placé ce script dans /var/www/html
L'apikey est en dur et en clair, il faut être attentif à la sécurité. Dans mon usage, il n'y a pas de risque particulier. On pourrait prévoir de ne pas l’écrire et la transmettre dans la requête GET.
L'appel se fait par 
`https://mon.url/raspiGet.php?number=%2B33600000000&text=Bonjour+a+tous
`
```
<?php
$apikey="123467890123467980123465789";
$url="https://127.0.0.1/raspisms/api/scheduled/";

$number=htmlspecialchars($_GET["number"]);
$text=$_GET["text"];
$data=array(
        'numbers'=>$number,
        'text'=>$text 
);
$header=array("X-Api-Key: ".$apikey);

$ch = curl_init();

curl_setopt($ch, CURLOPT_URL, $url);
curl_setopt($ch, CURLOPT_HEADER, TRUE);
curl_setopt($ch, CURLOPT_NOBODY, FALSE); 
curl_setopt($ch, CURLOPT_RETURNTRANSFER, TRUE);
curl_setopt($ch, CURLOPT_SSL_VERIFYHOST, FALSE); // remove SSL check
curl_setopt($ch, CURLOPT_SSL_VERIFYSTATUS, FALSE); // remove SSL check

curl_setopt($ch, CURLOPT_POST, 1);
curl_setopt($ch, CURLOPT_POSTFIELDS, $data);

curl_setopt($ch, CURLOPT_HTTPHEADER, $header);

$result=curl_exec($ch);
$headerSize = curl_getinfo( $ch , CURLINFO_HEADER_SIZE );
$headerStr = substr( $result , 0 , $headerSize );
$headerArray=explode("\r\n", $headerStr);
$bodyStr = substr( $result , $headerSize );

// on reccupère la réponse de raspiSMS et on la renvoi (header + contenu)
if (!curl_errno($ch)) {
        foreach ($headerArray as $line) {
                header($line);
        }
        print_r($bodyStr);
} else {
        header("HTTP/1.1 303 ".curl_error($ch));
        echo "HTTP/1.1 303 ".curl_error($ch);
}

?>
```

deajan, 2022-11-22:
@Vikingfr Ca m'a l'air pas mal. Je vais l'intégrer en sécurisant l'IP ayant le droit d'appel, et on devrait pas être trop mal.

deajan, 2022-12-12:
Petit retour sur le script @Vikingfr 
Merci pour le travail effectué.

Je l'ai posé dans `/usr/share/raspisms` et j'ai modifié le fichier `.htaccess` comme suit:
```
RewriteEngine on
RewriteRule ^assets - [L]
RewriteRule ^.well-known - [L]
RewriteRule ^data/public/ - [L]
RewriteRule ^raspiGet.php - [L]
RewriteRule . index.php
```

Enfin, j'ai restreint l'accès à cette URL via le reverse proxy.