RaspbianFrance/raspisms #204: CRITICAL: Error reading MSG SEND Queue

LaJobe, 2022-12-08:
Bonjour, RaspiSMS fonctionnait bien sur un Pi4 avec module SIM800L installé selon le modèle (diode+condensateur).
Depuis quelques jours, sans raison identifiable, plus rien ne fonctionne et le daemons.log enchaine les messages suivants à chaque seconde…

_RaspiSMS Daemon Phone 1.CRITICAL: Error reading MSG SEND Queue, error code :  [] []_

J'ai rebooté, éteint, mis à jour le Pi4 et la distribution officielle raspisms. Rien de mieux.

Comment effectuer un flush de la MSG SEND Queue ?
S'agit-il d'un problème connu et contournable ? Je n'ai rien trouvé en ligne.

LaJobe, 2022-12-09:
J'ai continué à investiguer et voici un complément d'infos.

Avec minicom, j'arrive à envoyer des SMS via les commandes AT. Mais "test sms" arrive en "ÜÑ-ñ" sur mon iPhone…
Lors de l'envoi d'un nouveau SMS avec le back office raspisms, j'ai un retour encadré vert qui précise que le message est bien parti. Or il n'est pas reçu et n'apparait pas non plus dans la liste des SMS envoyés.

Dans les daemons log il y a :

```
[2022-12-09T11:59:50.362607+01:00] RaspiSMS Daemon Phone 1.CRITICAL: Error reading MSG SEND Queue, error code :  [] []
[2022-12-09T11:59:50.864033+01:00] RaspiSMS Daemon Phone 1.CRITICAL: Error reading MSG SEND Queue, error code :  [] []
[2022-12-09T11:59:51.365502+01:00] RaspiSMS Daemon Phone 1.CRITICAL: Error reading MSG SEND Queue, error code :  [] []
[2022-12-09T11:59:54.959377+01:00] RaspiSMS Daemon Sender.INFO: Transmit sms send signal to phone 1 on queue QUEUE_ID_PHONE with message type 01. [] []
[2022-12-09T11:59:54.959546+01:00] RaspiSMS Daemon Sender.INFO: Scheduled 32 treated and deleted. [] []
[2022-12-09T11:59:55.545783+01:00] RaspiSMS Daemon Phone 1.CRITICAL: Error reading MSG SEND Queue, error code :  [] []
[2022-12-09T11:59:56.047388+01:00] RaspiSMS Daemon Phone 1.CRITICAL: Error reading MSG SEND Queue, error code :  [] []
[2022-12-09T11:59:56.549092+01:00] RaspiSMS Daemon Phone 1.CRITICAL: Error reading MSG SEND Queue, error code :  [] []
```

et puis plus loin :

```
[2022-12-09T12:04:05.164135+01:00] RaspiSMS Daemon Phone 1.CRITICAL: Error reading MSG SEND Queue, error code :  [] []
[2022-12-09T12:04:05.665684+01:00] RaspiSMS Daemon Phone 1.CRITICAL: Error reading MSG SEND Queue, error code :  [] []
[2022-12-09T12:04:06.167316+01:00] RaspiSMS Daemon Phone 1.CRITICAL: Error reading MSG SEND Queue, error code :  [] []
[2022-12-09T12:04:06.670756+01:00] RaspiSMS Daemon Phone 1.CRITICAL: Error reading MSG SEND Queue, error code :  [] []
[2022-12-09T12:04:07.172368+01:00] RaspiSMS Daemon Phone 1.CRITICAL: Error reading MSG SEND Queue, error code :  [] []
[2022-12-09T12:04:07.674012+01:00] RaspiSMS Daemon Phone 1.CRITICAL: Error reading MSG SEND Queue, error code :  [] []
[2022-12-09T12:04:07.674452+01:00] RaspiSMS Daemon Phone 1.INFO: Stopping Phone daemon with pid 945 [] []
[2022-12-09T12:04:08.372735+01:00] RaspiSMS Daemon Phone 1.INFO: Starting Phone daemon with pid 997 [] []
[2022-12-09T12:04:08.876252+01:00] RaspiSMS Daemon Phone 1.CRITICAL: Error reading MSG SEND Queue, error code :  [] []
[2022-12-09T12:04:09.377530+01:00] RaspiSMS Daemon Phone 1.CRITICAL: Error reading MSG SEND Queue, error code :  [] []
[2022-12-09T12:04:09.879025+01:00] RaspiSMS Daemon Phone 1.CRITICAL: Error reading MSG SEND Queue, error code :  [] []
[2022-12-09T12:04:10.380465+01:00] RaspiSMS Daemon Phone 1.CRITICAL: Error reading MSG SEND Queue, error code :  [] []
[2022-12-09T12:04:10.881805+01:00] RaspiSMS Daemon Phone 1.CRITICAL: Error reading MSG SEND Queue, error code :  [] []
```

deajan, 2022-12-12:
Est-ce que RaspiSMS recoit encore des SMS ?

LaJobe, 2022-12-12:
Oui, RaspiSMS reçoit et affiche comme il faut les messages qu’on lui envoit

LaJobe, 2023-01-02:
Toujours coincé avec ce problème…
Dois-je repartir d'une installation vierge ? Et si cela se représente ?

LaJobe, 2023-01-31:
Je suis reparti d'une clean Install de RaspiOS.
La version 11 Bullseye ne semble pas compatible… Je suis donc reparti en version 10 Buster et éviterai les mises à jour…

FredMa01, 2023-01-31:
RaspiSMS fonctionne bien sur Bullseye, malgré l'absence de release : https://github.com/RaspbianFrance/raspisms/issues/184