RaspbianFrance/raspisms #213: Bug MMS

Tom46fr, 2023-10-03:
Bonjour à tous,

J'ai testé RaspiSMS sur RPi et en SaaS.
Sur la version SaaS avec passerelle sur un téléphone Android, j'ai un étrange bug sur la réception des MMS.
A chaque réception j'obtiens non pas l'image mais simplement "[MEDIA ONLY]" alors que l’émission de MMS marche parfaitement bien et que la réception directement sur le téléphone est parfaitement fonctionnelle.

OsaAjani, j'ai essayer de prendre contact par email, sans succès...
Bref je suis pas contrôle un coup de pouce car l'usage de la solution SaaS est justement pour la gestion de MMS.

Merci de votre aide :) !
Tom.

OsaAjani, 2023-10-04:
Oups, j'avais complétement oublié de répondre au mail, je regarde ça ce soir.

J'avais essayé de reproduire le problème mais je n'avais pas réussi. Pouvez-vous essayer de démarrer l'application RaspiSMS Relay et d'envoyer un MMS vers le téléphone application ouverte ?

Tom46fr, 2023-10-04:
Pas de problème.

Appli ouverte j’ai certains MMS ou seul le texte passe, dans ce cas précis je n’ai même plus l’alerte MEDIA ONLY.

Sinon le résultat reste le même sur les nouveaux tests du jour.

Le téléphone fonctionne exclusivement en Wifi, et la réception des MMS sur l’appli message marche toujours parfaitement.

 

Je peux pas en dire beaucoup plus..

On se tient au courant s’il y a besoin de plus de test.

 

Tom.

OsaAjani, 2023-10-08:
Serait-il possible de récupérer les fichiers de logs dans la partie "réglages" de l'application et de me les envoyer par mail.