RaspbianFrance/raspisms #214: Syntaxe pour l'API si l'on veux passer des paramètres autre que "numbers"

Ma-Maison-Intelligente, 2023-10-25:
Bonjour et merci pour votre super travail.

Voulant utilisé RaspiSMS pour envoyé des SMS à partir d'un ESP32, je code le message POST en C++

Le code est le suivant, l'APIKey, le texte du SMS ainsi que le n° de téléphone sont stockés dans des variables

HTTPClient http;
      http.begin("https://app.raspisms.fr/api/scheduled");// J'utilise le site de test pour l'instant
      http.addHeader("X-Api-Key", apiKeyWhatsApp.c_str()); 
      http.addHeader("Content-Type", "application/x-www-form-urlencoded");

      String data = "text="+urlEncode(messageString)+"&numbers="+urlEncode(phoneNumber);
      
      int httpResponseCode = http.POST(data);
      
      http.end();
      
Ce code fonctionne très bien, je constate l'émission du SMS qui apparait dans SMS Envoyé.

La ou je rencontre un problème c'est quand je veux ajouter un critère conditional_groups. 
La documentation ne décrit pas la syntaxe à utiliser (ou alors je n'ai pas trouvé...).

Sur le site il y a un conditional_groups Hommes par exemple. Je veux utiliser ce critère pour m'assurer que le N° destinataire du SMS fait partie du groupe Hommes ce qui me permet de n'envoyer des SMS qu'a des personnes qui existent dans ma liste de contact sous RaspiSMS.

J'ai donc modifié mon code
String data = "text="+urlEncode(messageString)+"&numbers="+urlEncode(phoneNumber)+"&conditional_groups[]=Hommes"; 

J'ai essayé beaucoup d'autres syntaxes mais sans résultat : 
  - soit cela me retourne une erreur 
  - soit je n'ai pas d'erreur mais je ne vois pas de SMS envoyé sur le site de test.
  - quand il n'y a pas d'erreur, cela m'envoie systématiquement un SMS sur le numéro de téléphone que j'ai mentionné alors qu'il n'existe pas dans ma liste de contact (j'ai fait exprès pour tester).

J'ai aussi essayé d'envoyé des SMS à uniquement un conditional_groups sans préciser de numéro dans le POST.
Soit j'ai des retour d'erreur soit j'ai un résultat OK mais aucun SMS dans la boite d'envoie.

D'avance merci pour votre aide.

OsaAjani, 2023-10-25:
> conditional_groups (array | int), optional – L’ID du groupe conditionnel auquel envoyer le SMS ou un tableau d’IDs.

Deuxième option ^^

> conditional_groups (array | int), optional – L’ID du groupe conditionnel auquel envoyer le SMS ou un tableau d’IDs.

Ma-Maison-Intelligente, 2023-10-25:
Merci pour votre réponse mais malheureusement elle ne m’aide pas.

J’avais bien vu l’élément que vous mentionnez dans la documentation mais cela n’est pas clair pour moi.

En effet, je ne dispose que de connaissances limitées en informatique (autodidacte a l’aide des forums et autres tutos, merci au passage à tout les contributeurs).

Pourriez vous me préciser la syntaxe complète svp?

Ma-Maison-Intelligente, 2023-10-26:
Merci pour ce retour.

Je me doutai bien qu'il fallait entrer un nombre, conditional-groups étant de type "int". 
Comment et où faire la requête pour lister les ressources de type conditonal_group (quand je fait mes teste sur le site https://app.raspisms.fr et quand j'aurais installé RaspiSMS chez moi sur mon Raspberry) ?
Une fois un conditonal_group crée, il garde son ID même si on ajoute et supprime d'autre conditonal_group  créé avant ou après ?

J'avais essayé la syntaxes "&conditional_groups[]=1 mais aucun SMS n'apparait sur le serveur de Test https://app.raspisms.fr/sended/ (menu SMS Envoyé) alors que j'ai bien dans le menu Logs/Evénements : Ajout d'un Sms pour le 2023-10-26 10:01:10.

Je ne sais donc pas à qui a été envoyé le SMS.

Nouveau test fait en passant le critère &numbers=+33600000044 (N° qui n'est pas dans le répertoire de RaspiSMS) ET  &conditional_groups[]=1
Resultat:
  - La requête ne renvoie pas d'erreur 
  - Je vois le SMS envoyé au +33600000044 (!!! ce que justement je ne veux pas car ce n° n'est pas dans le répertoire !!! je ne sais pas si l'application supporte cette condition d'envoyer un SMS à un numéro ET que ce n° soit dans un groupe conditionnel)
  - dans les Logs j'ai 2 lignes : 
       * Ajout d'un Sms pour le 2023-10-26 10:10:12. 
       * Consume 1 credits of SMS quota.

Comme indiqué dans mon post précédant je ne suis pas informaticien donc si vous pouviez détailler vos réponse cela m'aiderai car ce qui est évident pour vous est loin de l'être pour moi.
Encore merci pour votre aide.

OsaAjani, 2023-10-26:
L'apparition du log de consommation d'un crédit correspond au numéro indiqué dans la requête.

Les infos sur la récupération d'une liste de ressources sont dans la doc avec des exemples de requête curl associées. 

L'id d'une ressource n'est jamais modifié.

Pour un code précis je suis désolé mais ce n'est pas vraiment le bon endroit pour ce type de support et je ne saurais pas donner la syntaxe précise en c++. 

Comme il ne s'agit pas d'un bug du système je vais clore l'issue et je vous encourage à aller poser la question sur un forum où vous trouverez peut-être plus facilement de l'aide, comme forum.raspberry-pi.fr.

closed 2023-10-26 by OsaAjani

Ma-Maison-Intelligente, 2023-10-26:
J'ai relu la doc et refait des teste, c'est maintenant OK pour moi pour récupérer les infos de la base de donnée.

Exemple (sur le serveur de test de RaspiSMS) faire un GET sur https://app.raspisms.fr/api/list/conditional_group/?api_key=dbb1fa7d163ca6f71ebd8f105e6d086c permet de récupérer les info de conditional_group
Le retour de cette requête donne:
{"id":476918,"id_user":3122,"name":"Femmes","condition":"exists(contact.gender) && contact.gender == 'female'","created_at":"2023-10-26 15:50:01","updated_at":null},{"id":476919,"id_user":3122,"name":"Hommes","condition":"exists(contact.gender) && contact.gender == 'male'","created_at":"2023-10-26 15:50:01","updated_at":null},{"id":476920,"id_user":3122,"name":"Tous","condition":"true","created_at":"2023-10-26 15:50:01","updated_at":null}]

 ******* LA SOLUTION par rapport au titre de ce topic *******

Avant de faire la requête POST il faut faire une requête GET (voir exemple ci-dessus sur le champs conditional_group ) pour récupérer les éléments nécessaire à notre requête POST 
Passer ensuite une requête POST avec les éléments qui vous intéressent
  Dans l'entête de POST il faut le site et l'APIKEY 
       https://app.raspisms.fr/api/scheduled/?api_key=dbb1fa7d163ca6f71ebd8f105e6d086c
  Dans le corp du POST il faut les autres paramètre
       text="Ceci_est_le_message_contenu_dans_le_SMS"&numbers="%2B33600000044"&conditional_groups[]=476919
       
ATTENTION ce POST envoi des SMS au N° mentionné ET aux N° se trouvant dans le groupe !!!