RaspbianFrance/raspisms #217: problème installation raspisms 3.0 sur ubuntu 22.04

Kocardi, 2024-01-01:
Bonjour
Je tente d'installer Raspisms sur Ubuntu en suivant la procedure sur le site "raspisms.fr" .
Tout a l'air de bien se passer jusqu'au moment ou je cherche le fichier ".credentials" qui n'a pas été créé.
J'ai le message "cat: /usr/share/raspisms/.credentials: Aucun fichier ou dossier de ce type"
Quelqu'un aurais une idée  SVP

Kocardi, 2024-01-16:
je suis le seul à avoir un soucis d'installation ?

GBInformatics, 2024-03-28:
> je suis le seul à avoir un soucis d'installation ?

non pas dutout, je suis également bloqué

vincent1890, 2024-06-15:
> Bonjour Je tente d'installer Raspisms sur Ubuntu en suivant la procedure sur le site "raspisms.fr" . Tout a l'air de bien se passer jusqu'au moment ou je cherche le fichier ".credentials" qui n'a [...]

Bonjour lors de l'installation dans la console a un moment celle-ci doit afficher :

####### DATABASE ######
Host: localhost
User: raspisms
Pass: 6gfhfdhjtn9Sdtfgjhfjrt542WwdtrhfjdxywrgCRl3d
Name: raspisms

You can find those in /usr/share/raspisms/.credentials_db
#######################

####### RASPISMS USER ######
Email: admin@example.com
Password: 6gfhfdhjtn9Sdtfgjhfjrt542WwdtrhfjdxywrgCRl3d
Admin: true

You can find those in /usr/share/raspisms/.credentials
############################

Donc si jamais les fichiers ne sont pas créer automatiquement vous pouvez les créer par vous meme les deux fichiers en cas de besoin en copiant collant l'affichage de la console dans les bon fichiers
vi /usr/share/raspisms/.credentials_db
vi /usr/share/raspisms/.credentials

PS: Les mots de passe indiqué ici sont des examples

vincent1890, 2024-06-15:
> Bonjour Je tente d'installer Raspisms sur Ubuntu en suivant la procedure sur le site "raspisms.fr" . Tout a l'air de bien se passer jusqu'au moment ou je cherche le fichier ".credentials" qui n'a [...]

Vous n'avez eu aucune erreur a l'installation ?

Vous pouvez tenter l'installation d'un deb NON officiellement sortie car cela n'est pas une release ! (A utiliser a vos risque et périls)
https://github.com/RaspbianFrance/raspisms-deb/blob/master/packages/latest.deb