RaspbianFrance/raspisms #218: Créer BDD via phinx KO

raynoxfr, 2024-03-04:
Bonjour,

Je cherche à démarrer raspisms sur docker.
J'ai donc créé un dockerFile avec les librairies nécessaires (je parviens à envoyer un SMS via gammu), ainsi qu'un docker-compose (qui me permet d'accéder à l'HTTP).

Toutefois, lorsque je crée la BDD via (en MySQL 8) :
`php vendor/bin/phinx migrate`

J'ai l'erreur suivante :

[image]

Avez-vous une idée à m'apporter afin de me débloquer ?
Merci

OsaAjani, 2024-03-04:
Je te conseils d'ajouter des logs dans le fichier add_created_at..., spécifiquement pour voir ce qui est retourné dans $database_name et dans $tables.

raynoxfr, 2024-03-04:
J'ai cherché pour afficher la valeur mais sans succès, j'ai essayé :
`$this->getOutput()->writeln($query);

En regardant les ibd de MySQL, je vois bien des tables dont phinxlog

[image]

Merci par avance

raynoxfr, 2024-03-04:
Pour info, je viens de tester avec MariaDB, et ca c'est fini correctement.
Il y a surement un problème de comptabilité avec MySQL 8 (dans mon cas)
[image]

closed 2024-03-04 by raynoxfr

GBInformatics, 2024-03-27:
> Pour info, je viens de tester avec MariaDB, et ca s'est fini correctement. Il y a surement un problème de comptabilité avec MySQL 8 (dans mon cas) ![image](https://private-user-images.githubusercontent.com/38424389/309833382-bc592a5f-8642-4ee0-943c-b17fa6d3799b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE1Mzk5MzIsIm5iZiI6MTcxMTUzOTYzMiwicGF0aCI6Ii8zODQyNDM4OS8zMDk4MzMzODItYmM1OTJhNWYtODY0Mi00ZWUwLTk0M2MtYjE3ZmE2ZDM3OTliLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzI3VDExNDAzMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQxYWRhNWNlNjYxMTk0MzljYTVkODNkMzYwODJjZGQ0NGJmNWI1ZmQ0YjRlMjg1MWE5ZjlmMTM0YzRjZDFjYTYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.n-1es6U9N7jIWtzDXMKJq_FeOaNQE4NKIo-KkSEUS9U)

Bonjour, je suis bloqué au même endroit que vous. J'ai une rpi 3B+ avec raspios Bullseye 32 bits. Est ce que c'est possible de m'aider s'il vous plait ?

raynoxfr, 2024-03-27:
Bonjour,
J'ai mis en place MariaDB en remplacement de MySQL 8 qui m'a permis de contourner le problème.
J'ai fait une installation complète (à partir du dépôt GitHub pour info)

GBInformatics, 2024-03-27:
> Bonjour, J'ai mis en place MariaDB en remplacement de MySQL 8 qui m'a permis de contourner le problème. J'ai fait une installation complète (à partir du dépôt GitHub pour info)

Vous n'avez pas suivi ce lien : https://raspisms.fr/download/ ? et sinon il est possible d'avoir une image pour le raspberry sans vos informations ? je suis perdu

raynoxfr, 2024-03-27:
Si j'ai suivi ce lien, mais j'ai fait l'installation "Installation pour les autres distributions Linux."
Cette seconde partie à mes yeux, il manque certaines parties comme la création de l'utilisateur pour se connecter, donc j'ai dézippé le .deb pour comprendre les étapes et les appliquer à mon installation, via le postinst du package.

J'utilise la solution Docker et je suis en train de créer une image (en cours de finalisation)
Tout dépendant de votre niveau d'informatique, mais le mieux est d'installer le .deb fourni sur le site, si vous n'êtes pas à l'aise.
Si vous êtes plus à l'aise avec l'informatique, je peux vous mettre à disposition une image docker, seul le fichier YAML est à définir et raspisms va démarrer avec sa base de données.

GBInformatics, 2024-03-27:
Pour ma part j'ai pris la solution du lien qui me paraissait "sure" mais ce sont des bouts d'éléments. Je vais supprimer mysql pour remplacer par MariaDB.
Je ne suis pas trop à l'aise avec le PHP

GBInformatics, 2024-03-27:
> Si j'ai suivi ce lien, mais j'ai fait l'installation "Installation pour les autres distributions Linux." Cette seconde partie à mes yeux, il manque certaines parties comme la création de l'utilisateur pour se connecter, donc j'ai dézippé le .deb pour comprendre les étapes et les appliquer à mon installation, via le postinst du package.
> 
> J'utilise la solution Docker et je suis en train de créer une image (en cours de finalisation) Tout dépendant de votre niveau d'informatique, mais le mieux est d'installer le .deb fourni sur le site, si vous n'êtes pas à l'aise. Si vous êtes plus à l'aise avec l'informatique, je peux vous mettre à disposition une image docker, seul le fichier YAML est à définir et raspisms va démarrer avec sa base de données.
> 
> Perso, je ne suis pas sur Raspberry, mais sur un ordinateur faisant tourner des images Docker

je n'ai pas mysql mais maria-db d'installé

GBInformatics, 2024-03-28:
> `php vendor/bin/phinx migrate`

J'ai trouvé la cause. Je suis passé sur mariadb mais le problème venait du mot de passe crypté. En le changeant et en utilisant la commande `php vendor/bin/phinx migrate`, je peux me connecter au compte raspisms.

Par contre, vous aussi après l'authentification vous êtes bloqués sur http://192.168.X.X/raspisms/connection/ -> 
Not Found

The requested URL was not found on this server.
Apache/2.4.56 (Raspbian) Server at 192.168.X.X Port 80

J'ai rémarqué que dans le chemin est ip_du_serveur/raspisms/connection sur le navigateur web mais sur le rpi le chemin n'est pas aussi long. C'est uniquement /usr/share/raspisms.

Savez-vous où se situe le dossier "connections" ?

GBInformatics, 2024-03-29:
> Si j'ai suivi ce lien, mais j'ai fait l'installation "Installation pour les autres distributions Linux." Cette seconde partie à mes yeux, il manque certaines parties comme la création de [...]

J'ai trouvé la cause, le mot de passe crypté dans les fichiers env.prod.php et phinx.yml

Vous avez le dossier connection dans raspisms ?