RaspbianFrance/raspisms #219: HTTPS "bug"

raynoxfr, 2024-03-05:
Bonjour,

Je reviens à la charge, sur mon image docker...
Je parviens maintenant à avoir la page avec un format attendu, tout en tournant dans un containeur pour raspisms et un 2nd pour mariadb, sans oublier d'activer le mode rewrite d'apache.

Selon moi il y a un "bug" sur la gestion de l'HTTPS.
En effet, lorsque je regarde mon paramétrage du env.prod.php :
`'STATIC_HTTP_URL' => 'https://sms.domaine.fr',`

et que je charge l'URL, j'ai :
[image]

L'accès se fait via traefik, qui permet de géner automatiquement l'HTTPS et les routes vers les différents containeurs.
Le paramétrage de l'HTTPS est géré uniquement au niveau de traefik, pour ainsi ne pas avoir à le gérer pour chaque conainteurs.
Jusqu'à maintenant, je n'ai jamais eu de souci particulier ou devoir modifier une valeur dans un programme suite à cette mise en place.

Je constate que même ayant défini plus haut "https://sms.domaine.fr", il y a des liens qui sont interrogés en http, d'ou la page ci-dessus.

Via le mode développeur de safari, je me rends compte qu'il y a des fichiers bloqués en "http" :

`[Warning] [blocked] The page at https://sms.domaine.fr/ was not allowed to run insecure content from http://sms.domaine.fr/assets/css/bootstrap.min.css. (sms.domaine.fr, line 13)`

La solution, que j'ai trouvé actuellement, ajouter dans le fichier env.prod.php :

$environment['HTTP_PROTOCOL'] = 'https://'

Afin d'obtenir un $http_pwd en https:// au lieu de http://.
Dans l'idéal, il faudrait récupérer les informations à partir de l'URL qui a été paramètre dans STATIC_HTTP_URL, afin d'être certain d'utiliser le bon protocole.

PS : je n'ai pas vu d'image ou de dockerfile, ce qui aurait pu être intéressant, pour déployer rapidement l'application, sans devoir suivre l'installation manuelle.

Merci

raynoxfr, 2024-03-05:
Au juste, est-ce que je cherche au mauvais endroit ?
Car je n'ai pas trouvé de documentation pour créer l'utilisateur pour se connecter. J'ai fini par aller dans le package DEB pour récupérer le "postinst" et je suis en train de chercher ce qu'il manque et comment configurer.

raynoxfr, 2024-03-05:
Je parviens à envoyer des SMS maintenant à partir de l'interface raspisms, impossible de les récupérer. J'ai trouvé sur le forum une commande à lancer et je ne vois pas de SMS en attente de lecture, doit y avoir un petit coup de tournevis...

Ensuite, lorsque tout sera fonctionnel, il serait intéressant de mettre l'image sur Hub docker, quand tout sera mis au propre.

Le dockerfile se base en faisant un clone github de clone.

OsaAjani, 2024-03-06:
`En effet, lorsque je regarde mon paramétrage du env.prod.php :
'STATIC_HTTP_URL' => 'https://sms.domaine.fr',`

Alors tu es tombé juste après un changement de la façon dont l'url est calculé par RaspiSMS. En fait `STATIC_HTTP_URL` ne devrait plus exister, il faudrait le calculer directement depuis les autres infos, je vais essayer de faire la modif rapidement dans le code et dans la doc. Cette adresse est uniquement utilisée pour calculer le chemin des medias dans les MMS.

Normalement l'appli essaie automatiquement de calculer s'il doit définir l'adresse en http ou en https, j'imagine que dans ton cas traffic passe la requête en HTTP et donc du point de vue de l'appli c'est une adresse http qui est demandée.

La façon correcte de faire est de créer une constante `HTTP_PROTOCOL` à https dans un des fichiers env, par example `env.prod.php`.

Pour la création de l'utilisateur ça utilise effectivement le controller console.

Si tu as un dockerfile qui fonctionne bien et que tu es d'accord, je suis chaud de l'ajouter sur dockerhub.

raynoxfr, 2024-03-06:
Parfait, je te tiens au courant.

Me reste la récupération des sms (faut que j'y travaille, avec gammu et voir si il arrive à voir les sms non lu).

Enfin je mettrai au propre, ce que j'ai fait et je t'enverrai les éléments et le dockerfile, je pense que ça pourra faciliter pour d'autres personnes.

Merci

GBInformatics, 2024-03-28:
> Je parviens à envoyer des SMS maintenant à partir de l'interface raspisms, impossible de les récupérer. J'ai trouvé sur le forum une commande à lancer et je ne vois pas de SMS en attente de lecture, [...]

Bonjour vous avez réussi à générer le fichier .credentials ?

raynoxfr, 2024-03-28:
Le fichier .credentials correspond au résultat de la commande :
`
php console.php controllers/internals/Console.php create_update_user --email="$APP_USER_EMAIL" --password="$APP_USER_PASSWORD" --admin="$APP_USER_ADMIN"`

$APP_USER_EMAIL : Email
$APP_USER_PASSWORD : Password
$APP_USER_ADMIN : true

Ensuite vous pourriez utiliser $APP_USER_EMAIL / $APP_USER_PASSWORD pour vous connecter.
Je n'ai pas suivi la procédure classique...

GBInformatics, 2024-03-28:
> $APP_USER_EMAIL : Email $APP_USER_PASSWORD : Password $APP_USER_ADMIN : true

alors pour que ca soir clair : dans mon terminal je mets la commande : php console.php controllers/internals/Console.php create_update_user --email="$APP_USER_EMAIL" --password="$APP_USER_PASSWORD" --admin="$APP_USER_ADMIN" ? Celle va me créer le fichier .credentials ensuite ?

raynoxfr, 2024-03-28:
non il va créer l'utilisateur en BDD selon les informations communiquées donc que vous. connaissez...
Donc le credentials n'aura plus d'intérêt, car vous connaissez les informations en entrée.

le credential permet de générer un compte et d'afficher les informations de connexion.

GBInformatics, 2024-03-28:
> il va créer l'utilisateur en BDD se

lors de l'installation j'ai communiqué l'adresse mail et le mot de passe du compte pour me connecter sur raspisms

raynoxfr, 2024-03-28:
ok donc le credentials n'est pas utile, il rappelle les informations, si vous les connaissez déjà

GBInformatics, 2024-03-29:
> ok donc le credentials n'est pas utile, il rappelle les informations, si vous les connaissez déjà

Je mets l'adresse mise lors de la confiugration, au moment où il demande "e-mail address for raspisms administrateur account".

vincent1890, 2024-06-15:
> `En effet, lorsque je regarde mon paramétrage du env.prod.php : 'STATIC_HTTP_URL' => 'https://sms.domaine.fr',`
>
> Alors tu es tombé juste après un changement de la façon dont l'url est calculé par [...]

Bonjour,

Est ce que le problème va etre resolu rapidement ?
[https://github.com/RaspbianFrance/raspisms/issues/222](https://github.com/RaspbianFrance/raspisms/issues/222)

J'ai exactement le même souci les chemins sont construit en absolu alors qu'il devrais etre des chemins relatif pour s'adapter au context utilisateur et etre http ou https si derriere un proxy par exemple.

Merci