RaspbianFrance/raspisms #220: Error installation RaspiSMS on Raspberry PI 3B+ 64Bits

GBInformatics, 2024-03-06:
Hello, I would like to configure an sms server with raspisms.
The aim is to receive sms alerts from my servers on my mobile. In France, only Free allows this via their API.
As I have a dual SIM card, I'm with Orange. At the moment it's an application on my phone that does the transfer.
Instead of using a smartphone, I'd like to use a Raspberry Pi 3b+ 64 bits with raspisms.
To install it, I use the following tutorial: [https://raspisms.fr/download/](https://jankoors.com/click?redirect=https%3A%2F%2Fraspisms.fr%2Fdownload%2F&dID=1709711281795&hashId=10cf1b7da2a6ac677d3b7f083e666ce394db4ad5dbc8ff67954bb65b2799c4ff0a139f&linkName=https://raspisms.fr/download/).
After the first command block I get no error. I then run sudo apt update && sudo apt upgrade.
For the "sudo apt install raspisms" command, I get the error :
raspisms@rpi-raspisms:~ $ sudo apt install raspisms
Reading package lists... Done
Build dependency tree... Done
Read status information... Done
Some packages cannot be installed. This may mean
you have asked for the impossible, or, if you are using the
unstable distribution, that some packages have not yet been
been created or have not been released from Incoming.
The following information should help you resolve the situation:

The following packages contain unsatisfied dependencies:
raspisms : Dependency: gammu but it is not installable
Depends on: python3-gammu but it is not installable
Recommends: apache2 but will not be installed
E: Unable to fix problems, faulty packages are in "keep as is" mode.

What is the solution to correct this?

vincent1890, 2024-06-15:
Bonjour avez vous suivi la procedure d'installation situé ici [https://raspisms.fr/download/](https://raspisms.fr/download/) ?

Il n'y a pas de dossier connections dans le dossier ou est installé RaspiSMS

GBInformatics, 2024-06-15:
Je n'vais pas vu, autant pour moi.

Oui tout à fait j'ai suivi le tutoriel du lien.

vincent1890, 2024-06-15:
> Je n'vais pas vu, autant pour moi.
> 
> Oui tout à fait j'ai suivi le tutoriel du lien.

Vous n'avez eu aucune erreur a l'installation ?

Vous pouvez tenter l'installation d'un deb NON officiellement sortie car cela n'est pas une release ! (A utiliser a vos risque et périls)
https://github.com/RaspbianFrance/raspisms-deb/blob/master/packages/latest.deb

Ou alors telecharger le git sur le Github localement pour mettre a jour l'installation

GBInformatics, 2024-06-16:
Pourquoi pas en soit mais je ne voudrais pas tout perdre (ensuite) si j'arrive à l'installation. Est ce que vous vouslez que j'essaie votre installation sur une rapsberry pi chez moi ?

GBInformatics, 2024-06-22:
Bponjour j'ai essayé avec docker. L'achitecture de ma raspberry n'est pas bonne.. je suis donc bloqué

GBInformatics, 2024-10-17:
Bonjour il faut utiliser la version raspberry pi os 32 bits Legacy Lite

closed 2024-10-17 by GBInformatics