RaspbianFrance/raspisms #221: dockerHub / dockerfile

raynoxfr, 2024-03-30:
Bonjour,

J'ai créé un dockerfile (partant du repo GitHub Raspisms), ainsi que tout ce qu'il faut pour démarrer.
Je n'ai pas pu la tester en totalité, puisque j'ai des problèmes de comptabilité de téléphone.

Il y a uniquement raspisms dans l'image et ce qu'il faut pour qu'il puisse tourner (hors BDD).
D'autres ajouts, tel que le choix de la BDD est dans un autre container (voir le docker-compose).

Sinon, voici le projet pour l'image docker :
https://github.com/raynoxfr/raspisms-dockerfile
Tu peux récupérer l'ensemble du code.

Le nommage des variables sont les mêmes que ceux présents dans les fichiers de configuration PHP.
Si le répertoire /config est vide, une configuration de base est créée, par défaut (en tenant compte des variables définies).
Lorsque la configuration de base est créée, c'est les fichiers de configuration de /config qui sont pris en compte.

**CREATE_ALL_SETTING**

Si à "true", variable d'environnement, qui permet de créer les tables en BDD, ainsi que l'utilisateur admin par défaut
(rappel des identifiants dans le fichier /config/credentials et dans le log du container).

**APP_HTTP_PROTOCOL**

J'ai été obligé d'ajouter une variable, via le dockerfile, afin de définir le protocol http:// ou https:// (auto-alimenté selon l'URL).
Etant sur traefik, l'HTTPS est généré par celui-ci, mais il communique en HTTP entre traefik et raspisms...
La valeur est initialisée dans le start.sh

**Reste à faire (optionnel)**

- Je n'ai pas mis en place de gestion de l'HTTPS dans l'image, via la configuration Apache.
- Configurer le port de la BDD (3306 en dur)

Je me tiens à disposition, si t'as des remarques.

GBInformatics, 2024-03-31:
Bonjour merci pour votre image, vous n'êtes pas sur Raspberry Pi, est ce que s'est compatible ?

raynoxfr, 2024-05-05:
Bonjour,

Je viens de voir le message, en effet je ne suis pas sur raspberry, mais oui elle doit être compatible, au pire suffit de changer l'image de base du Dockerfile par une ARM si vraiment il y a un problème.

Avez-Vous pu essayer depuis ?

Le dockerfile se base sur github et lance les instructions nécessaires, donc aucune raison que ça ne fonctionne pas, c'est le but de docker avoir une image, sans de prendre la tête des dépendances etc

GBInformatics, 2024-05-05:
Bonjour,

Je tente petite à petite de passer sur docker car sans je suis toujours bloqué au même point.
J'ai réussi à installer docker sur mon raspberry pi 3B+ 64Bits mais je suis bloqué car mariadb n'est pas compatible. Je vais tenté de réinstaller via debian 10 pour voir ce que ca donne.

raynoxfr, 2024-05-05:
Pourtant une simple recherche sur internet "raspberry mariadb docker" on trouve des images

https://hub.docker.com/r/tobi312/rpi-mariadb/

Docker si tu peux y passer c'est mieux, ta machine lance des containers paramétrable, elle reste toujours "propre" quand on veut plus tu supprimes le conteneur et quand tu veux le mettre à jour c'est aussi simple.
Dans l'idéal pour gérer ton docker, il faut mettre l'interface Web Portainer, et utiliser des fichiers de configuration Docker Compose (yaml).

GBInformatics, 2024-05-05:
je débute, je comprends le fonctionnenment mais la mise en production ce n'est pas aussi simple.
Vous, il est fonctionnel ?

raynoxfr, 2024-05-05:
Actuellement, je ne suis pas sur Raspberry, mais oui mon docker fonctionne ainsi sue l'image sue j'ai buildé de raspisms.

GBInformatics, 2024-05-06:
Bonjour, j'ai fait ce docker-compose.yml. Pouvez- vous regarder s'il vous paraît correct s'il vous plait ?[https://www.transfernow.net/dl/20240506xNCmlqZq](url)

GBInformatics, 2024-05-07:
Bonjour, j'ai réussi à installer les containeurs httpd, mariadb-server. Par contre impossible d'installer raspisms depuis le docker HUB, à chaque fois j'ai l'erreur : `WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm/v7) and no specific platform was requested
exec /bin/bash: exec format error`

A partir de la je suis bloqué.

GBInformatics, 2024-05-29:
Bonjour, je reviens vers vous pour savoir si vous avez réussi à installer sur Raspberry PI. Même avec l'image je n'avance pas. Je suis également bloqué par le module usb sim 800c. Vous aussi ?

GBInformatics, 2024-06-20:
Bonjour, je reviens vers vous pour savoir si vous avez réussi à installer sur Raspberry PI. Même avec l'image je n'avance pas.