RaspbianFrance/raspisms #222: Request assistance updating and configuring RaspiSMS Behind an nginx Reverse Proxy

vincent1890, 2024-06-13:
Hello,

I would like to know the procedure for updating RaspiSMS, specifically the "WEB" part. I have been using version 3 for about 2 years and I am having issues accessing it behind an "nginx" reverse proxy. I absolutely do not understand why it doesn't work, as I have multiple other servers currently behind the same reverse proxy. I am unable to configure RaspiSMS correctly, which is currently accessible locally via http://192.168.10.99:8083/raspisms/ and works well. However, I want to make it accessible from outside to use the interface and the API, all in HTTPS, thanks to the reverse proxy.

Thank you for your help.

Have a nice day.

GBInformatics, 2024-06-15:
Bonjour vous avez utilisé quoi pour créer le serveur raspisms ? Raspberry ou docker ?

Je vous pose la question car je suis bloqué, je n'arrive pas à finir l'installation

vincent1890, 2024-06-15:
> Bonjour vous avez utilisé quoi pour créer le serveur raspisms ? Raspberry ou docker ?
> 
> Je vous pose la question car je suis bloqué, je n'arrive pas à finir l'installation

J'ai utilisé RaspBerry pour l'instant mais des que j'aurais reussi a trouver le moyen de faire fonctionner RaspiSMS derriere un Reverse Proxy je compte bien faire en sorte d'utiliser docker à la place mais deja il faut trouver le pourquoi RaspiSMS ne fonctionne pas correctement derriere un Rev Proxy le code dois contenir une erreur mais je suis pas dev donc j'arrives pas a identifier l'endroit où RaspiSMS indique les urls en dur a la place d'urls relatives 

Si jamais vous avez reussi a faire fonctionner RaspiSMS derriere un RevProxy n'hesité pas a me dire

GBInformatics, 2024-06-15:
Bonjour, j'ai installé également sur Raspberry pi 32 bits.
Je suis bloqué après la connexion, il met affiché que le dossier n'exsite pas..

vincent1890, 2024-06-15:
> Bonjour, j'ai installé également sur Raspberry pi 32 bits. Je suis bloqué après la connexion, il met affiché que le dossier n'exsite pas..

J'ai répondu sur votre Issue https://github.com/RaspbianFrance/raspisms/issues/220

vincent1890, 2024-06-19:
[https://github.com/RaspbianFrance/RaspiSMS/issues/76](https://github.com/RaspbianFrance/RaspiSMS/issues/76)